MarlinFirmware/Marlin #8567: [FR] Live checking of bed boundaries

jjarven, 2017-11-26:
I mistakenly configured the bed size to be too small. The prints were mostly fine but some prints, the extruder stuck at some point (the bed limit) and extruded all the material to same place.

First I blamed a different slicer and settings but I finally figured out the issue and changed the boundaries and prints are now fine.

Thus, a check for bed x & y size with extruder head location and an error message stating the head would try to move out of bounds.

boelle, 2019-02-19:
@jjarven this has been added

in config you define bed size and min and max pos and during compile you will get an error if the configs dont match

remember to use latest bugfix 2.0

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

InsanityAutomation, 2019-06-22:
@boelle what you stated isn't what he's looking for at all. When softendstops trigger he's look for a serial output warning at runtime. I'd say this stays open and may toss a 2.1 milestone tag on it as I wouldn't consider it a blocker for 2.0

boelle, 2019-06-22:
ahhh... i agree, just needed an explanation