MarlinFirmware/Marlin #8568: PROBE_MANUALLY_STEP 0.025 does not work as expected

Hywelmartin, 2017-11-26:
------------------------------------------------------------
### Bug Report
- Description: I have activated this function i configuration.h 
// Set the steprate for papertest probing
#define PROBE_MANUALLY_STEP 0.025
- Expected behaviour: during delta auto calibration the effector should move in 0,025 increments as the display shows.
- Actual behaviour: the effector moves in 0,1mm increments, every 4th click of encoder
- Steps to reproduce:
  - set up delta auto calibration, manual probe and PROBE_MANUALLY_STEP 0.025
  - run G33 and turn knob...

my setup:
Mini kossel from Anycubic, slightly modified. No Probe
Marlin 1.1.6

Babystepping works.
I have 100 steps/mm

configuration.h and configuration_adv.h attached 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1503804/Configuration.zip)

AnHardt, 2017-11-26:
```cpp
//
// ENCODER SETTINGS
//
// This option overrides the default number of encoder pulses needed to
// produce one step. Should be increased for high-resolution encoders.
//
//#define ENCODER_PULSES_PER_STEP 1

//
// Use this option to override the number of step signals required to
// move between next/prev menu items.
//
//#define ENCODER_STEPS_PER_MENU_ITEM 5
```
???

Hywelmartin, 2017-11-26:
I have smooth function of the knob..
my settings are:.... Aha I added the wrong files in the first  post... SORRY

`#define ENCODER_PULSES_PER_STEP 4

//
// Use this option to override the number of step signals required to
// move between next/prev menu items.
//
#define ENCODER_STEPS_PER_MENU_ITEM 1

/**
 * Encoder Direction Options
 *
 * Test your encoder's behavior first with both options disabled.
 *
 *  Reversed Value Edit and Menu Nav? Enable REVERSE_ENCODER_DIRECTION.
 *  Reversed Menu Navigation only?    Enable REVERSE_MENU_DIRECTION.
 *  Reversed Value Editing only?      Enable BOTH options.
 */

//
// This option reverses the encoder direction everywhere.
//
//  Set this option if CLOCKWISE causes values to DECREASE
//
#define REVERSE_ENCODER_DIRECTION

//
// This option reverses the encoder direction for navigating LCD menus.
//
//  If CLOCKWISE normally moves DOWN this makes it go UP.
//  If CLOCKWISE normally moves UP this makes it go DOWN.
//
#define REVERSE_MENU_DIRECTION
`

Hywelmartin, 2017-11-26:
I will test 0,02 ( 2 microsteps?)

AnHardt, 2017-11-26:
```cpp
/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 80, 760*1.1 }  // default steps per unit for Kossel (GT2, 20 tooth)
```

````cpp
// Moves (or segments) with fewer steps than this will be joined with the next move
#define MIN_STEPS_PER_SEGMENT 6
```

6 x 1/100 = 0.06mm

Hywelmartin, 2017-11-26:
So support...

If I lower that to, 3. Will that have any drawbacks, like studdering printer while printing, overload from to many moves?

AnHardt, 2017-11-26:
For that short moves the most load is produced not by the stepper interrupt, but by planning them. `MIN_STEPS_PER_SEGMENT` is the chance to abort that move/segment before too much power is invested in a  move you can't see anyway. Because `DELTA_SEGMENTS_PER_SECOND`, especially when moving slow, most segments can be thrown away early. Lowering 'MIN_STEPS_PER_SEGMENT' mans shifting down the speed where your DELTA gets unusable because of stuttering.

Try it!

Hywelmartin, 2017-11-26:
I settled at MIN_STEPS_PER_SEGMENT 5 an didn't notice any odd behavior.....

closed 2017-11-26 by Hywelmartin

gbyczyns, 2017-12-02:
In my opinion MIN_STEPS_PER_SEGMENT makes sense only when printing.
Wouldn't it be better to disable it completely during the bed leveling procedure?

Hywelmartin, 2017-12-05:
I agree with @gbyczyns so I opened a feature request https://github.com/MarlinFirmware/Marlin/issues/8664