MarlinFirmware/Marlin #8573: Weird not continuous movement

Dabihu, 2017-11-26:
### Bug Report
- Description: Weird not continuous movement at a specific speed and configuration
- Expected behaviour: Linear movement
- Actual behaviour: Discontinuous movement
- Steps to reproduce:
Use default 1.1.6 configuration with these modification:
#define COREXY
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 100, 100, 4000, 500 }
(M502 used, but EEPROM disabled, so M500 have no effect)

Example command:
G1 F1000 X100 Y100

More options:
- F in range: 849..1154
- Diagonal move (only one motor used (X or Y) to movement)
- CoreXY or CoreYX

F1000 video (bad movement): https://youtu.be/wIX8tTg1UyI
F1155 video (good movement): https://youtu.be/CGDx70LtVvE

Behavior is independent from motor driver. The step signal is discontinous in such settings.
Clicking noise in the good video caused by a faulty idler not by the motor.
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1504299/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/1504300/Configuration_adv.h.txt)

coolio986, 2017-11-26:
I also have had this on version 1.1.3 - 1.1.5 (have not tried new versions) using CoreXY only. X/Y and E motors "pause" for a short period (100-200ms) then continue movement. I can confirm in the above post that it occurs with these conditions:

- F in range: 849..1154
- Diagonal move (only one motor used (X or Y) to movement)
- CoreXY or CoreYX

tcm0116, 2017-11-27:
I've noticed something similar on my Cartesian machine. Long travel movements appear to be split into 2 or more movements with a slight pause between.

thinkyhead, 2017-11-27:
@tcm0116 Does this only happen with bed leveling on? Or, does it not matter? Try adding a log line to the `planner.buffer_line` function so we can see whether the move is generating a single stepper block. Then, once we know whether it's a single stepper block, then we can check whether there's something in the stepper ISR that could lead to such a pause.

coolio986, 2017-11-27:
I have bed leveling turned off. I will add a log line and report back.

tcm0116, 2017-11-27:
For me, it was happening during printing with Bilinear leveling on during
long travel movements. As such, I do think it was happening at the grid
boundaries, so my issue may not be directly related to this one. However,
there could be an underlying issue with the planner stitching together
continuous movements in certain scenarios.

AnHardt, 2017-11-27:
Is a display involved?

bjarchi, 2017-11-27:
> For me, it was happening during printing with Bilinear leveling on during
long travel movements. As such, I do think it was happening at the grid
boundaries, so my issue may not be directly related to this one. However,
there could be an underlying issue with the planner stitching together
continuous movements in certain scenarios.

I saw this happen for a bit after I had started using bilinear and it was quite frustrating. I eventually realized that the problem was my z-jerk setting. At some point I had set z-jerk to 0.0 for testing and accidentally committed it with `M500`. With z-jerk at 0, the planner is **forced** to stop at any grid point where the sign of z-accel changes (as far as I could tell, at least). 

Resetting my z-jerk to a small non-zero value (0.3) restored continuous linear movement across grid point boundaries, even when the leveling direction changes.

Dabihu, 2017-11-27:
Default configuration doesn't have display, eeprom or bed leveling. So nothing. But of course everything is connected, but not working.

AnHardt, 2017-11-27:
What is the start position?
```
G1 F1000 X100 Y100
```
Is only then a 'one motor move' if the start position is somewhere with X==Y.
Is the start position in the videos constant?

Dabihu, 2017-11-28:
Video is misleading, because it's move from X100 Y100 to X0 Y0.
In CoreXY or CoreYX setup diagonal (45 degree) move use only one motor.
Briefly dX=dY then X motor is used, dX=-dY then Y motor is used.
Start, end, length doesn't matter:
G1 F1000 X0 Y0 <- good
G1 X20 <- good
G1 X120 Y100 <- bad
G1 X20 <-good
G1 X70 Y50 <-bad

coolio986, 2017-11-28:
I do not have a screen attached when printing, I am not using bed leveling. 
When printing at 15mm/s (900mm/m) doing a circle / square / rectangle/ anything, with 45deg infill this issue occurs for me.

The motor making the moves (X or Y, but not both), and E both pause for brief moments.

tcm0116, 2017-11-28:
@thinkyhead my issue is definitely unrelated to this one. Mine is occurring at the bilinear grid boundaries as the printer is coming to a complete stop as it crosses the boundary. It seems to be a function of the amount of change in Z over the grid and the speed of the movement. In any case, I'll move my issue elsewhere.

bjarchi, 2017-11-28:
@tcm0116 Did you see my earlier comment about checking your z-axis jerk setting? 

> my issue is definitely unrelated to this one. Mine is occurring at the bilinear grid boundaries as the printer is coming to a complete stop as it crosses the boundary. It seems to be a function of [...]

This is exactly what I saw when I first set up bilinear leveling, due to accidentally having set z-jerk to zero. 

Use `M205` to check - 'advanced settings', z jerk will be labeled Zx.xx ...
and use `M205 Zx.xx` (ex `M205 Z0.30`) to set to a non-zero value

bjarchi, 2017-11-28:
@coolio986 FWIW, this could occur with jerk set to 0 (or maybe even too low?) on any axis, if the planner is forced to stop due to motion constraints. 

If you don't have EEPROM enabled it's not super likely - the values in configuration.h you posted look OK - but it *could* be getting set by a startup script or something - in which case it would still persist until the next machine restart or until it's reset by gcode. Might be worth using `M205` to make sure something hasn't set one of your jerk values too low.

Roxy-3D, 2017-11-28:
Does this issue happen when you cross over from one EEPROM version number to the next one?   I'm seeing something similar except with the BL-Touch probes going wonkers and the nozzle crashing into the bed.  Once I do a M502, M500, M501 everything is OK again.

tcm0116, 2017-11-28:
@bjarchi I did see your comments about the jerk settings. My z axis jerk was set to the default of 0.4. I raised it all the way to 5, but still got the stops. I also played around with the acceleration, which only confirmed that the movement is being slowed/stopped at the bilinear grid lines. 

@Roxy-3D I have not tried to reset the EEPROM, so I will give that a try. However, the printer is working well otherwise, so I don't think that'll resolve the issue. I need to add some printouts to the planner to see exactly what's going on. It sort of seems like an issue with the jerk processing, but I need to investigate further.

Dabihu, 2017-11-28:
Start from X0 Y0
G1 F1000 X40 Y25
Logic analyser connected to Step pins (Channel0: X, Channel1: Y)
Full movement:
[image]
Zoom:
[image]
More zoom:
[image]
Timeline and changes in excel:
[X40_Y25.xlsx](https://github.com/MarlinFirmware/Marlin/files/1509278/X40_Y25.xlsx)
Raw logic data:
[X40_Y25.zip](https://github.com/MarlinFirmware/Marlin/files/1509280/X40_Y25.zip)

AnHardt, 2017-11-28:
@Dabihu 
Thanks for the numbers.
Does the stepper interrupt use doublestep mode from F1155 on?

Dabihu, 2017-11-28:
No. No difference, but it's continuous.
Same setup, F1155:
Full:
[image]
Zoom:
[image]
Timeline and changes in excel:
[F1155_X40_Y25.xlsx](https://github.com/MarlinFirmware/Marlin/files/1509870/F1155_X40_Y25.xlsx)
Raw logic data:
[F1155_X40_Y25.zip](https://github.com/MarlinFirmware/Marlin/files/1509871/F1155_X40_Y25.zip)

Bob-the-Kuhn, 2017-11-30:
@Dabihu - I've been able to re-create your issue.  I'm about four levels into the software but not yet at root cause.  I do know that it's getting screwed up because the start stepping rate is higher than the maximum/nominal rate.

Hopefully I'll have root cause tomorrow and a solution.

thinkyhead, 2017-12-04:
One of these changes might solve the issue. Please try each one and see if it helps:

```diff
- if (moves_queued && !UNEAR_ZERO(previous_nominal_speed)) {
+ if (moves_queued > 1 && !UNEAR_ZERO(previous_nominal_speed)) {
```
---
```diff
   // Always split the first move in two so it can chain
   if (!blocks_queued()) {
+    previous_nominal_speed = 0;
     DISABLE_STEPPER_DRIVER_INTERRUPT();
```
---
```diff
-   #define _BETWEEN(A) (position[A##_AXIS] + target[A##_AXIS]) >> 1
+   #define _BETWEEN(A) (position[A##_AXIS] + target[A##_AXIS]) / 2
```
Since I've reverted the changes in bugfix, you can start with one of these branches:
- https://github.com/thinkyhead/Marlin/tree/backup_1_split_first
- https://github.com/thinkyhead/Marlin/tree/backup_2_split_first

Dabihu, 2017-12-04:
Sorry thinkyhead, but this branches doesn't helps at all. It's worse than before.

Tried without any modification and the 3 modification separately, but all time this happened:

- Configuration changed:
#define COREXY
#define DEFAULT_AXIS_STEPS_PER_UNIT { 100, 100, 4000, 500 }

- No home, nothing. Just give this command after upload:
G1 F1000 X10 Y10

X motor moves like before (Length=sqrt(2)*10mm), non continuously to X=10,Y=10
After reach this position the X motor moves forward more AND Y motor too continuously sqrt(2)*10mm length then stops.

- Give the same command again:
G1 F1000 X10 Y10
X motor and Y motor moves continuously sqrt(2)*10mm length then stops.

- Repeat this as long as you want.

thinkyhead, 2017-12-04:
> Sorry thinkyhead, but this branches doesn't helps at all. It's worse than before.

Those branches would have been broken 30 minutes ago, as they contained the same errors.

Now however, they are patched.

Dabihu, 2017-12-04:
Better, but the same where we started.

- MarlinFirmware:bugfix-1.1.x merged, where the first of three modification was in the code, same movement.
- After that applied the other two modification in one step, same movement.

hg42, 2017-12-04:
@Dabihu the bugfix-1.2.x/2.0.x should not contain any of the three mods

Dabihu, 2017-12-04:
bugfix-1.1.x / planner.cpp:
- Line 1265 not changed:
if (moves_queued > 1 && !UNEAR_ZERO(previous_nominal_speed)) {
- Line 1444 modified:
#define _BETWEEN(A) (position[A##_AXIS] + target[A##_AXIS]) / 2
- Line 1446 added:
previous_nominal_speed = 0;

Or I don't understand what I should do.

hg42, 2017-12-05:
I'm not sure, @thinkyhead thinks in his head :-), but these three changes were attempts to solve a problem that is solved in both bugfix branches.

I suggest, you simply use the head of the branch without modifying it yourself

Apart from that,
the first seems to be in the bugfix branch, already. I guess thinkyhead knows why. It seems to be ok from my view. The count of queued moves will probably not go below zero, but it shouldn't hurt.
The second also shouldn't hurt, because dividing by two is the original idea, somewhat optimized by the shift, but I think the compiler will optimize it anyways.

The last one might cause ramp down, ramp up between moves, because I had this when I used this line while testing with the other bug (which could be caused by that bug).
I would not include it.

hg42, 2017-12-05:
but let me test it myself...

hg42, 2017-12-05:
ok, for me (using corexy and bugfix-2.0.x/32bit) the split move results in a ramp down and up in the middle.
Do you experience the same behavior?

hg42, 2017-12-05:
> Apart from that,
> the first seems to be in the bugfix branch, already. I guess thinkyhead knows why. It seems to be ok from my view. The count of queued moves will probably not go below zero, but it shouldn't hurt.

Well, something in my head didn't want to see that it's
  moves_queued > **1**
I thought about it as if it would be
  moves_queued > **0**
which would be the same like before.

So, this one is the reason why it ramps down and up between the first two moves in the current queue.

I am not sure, if @thinkyhead intentionally included this change in the latest commit.
I assume it's unintentionally, because he started from his backup branches.
This wasn't the variant, I tested here.
My patch was applied to the original PR #8611

**I suggest you remove the "> 1" from moves_queued**

Note, I am not using bed leveling currently. My tests only addressed the weird movements.
I don't know, if your original issue will be solved.

hg42, 2017-12-05:
I just read the whole thread and I guess I'm on the wrong track here.
I'll go to bed no and think about it again tomorrow. Sorry, if I told non-sense.

At least one fact remains: moves_queued > 1 doesn't work for me.

thinkyhead, 2017-12-05:
> `moves_queued > 1` doesn't work

Note that the bugfix branches were patched ~5 hours before the above post.

AnHardt, 2017-12-05:
@thinkyhead 's tree suggested patches are a typical reaction of an experienced programmer having no time to test himself but has to react somehow and fast.

If you have the idea an array index might become below zero it's a good idea to go back to the code used before.
```diff
- if (moves_queued && !UNEAR_ZERO(previous_nominal_speed)) {
+ if (moves_queued > 1 && !UNEAR_ZERO(previous_nominal_speed)) {
```
But the effect is - the second half of the splitted move will start at zero speed - always - like without splitting the first move the second started always with zero, but this time in the middle of the first (splitted) move.

In
```diff
   // Always split the first move in two so it can chain
   if (!blocks_queued()) {
+    previous_nominal_speed = 0;
     DISABLE_STEPPER_DRIVER_INTERRUPT();
```
the suspicion is a somehow uninitialized variable. Does not hurt, but would n't have worked before we invented splitting.

```diff
-   #define _BETWEEN(A) (position[A##_AXIS] + target[A##_AXIS]) >> 1
+   #define _BETWEEN(A) (position[A##_AXIS] + target[A##_AXIS]) / 2
```
makes sense when your suspicion is, one of the variables might not be an integer but float. With time to verify you will find - both are int32_t - but that takes a while.

Why this selection of suggestions? Because he believed in the correctness of his own code but not mine.

---------------------------------------------------------------

And now let's return here to this (https://github.com/MarlinFirmware/Marlin/issues/8573#issuecomment-347469235) problem. The 'split fist move' patch is not related to that and was by purpose splitted out to 'Pausing at bilinear grid boundaries #8595'
Sadly i have no idea what's going wrong here. Looks a bit as if the next timer interval in the Bresnham is calculated wrong sometimes. But under what circumstances?

AnHardt, 2017-12-05:
Did i understand right now? For the not continuous move on DELTAs it's not a requirement to have a 'one motor move only' but it occurs even if only one motor is involved.

AnHardt, 2017-12-05:
The stuttering moves are "single moves" - not further subdivided by grid lines, or delta-subdivision or similar. (now splitted by "first-move split" - but this issue is older)

Dabihu, 2017-12-05:
@AnHardt Sometimes I don't understand what everybody talking about, what I wrote is the most simple case, where the bug appear.
- Default settings with some modification. (Wrote above)
- Simple short or long move (G1 F1000 X10 Y10)
In CoreXY it's use only the X motor.
- Almost diagonal simple move (G1 F1000 X40 Y25)
Use X and Y motor. Step timings captured above with logic analyser.
- No leveling, nothing. That's all.
- Used Arduino IDE 1.8.5 compiler.

hg42, 2017-12-05:
@AnHardt 
> Note that the bugfix branches have this patched since just before @Dabihu last posted.

I don't think so...
He (using 32bit?) and I merged 13h and 12h ago. @thinkyhead added a followup 10h ago which disabled the `moves_queued > 1`.
So we both got the separated split moves (with ramps in between).
I don't see this follow up in bugfix-1.x.x, that explains why others did not have this problem.
That's why I got confused, because I read "it works" and " it still doesn't work" at the same time.
Also, I first tested with my version, which didn't include the "> 1", so I didn't expect, it wouldn't be solved in bugfix-2.0.x

but we got it finally, everything's ok :-)

hg42, 2017-12-05:
> He (using bugfix-2.0.x?) and I merged 13h and 12h ago.

no, @Dabihu uses 1.1.6. Which I can't (32bit). So I'm lost here.
And my assumption about the follow up commit was wrong for his case (but not mine).

I am on the way trying to reproduce the issue, because I also have a corexy and want to see, if bugfix-2.0.x does behave similar. I'll use @Dabihu's last description. Let's see.

hg42, 2017-12-05:
unfortunately my oscilloscope isn't able to capture this.
Does it create such pauses when running very slow, say F100?

At least all moves I tried from F100 to about F6000 were totally smooth.
I multiplied the length and my 0,0 is in the center, so I moved between -160, -100 and +160, 100 if that matters in some way. But the requested moves didn't do the same thing.

I also found, that @thinkyhead also removed the "> 1" from bugfix-1.x.x at a similar point in time, but used a different title, so @Dabihu got this other behavior, too.

AnHardt, 2017-12-05:
@Dabihu 
Please try (stepper.cpp)
```diff
@@ -742,12 +742,13 @@ void Stepper::isr() {
 
       eISR_Rate = adv_rate(e_steps[TOOL_E_INDEX], OCR1A_nominal, step_loops_nominal);
 
     #endif
 
-    SPLIT(OCR1A_nominal);  // split step into multiple ISRs if larger than  ENDSTOP_NOMINAL_OCR_VAL
-    _NEXT_ISR(ocr_val);
+//    SPLIT(OCR1A_nominal);  // split step into multiple ISRs if larger than  ENDSTOP_NOMINAL_OCR_VAL
+//    _NEXT_ISR(ocr_val);
+    _NEXT_ISR(OCR1A_nominal);
     // ensure we're running at the correct step rate, even if we just came off an acceleration
     step_loops = step_loops_nominal;
   }
 
   #if DISABLED(LIN_ADVANCE)
```
While cruising constantly this is about the only place where the timing could get a variation.
Do i have to dig in SPLIT() or do i have to search for something else.

Dabihu, 2017-12-05:
Because I don't know what cause the issue, I could not trigger with other parameters.
CoreXY needed, 100 axis step per unit needed, and F in the range 849..1154.
And as mentioned before not all direction cause the issue.
Maybe it depends on cpu clock frequency or system architecture too.

Dabihu, 2017-12-05:
@AnHardt I used bugfix-1.1 83a1a701032a100dbbfa617c3fafdae6bf962eb4 as base. It has the issue.
Changed what you wrote, but didn't helped. (Tried G1 F1000 X10 Y10)

hg42, 2017-12-05:
I can not reproduce this on bugfix-2.0.x and 32bit with
* 94cb412e45b8e719775960312a9241f39e2a5ae2 (should be around that mentioned commit from bugfix-1.1.x)
* 91fb6231b2a779c86b15a552add2e16aa09ce5f1 (next and last time the branches were merged)

Can you run bugfix-2.0.x? does it happen there, too?

hg42, 2017-12-05:
given that I have LPV1768 this may be an 8bit or AVR issue.
Or it is 1.1.x vs. 2.0.x.

Dabihu, 2017-12-05:
Tried bugfix-2.0.x 4d3c3d0b7384038e12c6eddb14bf717985b8dd7a too. Same result.
Just two modification in the Configuration.h:
#define COREXY
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 100, 100, 4000, 500 }

hg42, 2017-12-05:
do you know a commit, where it still worked?
the usual strategy would be to narrow the range between the working and failing commits by going up the tree with the working and down with the failing. I usually tag the results with OK1, OK2, ... and the failing with FAIL1, FAIL2, ...
At the end of the process, you hopefully get the one commit that caused the hassle.
Sometimes this is a really big commit (like merging), but often it's a simple commit and we could easiliy see, what change does cause this.
I would do this myself, but this only works if I can reproduce the problem (which I tried).

hg42, 2017-12-05:
I used your steps/unit which also worked.

Dabihu, 2017-12-05:
First iteration:
1.1.0-RC7: Working great!
1.1.0-RC8: Have this issue.

Bob-the-Kuhn, 2017-12-05:
I'll check out RC7.

I have a solution that I'm almost ready to show to the world.  I need to do some testing on delta configurations first.

tcm0116, 2017-12-05:
@Bob-the-Kuhn I've got some free time this evening if you want some help
testing on a Delta and/or Cartesian machine.

Bob-the-Kuhn, 2017-12-05:
I'll attach the file in an hour or two.

The root cause is the planner doesn't properly handle in all cases that the CORE machines always have one stepper that is running faster than the head.  The result is negative acceleration steps being passed to the stepper routine in a band of feed rates.  The stepper software doesn't react nicely to the negative steps.

hg42, 2017-12-05:
but why don't I see this on my machine (also corexy)?

Bob-the-Kuhn, 2017-12-05:
Here's the new planner.cpp for testing.
[planner.zip](https://github.com/MarlinFirmware/Marlin/files/1533348/planner.zip)

@hg42 - I expect that you're not using the affected feed rates.  If you give me your  steps_per_unit and default jerk settings I'll calculate the feed rate band where this occurs.

AnHardt, 2017-12-06:
@Bob-the-Kuhn 
Where did you start from. Diff gives a lot unrelated changes to me, in today's bugfix branches.

Bob-the-Kuhn, 2017-12-06:
I just double checked the uploaded ZIP file.  Those are my intended changes.

Jerk, speeds and rates are all related.

Dabihu, 2017-12-06:
@Bob-the-Kuhn Your source was around 972248c3334ae5432afdcd180fea80653d41beb1 but I can't find the correct one.

Bob-the-Kuhn, 2017-12-06:
As best I can tell, today's bugfix1.1.x planner.cpp is the same as it was about a week ago before all the split (and revert) activities.

Bob-the-Kuhn, 2017-12-06:
See PR #8680 for a better explanation of what I did.

Bob-the-Kuhn, 2017-12-06:
Yep - I'm not working off the latest code base.  I'll be back after I've figured out the problem and have done a few sanity checks.

hg42, 2017-12-06:
M92 X80.000000 Y80.000000 Z400.000000 E162.000000
M205 S0.000000 T0.000000 B20000 X20.000000 Y20.000000 Z0.400000 E5.000000

[33 more replies]

closed 2017-12-17 by Dabihu