MarlinFirmware/Marlin #8576: UBL 3 Point Tilting of mesh produces unexpected results

Koshu, 2017-11-27:
I build a new printer (with probe and z-max endstop) and played around with UBL to get the 3 point tilt (G29 J) compensation working. My understanding of what it should do or at least what I hoped it would do is the following (If it was not supposed to work that way then I guess this becomes an feature request rather than a bug report): 

You initially build your mesh (G29 P1 etc etc) which takes quite bit of time and than do 3 point tilt compensation right before every print. The 3 point tilt compensation then calculates the tilt around the x and y axes as well as displacements along the z axis and corrects them in the mesh. The result would be a perfect leveled mesh as long as the warping of the bed didn't change.

Now, while trying out this setup I noticed that my mesh started to look really weird with values that should not be possible (20 mm and more offset of the mesh). I found a way to reproduce the error. For readability I changed my mesh to an 3x3 Grid and Zeroed all Mesh points:

First the initial mesh:
```
>>G29 T
Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
  0.000    0.000  [ 0.000] 
 
  0.000    0.000    0.000  
 
  0.000    0.000    0.000  
(20,20)            (200,20)
(0,0)             (2,0)
ok
```

Then I executed the compensation:
```
>>G29 J V4 T
Bed X: 21.000 Y: 21.000 Z: 0.352
echo:busy: processing
echo:busy: processing
echo:busy: processing
Bed X: 199.000 Y: 21.000 Z: 0.312
echo:busy: processing
echo:busy: processing
Bed X: 110.000 Y: 199.000 Z: 0.387
echo:busy: processing
bed plane normal = [0.0002247,-0.0003090,1.0000000]
rotation matrix:+1.000000 +0.000000 -0.000225 
+0.000000 +1.000000 +0.000309 
+0.000225 -0.000309 +1.000000 
D constant: 0.3507303 

Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
 -0.293   -0.314   -0.334  
 
 -0.321  [-0.341]  -0.362  
 
 -0.349   -0.369   -0.389  
(20,20)            (200,20)
(0,0)             (2,0)
ok
```
So far so good. The values doesn't seem to be to much off and might or might not be correct. Also, I expected the mesh points near the probe points to be a bit closer than that. It starts to get really weird as soon as you execute G29 J multiple times:

```
Bed X: 21.000 Y: 21.000 Z: 0.355
echo:busy: processing
echo:busy: processing
echo:busy: processing
Bed X: 199.000 Y: 21.000 Z: 0.310
echo:busy: processing
echo:busy: processing
Bed X: 110.000 Y: 199.000 Z: 0.382
echo:busy: processing
bed plane normal = [0.0000281,0.0000281,1.0000000]
rotation matrix:+1.000000 +0.000000 -0.000028 
-0.000000 +1.000000 -0.000028 
+0.000028 +0.000028 +1.000000 
D constant: 0.7051403 

Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
 -1.298   -1.341   -1.384  
 
 -1.351  [-1.394]  -1.437  
 
 -1.404   -1.447   -1.490  
(20,20)            (200,20)
(0,0)             (2,0)
ok
echo:busy: processing
Bed X: 21.000 Y: 21.000 Z: 0.355
echo:busy: processing
echo:busy: processing
echo:busy: processing
Bed X: 199.000 Y: 21.000 Z: 0.312
echo:busy: processing
echo:busy: processing
Bed X: 110.000 Y: 199.000 Z: 0.385
echo:busy: processing
bed plane normal = [-0.0002388,0.0003020,1.0000000]
rotation matrix:+1.000000 +0.000000 +0.000239 
+0.000000 +1.000000 -0.000302 
-0.000239 +0.000302 +1.000000 
D constant: 1.7605686 

Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
 -4.413   -4.477   -4.541  
 
 -4.492  [-4.556]  -4.620  
 
 -4.571   -4.635   -4.699  
(20,20)            (200,20)
(0,0)             (2,0)
ok
```

Now, if the compensation would work as (I) expected than executing the command multiple times should always produce the exact same result as the tilt hasn't changed. Instead the error of the mesh gets larger and larger with each iteration.

**One possible Solution**
I digged through the code to find out what is causing this behavior but  I couldn't wrap my head around how the rotation matrix is constructed and how the rotation itself is executed. Might be an easy fix but that has to be done by someone who knows this code better than I do. Since I need this function for my new printer I decided to rewrite the function with a simpler approach and might as well share it. This is what I came up with:

```
    void unified_bed_leveling::tilt_mesh_based_on_3pts(const float &z1, const float &z2, const float &z3) {
      matrix_3x3 rotation;
      vector_3 v1 = vector_3( (UBL_PROBE_PT_1_X - UBL_PROBE_PT_2_X),
                              (UBL_PROBE_PT_1_Y - UBL_PROBE_PT_2_Y),
                              (z1 - z2) ),

               v2 = vector_3( (UBL_PROBE_PT_3_X - UBL_PROBE_PT_2_X),
                              (UBL_PROBE_PT_3_Y - UBL_PROBE_PT_2_Y),
                              (z3 - z2) ),

    //First we calculated the tilted plane using the point-normal form, with va being the normal and va being the receptor point.
    //The tilt of original mesh has already been substracted in the previous function. No extra steps required.
               va = vector_3(UBL_PROBE_PT_2_X, UBL_PROBE_PT_2_Y, z2),

               vn = vector_3::cross(v1, v2);

      for (uint8_t i = 0; i < GRID_MAX_POINTS_X; i++) {
        for (uint8_t j = 0; j < GRID_MAX_POINTS_Y; j++) {
          float x_tmp = mesh_index_to_xpos(i),
                y_tmp = mesh_index_to_ypos(j),
                z_tmp = z_values[i][j];

    //We translate the point-normal form of the plane to the general form and solve for z.
          float z_shift = -(vn.x*x_tmp + vn.y*y_tmp - vn.x*va.x - vn.y*va.y - vn.z*va.z) / vn.z;

          z_values[i][j] += z_shift;
        }
      }
    }
```
Instead of doing a accurate rotation of the points this function uses more of an projection by translating the offsets of the original (tilted) plane to the new measured plane. I will do a bit more testing but seems to work quite nicely:

Same steps as above. Original Mesh:
```
Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
  0.000    0.000  [ 0.000] 
 
  0.000    0.000    0.000  
 
  0.000    0.000    0.000  
(20,20)            (200,20)
(0,0)             (2,0)
ok
```

Executing G29 J multiple times:
```
Bed X: 21.000 Y: 21.000 Z: 0.355
echo:busy: processing
echo:busy: processing
echo:busy: processing
Bed X: 199.000 Y: 21.000 Z: 0.310
echo:busy: processing
echo:busy: processing
Bed X: 110.000 Y: 199.000 Z: 0.392
echo:busy: processing

Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
  0.416    0.393    0.370  
 
  0.385  [ 0.362]   0.340  
 
  0.355    0.332    0.309  
(20,20)            (200,20)
(0,0)             (2,0)
ok
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
ok

Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
  0.412    0.390    0.369  
 
  0.382  [ 0.361]   0.339  
 
  0.352    0.331    0.309  
(20,20)            (200,20)
(0,0)             (2,0)
ok
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
ok

Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
  0.417    0.393    0.369  
 
  0.387  [ 0.363]   0.339  
 
  0.357    0.333    0.309  
(20,20)            (200,20)
(0,0)             (2,0)
ok
```
Even if executed multiple times G29 J now produces always the same mesh (in margin of the measurement deviation). Further, the mesh points close to the probe point now have an accurate value.

Roxy-3D, 2017-11-27:
I'll try to check for an regressions in the 3-point leveling today.   It used to work exactly as you say you expect it to work.

AnHardt, 2017-11-27:
?  AxB = -(BxA)
Clockwise vs. anticlockwise definition of the points ?

Koshu, 2017-11-27:
@Roxy-3D Where would the regression come into place, with only 3 points the tilting plain completely defined, or for a different part of the algorithm?

@AnHardt Sorry, I don't quite follow. My Points are defined counterclockwise. I think the original algorithm checked for the resulting direction of the normal vector, so it shouldn't be a problem(?).

Koshu, 2017-11-27:
Update: In case someone is interested. Tested my approach with a 9x9 grid and I am happily printing at the moment. No Problems yet.

Roxy-3D, 2017-11-27:
The regression would have come from changes to the Marlin code base in which this feature wasn't tested.    Not many people use mesh tilting.   A well tuned mesh is usually sufficient for most printers.   And for people that do tilt their mesh, usually they use the grid base (n x n ) tilting.

@AnHardt   For the grid based tilting, the locations of the probed points is automatically taken into account.   The Least Squares Fit algorithm automatically factors that in.   But I'm wondering if you could be right that the order the points are probed matters in 3-Point leveling.   

Or here it is stated in different words:    The 3 points define a plane.   But if they are done in the wrong order, it is possible to get the sign of the plane normal vector in the exact opposite direction.   I guess that is what you are saying with Clockwise vs. anticlockwise definition of the points ?

AnHardt, 2017-11-27:
@Roxy-3D 
Does it need a definition of these 3 points at all?
Couln't we pick for example A[minX, minY], B[minX, maxY], C[maxX, minY] from the meshs points?

Roxy-3D, 2017-11-27:
From a backwards compatability perspective...    In order to emulate the original 3-point leveling...   The mesh is set to all zeros.    So, in that case we don't have any data to operate on.

But the bigger point is this:    If the person has a well defined mesh that exactly describes the print surface of the printer (such as a piece of glass)...    When that surface is taken off the printer and put in the freezer to release the print and then put back on the printer, it may not go back exactly the same.   A small piece of filament hair can find its way under the glass.   The mesh is still very valid, but now it is tilted.

AnHardt, 2017-11-27:
Even if z is unknown for all mesh-points we know where [x,y]to probe from the definition of the grid.

Roxy-3D, 2017-11-27:
Yes.   I thought you were suggesting to use the mesh point values to determine the tilt.   Maybe I mis-understood your comments.    

Here is a question for you...   Would it make sense to check that the 3 specified probe points are Clockwise in direction?   After thinking about what you said up above, I think it is possible the current 3-point tilt code can get the wrong answer if the points are taken in the wrong direction.   

And here is another question...   The LSF grid tilting code does work.   I wonder if it would make sense to just feed the 3 sampled points into that code and let it do a Least Squares Fit on it???   That has fairly robust checks for the D offset and such.

Roxy-3D, 2017-11-27:
@AnHardt    One more quick question for you:   Do you know a simple way to determine if the points are clockwise or counter clockwise.   I can do it by checking the angles between the points, but that is a lot of complexity for such a simple thing.

Roxy-3D, 2017-11-27:
@Koshu    I do like the code you substituted...     But I want to make sure I understand what you are thinking.  The mesh has no tilt.   But individual points within the mesh have an offset from a perfectly flat plane.  You have the comment up above:
```
//The tilt of original mesh has already been substracted in the previous function. No extra steps required.
```
I think what you really mean is:
```
//The Z values of each probed point have already been corrected using the mesh data for the points location.
```

Koshu, 2017-11-27:
@Roxy-3D Well, both I guess. The original mesh also had some kind of tilt (maybe only a very small one). That tilt shouldn't be part of the the new mesh, since we now know the actual tilt. The subtraction of the correction values from the three measured points takes care of that. But in the end I think we mean the same thing, we just have different words for it.

Roxy-3D, 2017-11-27:
@Koshu    Looking further at your code, you don't have any check in it for the direction of the plane normal vector.    
```
      /**
       * This vector is normal to the tilted plane.
       * However, we don't know its direction. We need it to point up. So if
       * Z is negative, we need to invert the sign of all components of the vector
       */
      if (normal.z < 0.0) {
        normal.x = -normal.x;
        normal.y = -normal.y;
        normal.z = -normal.z;
      }
```
If you swap your first and your last points for 3-point leveling in the Configuration.h file, do you still get the correct results?   My guess is you will be 'correcting' in the wrong direction.   Would you mind checking if that assertion is true?      (I don't want to replace the existing code without knowing the new code is better.)

AnHardt, 2017-11-27:
@Roxy-3D 
You got all the answers i could give you by your selves. :-)

If the test points are not defined by the user the direction is always correct and we can save the test (even if pretty cheap).

The advantage of the 3-point method is - it's cheap and simple.
It's drawback is the limitation to only 3 error prone points.
The drawback of the grid based method is it's expensive and complicated.
The advantage  is to use much more points - where the errors (when random - not systematic) level out each other. The result would be about the same as if probing the 3 points multiple times and averaging these results - just covering a larger area (with 3 points you can't cover more than half the bed - what does not matter because we assume the bed is flat here).
Application of the 'turn data' during print is the same for 3-point and grid.
Compared to the mesh methods the application during print is cheap ,fast and needs no subdividing of the moves.

For the described use case:  same build plate, same orientation, same warp but different tilt.
It would make most sense to me to probe 3 points where [x,y] we already have in the grid, calculate a new normal from them.
Calculate the normal from the same [x,y] points we already have in the stored grid.
Turn the stored grid by the **difference** between the old and the new normal.
Consecutive reprobing can't add up.
Works whether the grid was stored in EEPROM or not.
That way you haven't to calculate the turn for every move but only once for the mesh.
During printing use only the mesh-data.

Reprobing only 3-points is fastest. When reprobing more points the same algorithm as in 'grid' can be used to find the normals.

Koshu, 2017-11-27:
@Roxy-3D It's not necessary, the transition to point normal form takes care of the direction. If the normal vector points in the opposite direction it result in a double negation, so the result will be the same:

-(vn.x*x_tmp + vn.y*y_tmp - vn.x*va.x - vn.y*va.y - vn.z*va.z) / vn.z;

All the products in the the denominator contain a part of the normal vector and are divided by another part of the normal vector. But I can check tomorrow anyway. 

For the second question: There isn't really a rotation, rather a projection, for example if the original mesh is all zeros an the measured tilt is 5° it will look at every point of the mesh calculates at which z-value it will cut the plane (same x/y values). The value of that mesh point (in this case zero) is than added to the cut point. In comparison to an rotation this introduces a small error but at small angles this error is neglectable. On the other side this largely reduces the necessary calculation (like at which point to rotate around).

Koshu, 2017-11-27:
Decided to do the testing right now.
Switched back to a zeroed 3x3 grid.
If the probe point are counterclockwise (21/21 199/21 110/199) this is result (before and after):
```

Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
  0.000    0.000  [ 0.000] 
 
  0.000    0.000    0.000  
 
  0.000    0.000    0.000  
(20,20)            (200,20)
(0,0)             (2,0)
ok
ok

Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
  0.131    0.155    0.179  
 
  0.095  [ 0.119]   0.143  
 
  0.059    0.083    0.107  
(20,20)            (200,20)
(0,0)             (2,0)
ok
```

Clockwise (21/21 110/199 199/21) it produces:
```
Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
  0.000    0.000  [ 0.000] 
 
  0.000    0.000    0.000  
 
  0.000    0.000    0.000  
(20,20)            (200,20)
(0,0)             (2,0)
ok

ok

Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
  0.140    0.160    0.181  
 
  0.096  [ 0.116]   0.136  
 
  0.052    0.072    0.092  
(20,20)            (200,20)
(0,0)             (2,0)
ok
```
Looks fine to me

Roxy-3D, 2017-11-27:
Thank you for the explanation!    

Two questions:   First, it kind of looks like it is tilting the mesh about the origin.  In both examples, the (2,2) coordinate ended up with a much larger movement than any other coordinate.

Second question:  Can you tell me what happens if you do 3 or 4 iterations of your algorithm?   Does it converge and start to not tilt the mesh any more than it is?

Koshu, 2017-11-27:
@Roxy-3D Thats just my bed, I'm still building it and not everything is perfectly aligned yet. Here are the direct results of my probe (new measurement):

```
Bed X: 21.000 Y: 21.000 Z: 0.063
Bed X: 199.000 Y: 21.000 Z: 0.118
Bed X: 110.000 Y: 199.000 Z: 0.190
```

So, that the point (2,2) of the mesh is the highest is actually correct (without proper mesh measurement of course). 

Second question: That works, in my opening post I did three consecutive tilt measurments (without resets between). They all resulted in the same.

Roxy-3D, 2017-11-27:
> Second question: That works, in my opening post I did three consecutive tilt measurements (without resets between). They all resulted in the same.

This is very valuable if true!   Having the sequence converge instead of deviating wildly is very helpful.
I'm going to spend some time playing with this....

bjarchi, 2017-11-27:
@Koshu - FWIW I'm using `G29 J` just as you describe in the initial post and I can suggest a likely workaround while the bug discussion is ongoing. 

I use `G29 J` for transforming a stored mesh by 3-point probing, and I haven't seen the divergence you describe. I only had time to skim the extensive discussion (so far) but I can say that using the following sequence in my start-up g code has proven quite robust:

```
G29 L1
G29 J
```

So make sure that you're loading the stored mesh immediately before transforming it using your `G29 J` probe. 

Doesn't address the potential regression here, but it should give you good print results using this workflow while still staying in sync with mainline Marlin if that's of interest.

Roxy-3D, 2017-11-28:
@Koshu    I ended up going down a different path.    I believe I have the G29 J working to do 3-point leveling as you want it.    But what I did was probe the 3 specified points and feed those into the Least Squares Best Fit algorithm.     The reason for doing this is two fold.   First, it collapses all of the stand alone 3-point leveling code.   But the main benefit is it allows all tilting to use the same algorythm and support functions.   That makes it easier to maintain and support.

And...   the 3-Point leveling turns out to be a perfect test case for that.   Because a plane is defined by 3 non-linear points, after all the calculations are done to tilt the mesh...   Each of the 3 Point's Z values should be obtainable by just calling get get_z_correction(UBL_PROBE_PT_1_X, UBL_PROBE_PT_1_Y).   The value in the mesh ought to be identical to the value the probe measured.

If you have time to check it out, I would appreciate feed back.   It currently resides at:  https://github.com/Roxy-3D/Fast_G26_work

Koshu, 2017-12-01:
Was quite busy this week. Downloading your code now. Will report back when I did some testing.

Roxy-3D, 2017-12-01:
I'm debating if we should add an option to adjust the mesh such that the home point is at 0.000 mm.   

That might cause more confusion than it is worth.   But doing a 3-Point level does cause the home location to have a small offset some times.  It seems to me the home location should be nailed at 0.000 m.

Koshu, 2017-12-01:
Hmm, well that doesn't really work for me. The algorithm isn't producing the same result if executed multiple times (without reset in between):

```
(0,2)           (2,2)
(20,200)        (200,200)
  0.110    0.135    0.160  
 
  0.103  [ 0.129]   0.154  
 
  0.097    0.122    0.148  
(20,20)            (200,20)
(0,0)             (2,0)
ok
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing

Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
  0.227    0.275    0.323  
 
  0.214  [ 0.263]   0.311  
 
  0.202    0.250    0.298  
(20,20)            (200,20)
(0,0)             (2,0)
ok
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing

Bed Topography Report:

(0,2)           (2,2)
(20,200)        (200,200)
  0.344    0.418    0.491  
 
  0.324  [ 0.398]   0.471  
 
  0.304    0.377    0.451  
(20,20)            (200,20)
(0,0)             (2,0)
```

Something seems off when the rotation is calculated. Is there a benefit to do a a complete rotation calculation? 
Otherwise, doing it the other way round and adding the projection method to tilt the results of the LSB algorithm as well might be a viable solution(?). That way the code would create always the same result and we could still reuse the code for both methods.

Koshu, 2017-12-01:
> I'm debating if we should add an option to adjust the mesh such that the home point is at 0.000 mm.
>
> That might cause more confusion than it is worth. But doing a 3-Point level does cause the home [...]

Internally or for the frontend (Gcodes/LCD)?

Roxy-3D, 2017-12-01:
> Hmm, well that doesn't really work for me. The algorithm isn't producing the same result if executed multiple times (without reset in between):

Yes...  I did notice that.   I don't have an answer yet.   But...   The mesh is tilted such that it goes through all three probe points.    It maybe we should do the probing with correction enabled.

Koshu, 2017-12-01:
> It maybe we should do the probing with correction enabled.

Wouldn't that complicate the algorithm even more?

> I need to figure out if the mesh is getting tilted more and more... Of if it is just getting offset more and more.

To me it looks like its tilted more and more. In my last example the point in the upper right and lower left corner (after first and last run) have a deviation of 0.331 and 0.207. So, the points are moved down/up by a difference of 0.124. My probe shouldn't be that far off, also I can't tell for sure.

AnHardt, 2017-12-01:
For ordinary 3-point leveling the procedure before reprobing is not reboot but home.

Koshu, 2017-12-01:
> For ordinary 3-point leveling the procedure before reprobing is not reboot but home.

Sorry, I don't follow. I never rebooted. I homed once before trying the 3-point tilt compensation.

Roxy-3D, 2017-12-02:
As an option to G29's J commands.   Maybe a Z  (for Zero-Adjust) the home location.   Move the whole mesh up or down such that the nozzle just barely touches the bed at the home location with Z=0.000

thinkyhead, 2018-02-10:
Presuming solved.

closed 2018-02-10 by thinkyhead

bcdiaconu, 2020-11-07:
> Hmm, well that doesn't really work for me. The algorithm isn't producing the same result if executed multiple times (without reset in between):
>
> ```
> (0,2)           (2,2)
> (20,200)        (200,200) [...]

Hi @Roxy-3D,

The UBL 3 point does still not work properly but I spotted something here.

It seems that it adds the new reading of a point to last compensated mesh instead of the saved mesh. Lets take (2,2) we have a starting height of 0.160; than it sets at 0.323 which is Dh(i) = Z(i) - Z(i-1) = 0.323 - 0.160 = 0.163 which is quasi equal to 0.16; than 0.491 which by same Dh(i) formula would be D(i) = Z(i) - Z(i-1) = 0.491 - 0.323 = 0.168 therefore again cvasi equal to 0.16; 

And it seems there is also a difference between Dh(i) and Dh(i-1) which might give the final answer; D1 = Dh(1) - Dh(0) = 0.163 - 0.16 = 0.003; also D2 = Dh(2) - Dh(1) = 0.168 - 0.163 = 0.005; So D2 almost equal to D1 which seems to be the probe error.

Could You please help in debugging this issue?