MarlinFirmware/Marlin #8577: Compile error with last Bugfix 2.0.x

Bergerac56, 2017-11-27:
The last version of bugfix does not compile on my side. I get:

Compiling .pioenvs\LPC1768\src\src\gcode\parser.o
Marlin\src\gcode\motion\M290.cpp: In static member function 'static void GcodeSuite::M290()':
Marlin\src\gcode\motion\M290.cpp:63:37: error: no matching function for call to 'GCodeParser::boolval(char, bool)'
if (parser.boolval('P', true)) mod_zprobe_zoffset(offs);
^
In file included from Marlin\src\gcode\motion\../gcode.h:245:0,
from Marlin\src\gcode\motion\M290.cpp:27:
Marlin\src\gcode\motion\../parser.h:302:35: note: candidate: static bool GCodeParser::boolval(char)
FORCE_INLINE static bool        boolval(const char c)                          { return seenval(c) ? value_bool()         : seen(c); }
^~~~~~~
Marlin\src\gcode\motion\../parser.h:302:35: note:   candidate expects 1 argument, 2 provided
*** [.pioenvs\LPC1768\src\src\gcode\motion\M290.o] Error 1
 [ERROR] Took 112.12 seconds

coolio986, 2017-11-27:
Hi, I just compiled branch version 94cb412 of Bugfix 2.0.x
I did not have any issue. It compiled ok.

Can you post your Configuration.h and Configuration_adv.h?

Bergerac56, 2017-11-27:
@coolio986 - Here my config files
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1506832/Configuration.zip)

coolio986, 2017-11-27:
The error is in M290.cpp

Line 63,31:
`if (parser.boolval('P', true)) mod_zprobe_zoffset(offs);`

Should be 

`if (parser.boolval('P')) mod_zprobe_zoffset(offs);`

Since parser.boolval doesn't have a override for `boolval(const char c, const bool b)`

Also line 55,48:
`if (a == Z_AXIS && parser.boolval('P', true)) mod_zprobe_zoffset(offs);`

Should be

`if (a == Z_AXIS && parser.boolval('P')) mod_zprobe_zoffset(offs);`

MoellerDi, 2017-11-27:
got a similar error with bugfix-1.1.x:

Arduino: 1.8.1 (Mac OS X), Board: "Sanguino, ATmega1284 or ATmega1284P (16 MHz)"

sketch/Marlin_main.cpp: In function 'void gcode_M290()':
Marlin_main.cpp:9350: error: no matching function for call to 'GCodeParser::boolval(char, bool)'
           if (parser.boolval('P', true)) mod_zprobe_zoffset(offs);
                                       ^
sketch/Marlin_main.cpp:9350:39: note: candidate is:
In file included from sketch/Marlin_main.cpp:260:0:
sketch/gcode.h:313:32: note: static bool GCodeParser::boolval(char)
   FORCE_INLINE static bool     boolval(const char c)                          { return seenval(c) ? value_bool()      : seen(c); }
                                ^
sketch/gcode.h:313:32: note:   candidate expects 1 argument, 2 provided
exit status 1
no matching function for call to 'GCodeParser::boolval(char, bool)'

fiveangle, 2017-11-27:
Same, as of latest bf2:

```
avr-g++ -o .pioenvs\teensy20\src\src\gcode\probe\G30.o -c -fno-exceptions -std=gnu++0x -felide-constructors -fno-threadsafe-statics -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DF_CPU=16000000L -DLAYOUT_US_ENGLISH -DPLATFORMIO=30401 -DTEENSY2PP -DUSB_SERIAL -DARDUINO=10610 -DTEENSYDUINO=136 -I.pioenvs\teensy20\src -IC:\users\djohnson\.platformio\packages\framework-arduinoteensy\cores\teensy -IC:\users\djohnson\.platformio\packages\framework-arduinoteensy\cores -IMarlin -I.piolibdeps\U8glib-HAL_ID1932\src -IC:\users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\Wire -IC:\users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\Wire\utility -I.piolibdeps\LiquidCrystal_I2C_ID576 -IC:\users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\SPI -I.piolibdeps\TMC2130Stepper\src "-I.piolibdeps\Adafruit NeoPixel" -I.piolibdeps\LiquidTWI2 -I.piolibdeps\TMC26XStepper -IC:\users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\LiquidCrystal -IC:\users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\Servo Marlin\src\gcode\probe\G30.cpp
Marlin\src\gcode\motion\M290.cpp: In static member function 'static void GcodeSuite::M290()':
Marlin\src\gcode\motion\M290.cpp:63:37: error: no matching function for call to 'GCodeParser::boolval(char, int)'
if (parser.boolval('P', true)) mod_zprobe_zoffset(offs);
^
Marlin\src\gcode\motion\M290.cpp:63:37: note: candidate is:
In file included from Marlin\src\gcode\motion\../gcode.h:245:0,
from Marlin\src\gcode\motion\M290.cpp:27:
Marlin\src\gcode\motion\../parser.h:302:35: note: static bool GCodeParser::boolval(char)
FORCE_INLINE static bool        boolval(const char c)                          { return seenval(c) ? value_bool()         : seen(c); }
^
Marlin\src\gcode\motion\../parser.h:302:35: note:   candidate expects 1 argument, 2 provided
*** [.pioenvs\teensy20\src\src\gcode\motion\M290.o] Error 1
```

Haven't had time to dig into it yet.

-=dave

coolio986, 2017-11-27:
It appears that in gcode.h in #8017 `boolval `was revised using inline if to return the default value from seen rather than injecting a bool into the function with a default value `dval=false`.

Some areas have not been fixed with the updated structure. Please fix 1.1.x (Marlin_main.cpp)

```
/**
   * M290: Babystepping
   */
  inline void gcode_M290() {
    #if ENABLED(BABYSTEP_XY)
      for (uint8_t a = X_AXIS; a <= Z_AXIS; a++)
        if (parser.seenval(axis_codes[a]) || (a == Z_AXIS && parser.seenval('S'))) {
          const float offs = constrain(parser.value_axis_units((AxisEnum)a), -2, 2);
          thermalManager.babystep_axis((AxisEnum)a, offs * planner.axis_steps_per_mm[a]);
          #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
            if (a == Z_AXIS && parser.boolval('P', true)) mod_zprobe_zoffset(offs);
          #endif
        }
    #else
      if (parser.seenval('Z') || parser.seenval('S')) {
        const float offs = constrain(parser.value_axis_units(Z_AXIS), -2, 2);
        thermalManager.babystep_axis(Z_AXIS, offs * planner.axis_steps_per_mm[Z_AXIS]);
        #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
          if (parser.boolval('P', true)) mod_zprobe_zoffset(offs);
        #endif
      }
    #endif
  }
```

To:

```
/**
   * M290: Babystepping
   */
  inline void gcode_M290() {
    #if ENABLED(BABYSTEP_XY)
      for (uint8_t a = X_AXIS; a <= Z_AXIS; a++)
        if (parser.seenval(axis_codes[a]) || (a == Z_AXIS && parser.seenval('S'))) {
          const float offs = constrain(parser.value_axis_units((AxisEnum)a), -2, 2);
          thermalManager.babystep_axis((AxisEnum)a, offs * planner.axis_steps_per_mm[a]);
          #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
            if (a == Z_AXIS && parser.boolval('P')) mod_zprobe_zoffset(offs);
          #endif
        }
    #else
      if (parser.seenval('Z') || parser.seenval('S')) {
        const float offs = constrain(parser.value_axis_units(Z_AXIS), -2, 2);
        thermalManager.babystep_axis(Z_AXIS, offs * planner.axis_steps_per_mm[Z_AXIS]);
        #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
          if (parser.boolval('P')) mod_zprobe_zoffset(offs);
        #endif
      }
    #endif
  }
```

fiveangle, 2017-11-27:
They were changed, but Scott introduced this regression by re-injecting the old method into M290 routines with https://github.com/MarlinFirmware/Marlin/commit/2060ba3556df97621eae8051845f1224f764cb82 (bf11) and https://github.com/MarlinFirmware/Marlin/commit/be00e421a76cc82a3e68cf801e211b0f450ea393 (bf20).

I'm making PRs to fix now.

-=dave

closed 2017-11-28 by thinkyhead

Bergerac56, 2017-11-29:
Thanks the team. It compiles fine now

commit 4793c34c (2017-11-28): Fix M290 boolval bug

Fix #8577
commit 46776de9 (2017-11-28): Fix M290 boolval bug

Fix #8577