MarlinFirmware/Marlin #8578: K-Factor Calibration Pattern Generator does not extrude

ctday, 2017-11-27:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

ctday, 2017-11-27:
The K-Factor Calibration Pattern generated by http://marlinfw.org/tools/lin_advance/k-factor.html does not actually print. The problem is that the generated gcode issues M83 before G90. The latter overrides the former, putting all axes into Absolute mode. The correct sequence is G90 then M83, which leaves the extruder axis in Relative mode and the other axes in Absolute mode. This order prints properly.

Sineos, 2017-11-27:
Thanks for spotting: https://github.com/MarlinFirmware/MarlinDocumentation/pull/109

Had this fixed in my dev branch already tho it should only affect the nozzle prime. The pattern itself is printed under G91.

Sineos, 2017-11-27:
If you want to test: https://github.com/Sineos/MarlinDocumentation/tree/rotate_pattern/_tools/lin_advance

Just copy the .js and .html in a folder and run the html. Feedback would be appreciated.

bjarchi, 2017-11-27:
> The K-Factor Calibration Pattern generated by http://marlinfw.org/tools/lin_advance/k-factor.html does not actually print. The problem is that the generated gcode issues M83 before G90. The latter [...]

Additionally this must be printer specific, as I've used the version on the main documentation site (and am testing the development versions) with no extrusion issues.

thinkyhead, 2017-11-29:
Merged the patch.

closed 2017-11-29 by thinkyhead