MarlinFirmware/Marlin #8579: 2.x fan commands don't work until a movement command runs

aplocher, 2017-11-27:
------------------------------------------------------------
### Bug Report
- Description: When I set the fan speed (either through the Octoprint terminal or using the Octoprint UI) it responds "OK" but doesn't actually perform the command until after I move the X-axis (for example) slightly.

  Also upon powering on the printer, the fan is (annoyingly!) spinning at full speed until I go move one of the axis.

  I'm on an ANet A8 using the latest code in the 2.x branch. Built with Arduino IDE

- Expected behaviour: Should change the fan speed immediately when I run the M106 command
- Actual behaviour: Continues with the fan speed it was currently running at, until I move one of the axis
- Steps to reproduce:

Send: M106 S255
Recv: ok
(no change occurs)

Send: G91
Recv: ok
Send: G1 X10 F6000
Recv: ok
Send: G90
Recv: ok
(now the fan speed changes to 255)

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1506980/Marlin.zip)

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

AnHardt, 2017-11-27:
Sorry for not having locked into your config.
Only guessing from the symptoms I assume you confused 'part cooling fan', 'hotend fan' and 'board fan' and configured all of them to one pin.

aplocher, 2017-11-27:
Hmm I don't think I changed any wires when I jumped from 1.1.6 to 2.x (I was messing with some stuff yesterday but I think that was way earlier and 1.1.6 was fine).

Even if that was the case, it wouldn't explain why it consistently does work after the G1 command.  If I set the fan speed, and then G1 to move the hotend, it execute my fan command at that time, but not immediately when I call the M106 command like it used to.

Thanks

AnHardt, 2017-11-27:
```cpp
/**
 * Controller Fan
 * To cool down the stepper drivers and MOSFETs.
 *
 * The fan will turn on automatically whenever any stepper is enabled
 * and turn off after a set period after all steppers are turned off.
 */
//#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  //#define CONTROLLER_FAN_PIN FAN1_PIN  // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SECS 60          // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255        // 255 == full speed
#endif
```
Doesn't this lock like your symptom?

aplocher, 2017-11-27:
Nope, don't think so.  It powers on and is running at max fan speed until I move an axis.  I've had it on for well over an hour or two before finally getting annoyed  (and deafened) by the fan.  Simply moving X (or any axis) 10 mm turns the fan off.  

I believe when it powers on it typically turns the fan on for a quick sec and then turns off.  My guess is, it's running the command to turn the fan off but it doesn't get executed until after an axis moves (which is the same behavior I'm seeing from the GCode terminal in Octoprint).

Every command I use with M106 works as long as I move to another XYZ coord.  But it doesn't until I do that.

aplocher, 2017-11-27:
I see what you're saying with the fan pin.  I diff'ed my config against my 1.1.6 one and a previous 2.x one (from about a month ago) that I used, and nothing fan-related has changed.  Still don't think that behavior would be quite right even if that was misconfigured...

ManuelW77, 2017-11-28:
Hey there, i have the same with the latest 1.1.6 Bugfix.
I can't turn my Part Fan on with M106 or off with M107. 
But the Fan will go or off on when i home the Nozzle.

AnHardt, 2017-11-28:
Arrg! This could be by design! I remember a discussion about synchronizing fan activity with the moves. During a print the part cooling fan is mostly used for bridges and at the end when the layers do not last very long. If the fan speed would be set immediately, then the bridge-move is cued into the planer buffer, then the fan is shot down again, the fan would be off again before the related move is executed.

thinkyhead, 2017-11-28:
> Hey there, i have the same with the latest 1.1.6 Bugfix.

@ManuelW77 With `bugfix-1.1.x` _and_ with `1.1.6`?

aplocher, 2017-11-28:
@AnHardt I think what you said can kinda make sense, but there should be some behavioral modifications.  e.g. don't run fan at full speed at power-on.  

I'm not sure if there's any "Marlin-flavor" of g-code (like custom commands for marlin) but if there is, I would suggest adding a custom "Execute Now" type thing that allows fan speeds to be modified immediately.  (e.g. M106 255; Z123; to change to 255 NOW).  But it seems like a feature like that would cause a lot of headaches from the slicer developers since other firmwares probably don't behave that way.

thinkyhead, 2017-11-28:
Over-zealous optimization may have gone awry. See if the newest `bugfix-1.1.x` and/or `bugfix-2.0.x` exhibit the problem.

AnHardt, 2017-11-29:
Sorry. My recognition was right, but fals alarm. When no blocks are planned the fan speed update is done directly.

```cpp
void Planner::check_axes_activity() {
  unsigned char axis_active[NUM_AXIS] = { 0 },
                tail_fan_speed[FAN_COUNT];

  if (blocks_queued()) {

    #if FAN_COUNT > 0
      for (uint8_t i = 0; i < FAN_COUNT; i++)
        tail_fan_speed[i] = block_buffer[block_buffer_tail].fan_speed[i];
    #endif

...

  }
  else {
    #if FAN_COUNT > 0
      for (uint8_t i = 0; i < FAN_COUNT; i++) 
        tail_fan_speed[i] = fanSpeeds[i];
    #endif

...

  }

```

Sorry again. I can't reproduce your problem.

closed 2017-11-29 by thinkyhead

aplocher, 2017-11-29:
Thanks guys, I'll verify this tonight.  Will let you know if any problems arise.

commit dfcef11b (2017-11-28): Fix #8579

Fan speed code fell into a curly brace
commit 16baa486 (2017-11-28): Fix #8579

Fan speed code fell into a curly brace