MarlinFirmware/Marlin #8585: Marlin  does not compile to STM32 using default example config

compihu, 2017-11-28:
### Compile Error
When I compile for STM32 with default example configuration the compilation fails:

Marlin\src\HAL\HAL_STM32F1\HAL_spi_Stm32f1.cpp:49:8: error: 'SPISettings' does not name a type
[Marlin2_stm32f1_buildlog.txt](https://github.com/MarlinFirmware/Marlin/files/1509898/Marlin2_stm32f1_buildlog.txt)

coolio986, 2017-11-28:
It appears the issue is a failure to call  "\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src\SPI.h"

Even doing a `#include <SPI.h>` In `HAL_spi_Stm32f1.cpp` has no effect.

c_cpp_properties.json has the includePath set correctly, but yet the standard location cannot be found.

thinkyhead, 2017-11-28:
Possibly some other file named `SPI.h` is obscuring the arduino core version…

coolio986, 2017-11-29:
@thinkyhead 
I checked the source for the teensy35 since that uses a very similar structure to stm32f1. The teensy35 compiles without issue, while also referencing the same libraries

Differences between the two consist of the stm32f1 having a build_flags for python in platformio.ini  and teensy35 using libraries included in its HAL folder.

coolio986, 2017-11-30:
I have continued to look into this issue, but I have yet to find a solution. Since I am newer to platformio, I don't quite know the ins and outs of it like arduino ide. It may be wise to have some of the devs look into this one.

compihu, 2017-12-10:
Will STM32 be supported anytime soon?

compihu, 2017-12-10:
Where should I find the declaration of SPISettings?

compi@vecsespc:~/work/ext/Marlin/Marlin$ grep -r SPISettings .
./src/HAL/HAL_STM32F1/HAL_spi_Stm32f1.cpp:static SPISettings spiConfig;
./src/HAL/HAL_STM32F1/HAL_spi_Stm32f1.cpp:  spiConfig = SPISettings(clock, MSBFIRST, SPI_MODE0);
./src/HAL/HAL_TEENSY35_36/HAL_spi_Teensy.cpp:static SPISettings spiConfig;
./src/HAL/HAL_TEENSY35_36/HAL_spi_Teensy.cpp:  spiConfig = SPISettings(clock, MSBFIRST, SPI_MODE0);

victorpv, 2017-12-12:
It's supported, I'm printing fine in my custom board.
The issue is because the SPI include was changed in this commit:
https://github.com/MarlinFirmware/Marlin/commit/60af705e6d68bc972fbdcd232e78a775efe53128#diff-8eb2d10518202d127cc150c0ef73d546

You need to change this line:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-`2.0.x/Marlin/src/HAL/HAL_STM32F1/HAL_spi_Stm32f1.cpp#L40`

Should read 
`#include "SPI.h"`

I'll try to correct that in my next PR, someway I missed it in my latest PR, I also had the same error that when I rebased a few days ago and got the change.

victorpv, 2017-12-12:
@thinkyhead feel free to reference me in any STMF1 issue if I don't happen to reply, some times I may go quite a few days without having enough time check what's going on in general, but if you reference me I should get a flag when I login to github.

victorpv, 2017-12-12:
@compihu just so you know, I use Sloeber, not platformio or Arduino IDE, but you find any other error let me know and I'll try to test. 
So far I can confirm ADC, steppers, serial, serialUSB, sdcard on SPI1, LCD, i2c1 all work fine.
I printed a 5 hours job the other day from sdcard without a hicup.

victorpv, 2017-12-12:
I just sent a PR to fix this:
https://github.com/MarlinFirmware/Marlin/pull/8762

compihu, 2017-12-12:
@victorpv I changed SPI.h inclusion to the same as appears in your pull request on top of #c555a214d2cb5704bb323f99b15f0132300336b4
It seems that it did not fix the SPI issue and Timer code got broken meanwhile:

`> Executing task: platformio run --environment STM32F1 <

[Tue Dec 12 18:25:21 2017] Processing STM32F1 (build_flags: !python Marlin/src/HAL/HAL_STM32F1/stm32f1_flag_script.py; lib_deps: U8glib-HAL, LiquidCrystal_I2C@1.1.2, TMC2130Stepper, Adafruit NeoPixel, https://github.com/lincomatic/LiquidTWI2.git, https://github.com/trinamic/TMC26XStepper.git, https://github.com/ameyer/Arduino-L6470.git; platform: ststm32; board: genericSTM32F103RE; framework: arduino; src_filter: +<src/*> -<src/config>)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting Marlin.ino
Collected 36 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <U8glib-HAL> v0.4
|-- <LiquidCrystal_I2C>
|   |-- <Wire>
|-- <TMC2130Stepper> v2.0.0
|   |-- <SPI> v1.0
|-- <Adafruit NeoPixel> v1.1.2
|-- <LiquidTWI2>
|   |-- <Wire>
|-- <TMC26XStepper>
|   |-- <SPI> v1.0
|-- <Arduino-L6470>
|-- <SPI> v1.0
|-- <LiquidCrystal>
|-- <EEPROM>
|-- <Wire>
|-- <Servo(STM32F1)> v1.1.2
|-- <STM32ADC> v1.0
Compiling .pioenvs/STM32F1/src/src/HAL/HAL_LPC1768/LPC1768_Servo.o
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfigPre.h:29:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:26,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../../Configuration.h:289:2: warning: #warning temp_sensor set to 998/999 (fake) [-Wcpp]
#warning temp_sensor set to 998/999 (fake)
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfigPre.h:29:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:26,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../../Configuration.h:447:2: warning: #warning "disabled temperature protection" [-Wcpp]
#warning "disabled temperature protection"
^
In file included from /home/compi/.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/boards.h:39:0,
from /home/compi/.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/wirish.h:54,
from /home/compi/.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/Arduino.h:30,
from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h:43,
from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL.h:50,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
/home/compi/.platformio/packages/framework-arduinoststm32/STM32F1/variants/generic_stm32f103r/board/board.h:41:0: warning: "CYCLES_PER_MICROSECOND" redefined [enabled by default]
#define CYCLES_PER_MICROSECOND  72
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfigPre.h:27:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:26,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../core/macros.h:47:0: note: this is the location of the previous definition
#define CYCLES_PER_MICROSECOND (F_CPU / 1000000L) // 16 or 20 on AVR
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h:61:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL.h:50,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:49:25: warning: extra tokens at end of #ifdef directive [enabled by default]
#ifdef MCU_STM32F103CB  || defined(MCU_STM32F103C8)
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h:61:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL.h:50,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h: In function 'void HAL_timer_set_count(uint8_t, hal_timer_t)':
[... 761 lines cut ...]
from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_STM32F1/persistent_store_impl.cpp:30:
Marlin/src/HAL/HAL_STM32F1/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h: At global scope:
Marlin/src/HAL/HAL_STM32F1/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h:157:12: warning: 'int freeMemory()' defined but not used [-Wunused-function]
static int freeMemory() {
^
*** [.pioenvs/STM32F1/src/src/HAL/HAL_STM32F1/persistent_store_impl.o] Error 1
===================================================================================================================================================================================================== [ERROR] Took 1.68 seconds =====================================================================================================================================================================================================

[... cut ...]
===================================================================================================================================================================================================== [ERROR] Took 1.68 seconds =====================================================================================================================================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.`

victorpv, 2017-12-12:
not sure if you need to do it in platformio, but in Sloeber I have the STM32F1/libraries/SPI library, from the stm32duino core files, added to the library list in the project.
Perhaps with that is how it finds SPI.h. Try that, otherwise you could also try <SPI.h> instead of "SPI.h"

victorpv, 2017-12-12:
I need to check in the timer error, I think something went wrong with my commits and something is missing, since me local copy compiles and runs fine.

victorpv, 2017-12-12:
@compihu I see one of the timer files is not updated. 
I'll have to commit that file back to my fork, then send a PR.
Wait for further testing until I can review it this evening. I think someway I mixed files when rebasing, since I got many errors and had to resolve loads of conflicts. At that point I may have left that file out.

p3p, 2017-12-12:
@victorpv I know it's a little annoying having to support a build chain you don't use yourself, I use Eclipse myself for development, but we are trying to standardise on platformio for all platforms (not the painful Atom IDE plugin, you can ignore that as long as the command line build works), you can also add a simple default build test for your platform to Travis just to make sure there aren't any silent build issues introduced later.

victorpv, 2017-12-12:
Thanks p3p, I'll try to learn how to use Travis.
I used platformio with the Re-Arm, but I find so much easier Eclipse that I keep using it for the STM. 
But I think the errors @compihu is having are due to some mess up I did while rebasing to the latest 2.0, not due to plaformio or Sloeber.

victorpv, 2017-12-12:
@compihu replace the file in your disk with this one:
https://github.com/victorpv/Marlin/blob/bugfix-2.0.x_multihost/Marlin/src/HAL/HAL_STM32F1/HAL_timers_Stm32f1.h

That has the latest changes. Don't download that whole branch, it's a merge to add multiserial support from TCM but not tested yet. You can download if you want to test it, but I haven't tested anything other than compilation.

p3p, 2017-12-12:
My response wasn't directly related to this issue, I had just seen a few comments scroll past about build issues and you responding that you build with Sloeber so I was just making sure you knew it was recommended that all platforms do build with platformio, Adding a default config Travis test should be pretty painless (one day we will run the whole test suite against each platform but that's not possible yet), you can just modify the LPC1768 one (https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.travis.yml#L422)

victorpv, 2017-12-12:
I thought your comment was referring to both, this case and in general. I need to check how Travis works and hopefully get it set for autobuilding the STM HAL. Thanks for the pointer.

victorpv, 2017-12-12:
@compihu btw just tested the multihost branch I linked above and seems to work, using SerialUSB + usart 1
Usart1 connected to an ESP01 with ESP3d Firmware. There are things to iron out with the ESP communication.

victorpv, 2017-12-13:
@alexxy seems like you have more experience with platformio. Can you check if updating the file metioned above from the multi-serial branch compiles in platformio?
And would you be able to help with the Travis integration?
One last question, what F1 board do you have?

alexxy, 2017-12-13:
@victorpv latest changes doesnt compiles for me with PIO
Where StepperTimer and TempTimer were defined?
PS I have stm32f103RET board...

```
Compiling .pioenvs/STM32F1/src/src/HAL/HAL_LPC1768/include/digipot_mcp4451_I2C_routines.o
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h:61:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL.h:50,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h: In function 'void HAL_timer_set_count(uint8_t, hal_timer_t)':                                                                                                                                 
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:120:5: error: 'StepperTimer' was not declared in this scope                                                                                                                                    
StepperTimer.pause();
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:126:5: error: 'TempTimer' was not declared in this scope
TempTimer.pause();
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h: In function 'hal_timer_t HAL_timer_get_count(uint8_t)':
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:140:12: error: 'StepperTimer' was not declared in this scope                                                                                                                                   
temp = StepperTimer.getCompare(STEP_TIMER_CHAN);
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:143:12: error: 'TempTimer' was not declared in this scope
temp = TempTimer.getCompare(TEMP_TIMER_CHAN);
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h: In function 'void HAL_timer_set_current_count(uint8_t, hal_timer_t)':                                                                                                                         
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:154:26: error: 'StepperTimer' was not declared in this scope                                                                                                                                   
case STEP_TIMER_NUM: StepperTimer.setCount(count); break;
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:155:26: error: 'TempTimer' was not declared in this scope
case TEMP_TIMER_NUM: TempTimer.setCount(count); break;
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h: In function 'hal_timer_t HAL_timer_get_current_count(uint8_t)':                                                                                                                               
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:163:12: error: 'StepperTimer' was not declared in this scope                                                                                                                                   
temp = StepperTimer.getCount();
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:166:12: error: 'TempTimer' was not declared in this scope
temp = TempTimer.getCount();
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h: In function 'void HAL_timer_isr_prologue(uint8_t)':
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:180:5: error: 'StepperTimer' was not declared in this scope                                                                                                                                    
StepperTimer.pause();
^
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_timers_Stm32f1.h:186:5: error: 'TempTimer' was not declared in this scope
TempTimer.pause();
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL.h:50:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h: At global scope:
Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_STM32F1/HAL_Stm32f1.h:157:12: warning: 'int freeMemory()' defined but not used [-Wunused-function]                                                                                                                          
static int freeMemory() {
^
Compiling .pioenvs/STM32F1/src/src/HAL/HAL_LPC1768/main.o
*** [.pioenvs/STM32F1/src/src/HAL/HAL_LPC1768/LPC1768_Servo.o] Error 1
```

victorpv, 2017-12-14:
@alexxy you would have to replace this file in your folder with this copy:
https://github.com/victorpv/Marlin/blob/bugfix-2.0.x_multihost/Marlin/src/HAL/HAL_STM32F1/HAL_timers_Stm32f1.h

I tried Travis CI in my branch, but platformio apparently doesn't have board definition for MCUs with more than 64 pins (RxT6).

compihu, 2017-12-21:
Here is the build log after #2ec4113cb26f1a3b70a9f513f7344e57c7ef1e45
[MarlinBuildSTM32_2ec4113cb26f1a3b70a9f513f7344e57c7ef1e45.log](https://github.com/MarlinFirmware/Marlin/files/1578191/MarlinBuildSTM32_2ec4113cb26f1a3b70a9f513f7344e57c7ef1e45.log)

Bob-the-Kuhn, 2018-01-03:
@victorpv - there was a mention of a need for PINS_DEBUG support for the STM32.

Here's an untested & uncompiled attempt.

I got the STM32 specific items by poking around your HAL_STM32F446 directory.

It also includes examples from the LPC1768 & the Due as I expect I've left a lot to the imagination.

[HAL_pinsDebug_STM32.zip](https://github.com/MarlinFirmware/Marlin/files/1599560/HAL_pinsDebug_STM32.zip)

victorpv, 2018-01-04:
Thanks Bob, I have been trying to get Travis to pass and had not looked at that yet. I will check your version  and try to get it working.

compihu, 2018-02-11:
@thinkyhead @victorpv 

Guys did someone actually test this before closing the issue?!

[Marlin_bugfix-2.0.x_build_log_stm32F1_89165326b4bc627ee080cd2acd073c0097d80ddd.txt](https://github.com/MarlinFirmware/Marlin/files/1714640/Marlin_bugfix-2.0.x_build_log_stm32F1_89165326b4bc627ee080cd2acd073c0097d80ddd.txt)

thinkyhead, 2018-02-14:
> Guys did someone actually test this before closing the issue?!

Not at all. Just closing issues with no activity because they tend to pile up. Thanks for the update.

So let's see…
```
Marlin/src/HAL/HAL_STM32F1/persistent_store_impl.cpp:86:5: error: stray '`' in program
if (writing) value[i] = c`;
```
Removing the stray ` seems to solve that specific issue.

Now if you compile `bugfix-2.0.x` what other errors do you see?

I'd do it myself, but have a couple hundred more issues I'm anxious to get through. I look forward to your followup report.

compihu, 2018-02-14:
@thinkyhead @victorpv 
Unfortunately the original SPI.h inclusion related problem still persist. SPI.h does not declare architecture dependent settings and configs itselp but as I can guess it should include an architecture dependent SPI header file (as it does for LPC1768) where these declarations should present. Unfortunately I cannot find such a header for STM32.
[Marlin_bugfix-2.0.x_build_log_stm32F1_93c0198b2dec4497c42d20cd01a6dac909416da7.txt](https://github.com/MarlinFirmware/Marlin/files/1723937/Marlin_bugfix-2.0.x_build_log_stm32F1_93c0198b2dec4497c42d20cd01a6dac909416da7.txt)

thinkyhead, 2018-02-14:
To resolve the `CYCLES_PER_MICROSECOND` conflict, try this in `macros.h`.
```cpp
#if !defined(__STM32F1__) && !defined(TARGET_STM32F1)
  #define CYCLES_PER_MICROSECOND (F_CPU / 1000000L) // 16 or 20 on AVR
#endif
```

thinkyhead, 2018-02-14:
@MarlinFirmware/32bit-maintainers

victorpv, 2018-02-15:
Compihu I'm not understand what header you are looking for.
You need to include the SPI.h from the HAL folder in marlin, and the SPI.h from the SPI library in the stm32duino.
I use sloeber (Eclipse arduino plugin) and don't have any problem. I have tried to compile with platformio when I have had time, but so far it has not worked.

compihu, 2018-02-16:
@victorpv I thought that PlatformIO is the new "official" IDE for Marlin development as platformio.ini added to Marlin back in mid-2016. If Sloeber is the new way to go, would you please show me where can I find project files for it in the tree? Another question: Is STM32 port of marlin based on Arduino framework? If not, how Sloeber (which is an Arduino IDE) comes into the picture?

BTW:
`compi@lothiriel:~/work/ext/Marlin/Marlin/src$ grep -r SPISettings .
./HAL/HAL_LPC1768/LPC_SPI.h:class SPISettings {
./HAL/HAL_LPC1768/LPC_SPI.h:    SPISettings(int a, int b, int c) {};
./HAL/HAL_LPC1768/LPC_SPI.h:    void beginTransaction(SPISettings foo) {};
./HAL/HAL_STM32F1/HAL_spi_Stm32f1.cpp:static SPISettings spiConfig;
./HAL/HAL_STM32F1/HAL_spi_Stm32f1.cpp:  spiConfig = SPISettings(clock, MSBFIRST, SPI_MODE0);
./HAL/HAL_STM32F1/HAL_spi_Stm32f1.cpp:  spiConfig = SPISettings(spiClock, bitOrder, dataMode);
./HAL/HAL_STM32F7/HAL_spi_STM32F7.cpp:static SPISettings spiConfig;
./HAL/HAL_STM32F7/HAL_spi_STM32F7.cpp:  spiConfig = SPISettings(clock, MSBFIRST, SPI_MODE0);
./HAL/HAL_STM32F7/TMC2660.cpp:  STEPPER_SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE3));
./HAL/HAL_TEENSY35_36/HAL_spi_Teensy.cpp:static SPISettings spiConfig;
./HAL/HAL_TEENSY35_36/HAL_spi_Teensy.cpp:  spiConfig = SPISettings(clock, MSBFIRST, SPI_MODE0);
./HAL/HAL_TEENSY35_36/HAL_spi_Teensy.cpp:  spiConfig = SPISettings(spiClock, bitOrder, dataMode);
`

Would you please enlighten me where SPISettings is defined for STM32F1? I can find it for LPC1768 only.

victorpv, 2018-02-17:
@compihu sorry  if I haven't been clear *I* do not use platformio to compile, because so far I haven't been able to get to compile right, just me. I know other people have used the arduino IDE just fine too, and other may have used Plaformio and got it working, but may not have submitted whatever they have to do.
Platformio is the "official" IDE for Marlin, and the target is to get it to compile with it.
SPISettings should be within the SPI library in the stm32duino core (\STM32F1\libraries\SPI\src)

compihu, 2018-02-17:
@victorpv I guess the stm32duino SPI headers will be automatically included by Sloeber and not by Platformio. Last time I checked Sloeber did things for me I've never asked to do...

victorpv, 2018-02-17:
@p3p suggested to disable and option in platformio to disable it from trying to find the libraries itself, apparently was having a similar problem with the LPC HAL. I still need to have time to test it out. I guees after disabling that, we have to tell it where to find the library in some other config file.

alexxy, 2018-02-18:
class SPISettings defined in 

${HOME}/.platformio/packages/framework-arduinoststm32/STM32F1/libraries/SPI/src/SPI.h

In PlatformIO on linux, but it doesnt included by default. So it causes problems.

teemuatlut, 2018-02-18:
I think that's yet another case of the compiler not making a difference between
`#include "SPI.h"` and `#include <SPI.h>`.
When compiling for STM32, it'll try to use the SPI.h file from HAL folder.

compihu, 2018-02-19:
@teemuatlut It DOES make difference;) "" searches the directory when the compiled unit resides FIRST, then the inlude path list. <> omits the local search. In similar situations it is a good habbit to put the parent foder of the include folder to the include path and include the specific file by <folder/file.h> (like boost used to do it for example)

teemuatlut, 2018-02-19:
And yet `#include <SPI.h>` will find the file in the HAL folder and not the library.

Try it yourself. Add `#define SPI_FROM_LIB` to STM32F1/libraries/SPI/src/SPI.h and then `#define SPI_FROM_HAL` to HAL/SPI.h
Test it with this in HAL_spi_Stm32f1.cpp
```cpp
// --------------------------------------------------------------------------
// Includes
// --------------------------------------------------------------------------
#include <SPI.h>
#ifdef SPI_FROM_LIB
  #error #define SPI_FROM_LIB
#endif
#ifdef SPI_FROM_HAL
  #error #define SPI_FROM_HAL
#endif

#include "../HAL.h"
```

compihu, 2018-02-19:
Tis is nothing to do with "" and <> includes it's about the include path order.

teemuatlut, 2018-02-19:
Then perhaps you'd like to tell us why `#include <SPI.h>` finds the HAL/SPI.h file if it is supposed to omit the local folders?

p3p, 2018-02-19:
The STM32F1 platform seems to use a flag script file derived from the LPC1768 version, not sure why this is needed for an Arduino based platform, this has the line `"-IMarlin/src/HAL"` adding the HAL folder to the include search path.

alexxy, 2018-02-19:
Also there seems like header conflict. Since SPI.h exist here

Marlin/src/HAL/SPI.h

and also in 

${HOME}/.platformio/packages/framework-arduinoststm32/STM32F1/libraries/SPI/src/SPI.h

Which has needed defines

compihu, 2018-02-19:
I think this would be the way to go. But I do not know how many changes would it take to remove src/HAL from include paths, add src instead and prefix all HAL inclusions with HAL/

thinkyhead, 2018-02-19:
Try this:

`Marlin/src/HAL/HAL_STM32F1/HAL_spi_Stm32f1.cpp`:
```diff
- #include "SPI.h"
+ #include "../SPI.h"
```

thinkyhead, 2018-02-20:
Seems we've settled on the fact the the LPC build script is doing some clever "spoofing" to put the `src/HAL` and `src/HAL/HAL_LPC1768/include` folders at the front of the library include paths. This is apparently done to get `#include <Arduino.h>` to include `src/HAL/HAL_LPC1768/includes/Arduino.h` from any library or source file that uses that form.

As a side-effect, this causes `#include <SPI.h>` to pick up the one in `src/HAL` first. As long as the "spoof" method is being used to redirect includes we have no alternative but to rename the local copy of `SPI.h` to something else, such as `HAL_SPI.h` or `MarlinSPI.h`.

In the near future we can work out a better include structure that gets us what we want, or perhaps we can move all the files that we want to "spoof" into the `include` subfolder and drop the `-IHAL` argument from the build script.

That way if a file inside of `include` wants to intentionally spoof `SPI.h` it can do so, and if it needs to use the contents of the `SPI.h` file in the `HAL` folder, it need only contain one line: `#include "../../SPI.h"`.

thinkyhead, 2018-02-20:
The STM32F1 build script is putting the `Marlin/src/HAL` folder at the front of the library include paths. This is apparently done to get `#include <...>` to look in `Marlin/src/HAL/...`. Unfortunately, this affects every library or source file that uses that form, and causes `#include <SPI.h>` to find the one in `Marlin/src/HAL` first.

If the include path is truly necessary, then we can rename the local copy of `SPI.h` to something else, such as `HAL_SPI.h` or `MarlinSPI.h`. But before taking this step I'll see if I clean up Marlin's `#include` usage and drop the added HAL path from the build script.

youprint, 2018-02-20:
I succefully compiled MArlin on a STM32F103RE using Arduino IDE, but with a lot of pain. I had to mix old code from Victor_PV repo (old SPI file) with the new code on this repo. LCD, SD card is working but not motors, i think there is a pb with the HAL timers. I made a complete test platform : 
https://imageshack.com/a/img923/9199/G1Y5Sz.jpg
If we decide to stay on this HAL and the STM32duino Bootloader, it will be easier to compile it with Aduino IDE 1.8.5. Just try it, but be sure to flash the STM32 bootloader before.
I'm going to create a repo and a tutorial to flash a STM32F103RE, VE or ZE from scratch.
If we have all the same way to compile it will be easier to work.
I tried to compile with Atom but without any success.

thinkyhead, 2018-02-20:
Yeah, I'm trying with PlatformIO at the moment.

The compiler is getting a command-line parameter `-Wc++0x-compat` so it won't allow C++11 features, even though the `-std=gnu++11` is also included via the script. So I'm trying to see if the compat flag can be negated.

youprint, 2018-02-20:
I'm still building my test platform
https://imageshack.com/a/img924/5208/hKvaWe.jpg
I have the dream to drive my news L6470 with the STM32 chip, I know there is an existing library for AVR but i don't know if it is working on STM via Arduino IDE.

thinkyhead, 2018-02-20:
I've given up for tonight. I can't find any non-hackish way to get the "gccarmnoneeabi" compiler to like C++11. I'll need to get well-acquainted with building for all our targets and make sure they can all be built from my environment … not just for my own testing but also to be able to write up understandable build instructions for end-users. I don't want anyone to have to deal with the kinds of pitfalls we devs seem to enjoy so much.

thinkyhead, 2018-02-20:
That build platform is going to be awesome. I could do something similar with the bits and pieces I have lying around here. Hmm…

youprint, 2018-02-21:
I just tried with sloeber (eclipse Arduino plug in) and it’s working too with my modded code. If I use the code from the repo it doesn’t work, still a pb with spi. 
I’m going to work on a simple configuration for a 4 axis laser CNC for STM32 (blue pill) since I found I working HAL. 
I’m going to test the STMF7 HAL too...

thinkyhead, 2018-02-21:
> If I use the code from the repo it doesn’t work

Same include path issue? I will patch that later today.

youprint, 2018-02-22:
Already did it but some spi timer issues. I really think we need to focus compiling with Arduino with stm32duino libs since it is written to natively work with it. The other choice is to import these libs into platformio

compihu, 2018-02-22:
Does this mean that STM32 HAL is based on arduino framework? Please, please, please say no.... Pretty please with sugar on top...

youprint, 2018-02-22:
The STM32F1 Hal was written by Victor who is an active member of the STM32duino community. All his work is based on these special modded libs. But I don’t think it’s difficult to get it work with platformio but it need some work. The STMF7 hal use a special core created by Danieleff, it seems using standard STM libs, i’m just waiting for my dev board to try it.

thinkyhead, 2018-02-23:
I did a patch to deal with some bad paths, but it only improved the LPC1768 situation. I still need to mess around more to resolve the same issue with STM32F1. I have hesitated because if it's supposed to be possible to build for STM32F1 in Arduino IDE, the kinds of changes I would make could break that compatibility.

Word has it that carefully placed `platform.local.txt` file can provide additional build arguments for the compiler. If we can get those set up and working within the latest Arduino IDE then we have a universal solution that will allow both Arduino IDE and PlatformIO compatibility from the same code-base.

thinkyhead, 2018-02-23:
Getting close to being able to build for STM32F1, but there are now some new compile issues that need working out. https://github.com/thinkyhead/Marlin/tree/bf2_stm32f1_fixup

[16 more replies]

PR #8818 (2017-12-18): [1.1.x] Revert #8735 "Initial step correction"
This reverts commit ec028bf747ff7fe99c4fac29f0df62491db2528c from #8735.

This change seems to fix #8808 and fix #8585.

@Bob-the-Kuhn — We'll need to look at the unwanted side-effects of the original change and see what needs tweaking.

See also #8819
PR #8819 (2017-12-18): [2.0.x] Revert "Initial step correction"
This reverts commit 5cf6a062e315e7c0936f60ba6bf11d7f8fb1f20c from #8736

This change seems to fix #8808 and fix #8585.

@Bob-the-Kuhn — We'll need to look at the unwanted side-effects of the original change and see what needs tweaking.

See also #8818
closed 2018-04-26 by thinkyhead