MarlinFirmware/Marlin #8590: FR : "Antiblob" function for filamentwidth sensor

Erlandsson, 2017-11-28:
I had a tip here about a function for the widthsensor part in another thread. I was told to start a Feature Request about it.

I have issues with the sensor flipping out and reports extremely small diameter sometimes, 
so the extrusion runs up the roof for a second or so.
It seems it is not just me having these problems. I have tried all tips i got abous shielded wires, separated from power, regrounding , tried adjusting led color and brightness. 

A small add that checks the output and if it is abnormal or very much above the last read ones, 
then it will use the last good ones, until it gets normal reading again. Justabout now it happended again.
Running fine, then suddenly outputting 253% for a second or two.. Happens randomly.

if it reads: 2.6, 2.6, 2.6, 2.65, 2.68, 2.70, 2.68, **1.0**, **1.0**.. then that is obviously an error, then use the last ones read until it gets normal readings again..

I am talking to some guys who might help me get some rudimentary "blobsensor" function into the sensor firmware without too much delay on the output. As i cant get in touch with the original coder/s on thingiverse. I have no clue myself how to program so...

psavva, 2017-12-03:
I have checked the implementation, and there is a min and max reading which you can set.
MEASURED_UPPER_LIMIT and MEASURED_LOWER_LIMIT need to be configured as boundaries.

If the reading falls outside of these boundaries, then the nominal with will be used, which was the previous reading if I understood the implementation correctly.

Erlandsson, 2017-12-03:
Yes, but it does not work.. i have set it to max 3.1, but anyway the display indicate when it happens that the filament is 1mm, and the extruder goes nuts and blobs extremely.

psavva, 2017-12-03:
So this is Not a FR, but a bug report then. 
Suggest to change the title, and provide config files.  Which filamnet width sensor are you using?  I'll order/build the same to replicate...

Erlandsson, 2017-12-08:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1543071/Configuration_adv.zip)

I am using "tinywise" on thingiverse with Marlin 1.1.6 on a ramps board.
The min, max settings in firmware does not work. I sometimes get 1mm readings and extruder then goes haywire, not using the max setting.

empakoso, 2017-12-12:
I too am using the filament width sensor.  The min max is not being respected in the configuration_adv file.  Could someone take a look at how this MIN and MAX is used to abort the calculation.  I too am getting blobs.

@filipmu 
Is this the intention of the MIN MAX values in the Marlin fw?

filipmu, 2017-12-12:
MIN and MAX limits should be applied to incoming sensor reading. This way if the sensor fails for some reason (dirt, loose wire) and returns a reading of 0 or beyond the normal filament range marlin should use the MIN or MAX value for calculations. This was in the first version of the code but may have been lost in future updates. 

Filip

empakoso, 2017-12-12:
I would prefer that if the MIN or MAX values are exceeded that we use the nominal value entered in the configuration file to calculate.
(thank you for the quick reply ...as always :)

thinkyhead, 2017-12-13:
> I would prefer that if the MIN or MAX values are exceeded that we use the nominal value entered in the configuration file to calculate.

Try this in place of the existing function…
```cpp
  int Temperature::widthFil_to_size_ratio() {
    float temp = filament_width_meas;
    if (!WITHIN(temp, MEASURED_LOWER_LIMIT, MEASURED_UPPER_LIMIT)
      temp = filament_width_nominal;  // Assume sensor cut out
    return filament_width_nominal / temp * 100;
  }
```
And be sure to set the min/max pretty close to your nominal width. For example:
```cpp
  #define MEASURED_UPPER_LIMIT         1.85
  #define MEASURED_LOWER_LIMIT         1.65
```

thinkyhead, 2017-12-13:
Also please test #8772 and see if it makes the filament width sensor more useful. With those changes it should operate correctly in both normal linear extrusion mode (i.e., `M200 D0`) and volumetric extrusion mode (e.g., `M200 D1.75`), whereas previously it would have only operated correctly in volumetric mode.

empakoso, 2017-12-13:
I am on it.  I will be away from my printer until the weekend for feedback.  I've already uploaded all your changes.
Will this fix my issue #8767 ?

thinkyhead, 2017-12-13:
> Will this fix my issue #8767 ?

Yes, it will.

empakoso, 2017-12-20:
Ok, with this latest fix, I am now testing the MIN and MAX settings. I’ve set the following in the FW:
Max: 1.85
Nom: 1.75
Min: 1.65
So roughly +/-5% range.

Everything is working so far.  The real test is if this fix will eliminate the previous 200% blobs on the print.

As I am writing this, wouldn’t the MIN and MAX be better suited as a percentage?  For example, what if the filament width is changed? Then the fw would have to be uploade each time.

empakoso, 2017-12-20:
Seems to work on my first print. No blobs.  In fact I am proud to say it’s the best quality print of the calibration cube I have printed to date.  This sensor has definitely made an improvement.

thinkyhead, 2017-12-20:
I imagine that combining the sensor with `LIN_ADVANCE` would make our heads explode.

empakoso, 2017-12-20:
Oh geez! Dynamic is definitely the name of the game, then. Wow!

empakoso, 2017-12-21:
Confirmed! No more intermittent blobs on the print. The sensor is calculating based on the MIN and MAX and is doing a fine job now.

Erlandsson, 2017-12-22:
I don't get it, how do i get these fixes? copy thinkyhead's codesnippet or download a bugfix?

empakoso, 2017-12-22:
Download the bugfix 1.1.#
Then setup the config files from scratch( is how I prefer to do it).

Erlandsson, 2017-12-23:
thanks. ok now i know what do do this christmas.. Merry Christmas all..

empakoso, 2017-12-24:
The bugfix-1.1.# (current branch files)
And yes, it’s based on the previously released 1.1.7
Good luck and Merry Christmas

Erlandsson, 2017-12-24:
Little confused here..  #define FILWIDTH_ERROR_MARGIN ?
Is that that a replacement for MIN and MAX diameter?
because i can't find MIN MAX settings.

psavva, 2017-12-24:
@Erlandsson, yes FILWIDTH_ERROR_MARGIN , this replaces the min max settings.

psavva, 2017-12-24:
As per Thinkyhead, 
"This change allows a single "error margin" setting to validate filament width sensor even if the nominal filament width is altered at runtime."

empakoso, 2017-12-24:
https://github.com/MarlinFirmware/Marlin/issues/8767

This should explain it better:
https://github.com/MarlinFirmware/Marlin/pull/8845

Erlandsson, 2017-12-24:
Hmm this means that i no longer can run filament that is 2mm with default width at 2.8. I liked that you could have much smaller width than 2.8 like 2.0 but not too much over like 3.1. but i will test and se how it goes. Are there any info about HOW and WHAT it does? Like instructions for the function?

thinkyhead, 2017-12-25:
> Hmm this means that i no longer can run filament that is 2mm with default width at 2.8.

If you're using 2.0mm filament you would use `M404 W2` to set the nominal filament diameter to 2.0mm. That will ensure the correct percentage difference is displayed on the LCD. `FILWIDTH_ERROR_MARGIN` should be set to a value smaller than 0.5 to eliminate as many bad readings as possible.

closed 2017-12-25 by thinkyhead

Erlandsson, 2017-12-25:
Ok, ill try that. just one more question. I i use M404 W* Then i can have whatever setting in the slicer?
I can set nominal M404 W2.7 and still have it set to 3.0mm in the slicer? Slicersettings is not used at all if i understand correct?

filipmu, 2017-12-26:
You should set your slicer filament width to 2.7mm also since these need to be the same for the scaling to work. 

Filip

Erlandsson, 2017-12-29:
This is excellent. Works great now. Thanks all involved.