MarlinFirmware/Marlin #8591: UBL 3 point level auto correction returns "Error:Attempt to probe off the bed." on Kossel Anycubic

apprentice10, 2017-11-28:
### Bug Report (FIXED? MAYBE, TO SAVE TIME JUMP AT THE END FOR THE POTENTIAL FIX)

- Description: UBL 3 point level auto correction returns "Error:Attempt to probe off the bed." when testing second point on delta printer (specifically on kossel Anycubic)

- Expected behaviour: Test all 3 points to correct mesh

- Actual behavior: Test first point with success on second point will terminate the process and return "Error:Attempt to probe off the bed."

- Steps to reproduce: After previously going thru the step by step UBL, and homing, I try to run the UBL tool 3 point leveling.

22:40:43.668 : N23 M111 S32*97
22:40:43.672 : echo:DEBUG:LEVELING
22:40:58.871 : N24 G29 J*78
22:40:58.888 : >>> probe_pt(80.00, 0.00, no stow)
22:40:58.888 : current_position=(0.00, 0.00, 239.66) :
22:40:58.891 : >>> do_blocking_move_to(65.00, 9.00, 239.64)
22:40:58.898 : destination=(0.00, 0.00, 239.66) : set_destination_from_current
22:40:58.898 : destination=(0.00, 0.00, 239.64) : prepare_uninterpolated_move_to_destination
22:40:58.898 : current_position=(0.00, 0.00, 239.64) : zone border move
22:40:59.953 : current_position=(65.00, 9.00, 239.64) : xy move
22:41:00.049 : <<< do_blocking_move_to
22:41:00.049 : current_position=(65.00, 9.00, 239.64) : set_probe_deployed
22:41:00.049 : deploy: 1
22:41:00.052 : current_position=(65.00, 9.00, 239.64) : >>> run_z_probe
22:41:00.054 : current_position=(65.00, 9.00, 239.64) : >>> do_probe_move
22:41:00.257 : >>> do_blocking_move_to(65.00, 9.00, -10.00)
22:41:00.259 : destination=(65.00, 9.00, 239.64) : set_destination_from_current
22:41:00.263 : current_position=(65.00, 9.00, 239.64) : xy move
22:41:00.267 : destination=(65.00, 9.00, -10.00) : prepare_uninterpolated_move_to_destination
22:41:00.271 : current_position=(65.00, 9.00, -10.00) : z lower move
22:41:04.507 : <<< do_blocking_move_to
22:41:04.511 : current_position=(65.00, 9.00, -0.04) : sync_plan_position_kinematic
22:41:04.520 : current_position=(65.00, 9.00, -0.04) : <<< do_probe_move
22:41:04.520 : 1st Probe Z:-0.04
22:41:04.520 : >>> do_blocking_move_to(65.00, 9.00, 4.96)
22:41:04.520 : destination=(65.00, 9.00, -0.04) : set_destination_from_current
22:41:04.520 : destination=(65.00, 9.00, 4.96) : prepare_uninterpolated_move_to_destination
22:41:04.525 : current_position=(65.00, 9.00, 4.96) : z raise move
22:41:04.528 : current_position=(65.00, 9.00, 4.96) : xy move
22:41:04.753 : <<< do_blocking_move_to
22:41:04.754 : current_position=(65.00, 9.00, 4.96) : >>> do_probe_move
22:41:04.959 : >>> do_blocking_move_to(65.00, 9.00, -10.00)
22:41:04.959 : destination=(65.00, 9.00, 4.96) : set_destination_from_current
22:41:04.964 : current_position=(65.00, 9.00, 4.96) : xy move
22:41:04.964 : destination=(65.00, 9.00, -10.00) : prepare_uninterpolated_move_to_destination
22:41:04.970 : current_position=(65.00, 9.00, -10.00) : z lower move
22:41:05.178 : <<< do_blocking_move_to
22:41:05.180 : current_position=(65.00, 9.00, -0.03) : sync_plan_position_kinematic
22:41:05.186 : current_position=(65.00, 9.00, -0.03) : <<< do_probe_move
22:41:05.190 : current_position=(65.00, 9.00, -0.03) : <<< run_z_probe
22:41:05.190 : 2nd Probe Z:-0.03 Discrepancy:-0.01
22:41:05.190 : >>> do_blocking_move_to(65.00, 9.00, 4.97)
22:41:05.192 : destination=(65.00, 9.00, -0.03) : set_destination_from_current
22:41:05.197 : destination=(65.00, 9.00, 4.97) : prepare_uninterpolated_move_to_destination
22:41:05.199 : current_position=(65.00, 9.00, 4.97) : z raise move
22:41:05.204 : current_position=(65.00, 9.00, 4.97) : xy move
22:41:05.430 : <<< do_blocking_move_to
22:41:05.430 : <<< probe_pt
22:41:05.430 : >>> probe_pt(-40.00, 69.28, no stow)
22:41:05.430 : current_position=(65.00, 9.00, 4.97) :
22:41:05.430 : Error:Attempt to probe off the bed.

BACKGROUND (SINCE IS FIRST TIME)
Hello! I've been using Marlin on my Anycubic Kossel for about 2 weeks, I'm trying to run the bugfix 1.1.x for the UBL feature. I don't know much about C++ (I can read but hardly program it) I come from visual basic and maybe Python.

**## THE CODE**:
Basically I found out that the 3 point tool variable DELTA_PROBEABLE_RADIUS didn't had in count the X_PROBE_OFFSET_FROM_EXTRUDER and the Y_PROBE_OFFSET_FROM_EXTRUDER witch would cause it to believe it was out side of the bed when testing a point near the edge.
what I try to do is subtract the largest absolute value between X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER to the DELTA_PROBEABLE_RADIUS
 In the configuration.h file I change line 1012 from this:

`  #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10)`

...to this "monster" (please bear with me, I couldn't get the #if and #elif to run abs() )

    #if X_PROBE_OFFSET_FROM_EXTRUDER >= 0 && Y_PROBE_OFFSET_FROM_EXTRUDER >=0 //both positive
      #if X_PROBE_OFFSET_FROM_EXTRUDER >= Y_PROBE_OFFSET_FROM_EXTRUDER
        #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 5 - X_PROBE_OFFSET_FROM_EXTRUDER)
      #elif Y_PROBE_OFFSET_FROM_EXTRUDER > X_PROBE_OFFSET_FROM_EXTRUDER
        #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 5 - Y_PROBE_OFFSET_FROM_EXTRUDER)
      #endif
    #elif X_PROBE_OFFSET_FROM_EXTRUDER < 0 && Y_PROBE_OFFSET_FROM_EXTRUDER >=0 // X negative
      #if (X_PROBE_OFFSET_FROM_EXTRUDER * -1) >= Y_PROBE_OFFSET_FROM_EXTRUDER
        #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 5 - (X_PROBE_OFFSET_FROM_EXTRUDER * -1))
      #elif Y_PROBE_OFFSET_FROM_EXTRUDER > (X_PROBE_OFFSET_FROM_EXTRUDER * -1)
        #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 5 - Y_PROBE_OFFSET_FROM_EXTRUDER)
      #endif
    #elif X_PROBE_OFFSET_FROM_EXTRUDER >= 0 && Y_PROBE_OFFSET_FROM_EXTRUDER < 0 // Y negative
      #if X_PROBE_OFFSET_FROM_EXTRUDER >= (Y_PROBE_OFFSET_FROM_EXTRUDER * -1)
        #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 5 - X_PROBE_OFFSET_FROM_EXTRUDER)
      #elif (Y_PROBE_OFFSET_FROM_EXTRUDER * -1) > X_PROBE_OFFSET_FROM_EXTRUDER
        #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 5 - (Y_PROBE_OFFSET_FROM_EXTRUDER * -1))
      #endif
    #elif X_PROBE_OFFSET_FROM_EXTRUDER < 0 && Y_PROBE_OFFSET_FROM_EXTRUDER < 0 // Both negative
      #if (X_PROBE_OFFSET_FROM_EXTRUDER * -1) >= (Y_PROBE_OFFSET_FROM_EXTRUDER * -1)
        #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 5 - (X_PROBE_OFFSET_FROM_EXTRUDER * -1))
      #elif (Y_PROBE_OFFSET_FROM_EXTRUDER * -1) > X_PROBE_OFFSET_FROM_EXTRUDER
        #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 5 - (Y_PROBE_OFFSET_FROM_EXTRUDER * -1))
      #endif
    #endif

SHEET TO TEST VALUES
https://docs.google.com/spreadsheets/d/1RRCQQChCxQctMFHVPF6AfeUv-zvuVGpjmMZHkIlJUe0/edit?usp=sharing

I made this sheet so I could have track of the variables.

THANK YOU
It's really ammasing what you guys are doing!!! keep it up!

MY APOLOGIES
For those that read this "so long" threat and feel they lost their time reading it all.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1511644/Marlin.zip)

Roxy-3D, 2017-11-28:
You These defined:
// Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
#define DELTA_PRINTABLE_RADIUS 90.0 // mm <-KOSSEL PLUS DEFAULT(140.0)

#define X_PROBE_OFFSET_FROM_EXTRUDER -15     // <-KOSSEL PLUS DEFAULT(0) X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 9   // Y offset: -front +behind [the nozzle]

Using these values this code sets the probeable radius:

    #elif X_PROBE_OFFSET_FROM_EXTRUDER < 0 && Y_PROBE_OFFSET_FROM_EXTRUDER >=0 // X negative
      #if (X_PROBE_OFFSET_FROM_EXTRUDER * -1) >= Y_PROBE_OFFSET_FROM_EXTRUDER
        #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 5 - (X_PROBE_OFFSET_FROM_EXTRUDER * -1))
      #elif Y_PROBE_OFFSET_FROM_EXTRUDER > (X_PROBE_OFFSET_FROM_EXTRUDER * -1)
        #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 5 - Y_PROBE_OFFSET_FROM_EXTRUDER)
      #endif

which ends up doing this:

#define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 5 - (X_PROBE_OFFSET_FROM_EXTRUDER * -1))

which cranks out to 90-5-15 = 70 mm.

The 2nd probe point is defined as:

  #define UBL_PROBE_PT_2_X _PX(DELTA_PROBEABLE_RADIUS, 120) // =-40
  #define UBL_PROBE_PT_2_Y _PY(DELTA_PROBEABLE_RADIUS, 120) // =69.2

This is outside the probeable radius.  (40*40+69*69)^.5 = 79.7 mm.   The firmware is trying to protect you from damaging your machine.

apprentice10, 2017-11-28:
I'm sorry, just to clarify, the -40 in the comment was from before I change the code, the actual value is 70 * cos(RADIANS(9)) = -35 witch wont do much a difference in your statement (35 * 35 + 69 * 69)^.5=77.3.

As I understood (correct me if I'm wrong) the DELTA_PRINTABLE_RADIUS (in my case 90) is the actual radius on witch the nozzle can move without taking notice the probe, if my probe has an offset on X, Y axis shouldn't I have to take those values from the DELTA_PROBEABLE_RADIUS?
on the sheet I sent you you can test those values with and without taking the probe's offset

https://docs.google.com/spreadsheets/d/1RRCQQChCxQctMFHVPF6AfeUv-zvuVGpjmMZHkIlJUe0/edit?usp=sharing

I case I have to take back the code how can I fix the "Error:Attempt to probe off the bed."?
Thank you.

Roxy-3D, 2017-11-28:
Shrink the 120 down to a number that works....
```
  #define UBL_PROBE_PT_2_X _PX(DELTA_PROBEABLE_RADIUS, 120) // =-40
  #define UBL_PROBE_PT_2_Y _PY(DELTA_PROBEABLE_RADIUS, 120) // =69.2
```

apprentice10, 2017-11-29:
Ok! thank you!

closed 2017-11-29 by apprentice10