MarlinFirmware/Marlin #8592: Kossel Anycubic freeze while printing

apprentice10, 2017-11-28:
### Bug Report
- Description: Kossel Anycubic freeze while printing
- Expected behaviour: not freezing.
- Actual behaviour: short after I launch the print from the SD card (2 to 4th layer) the printer would simply stop everything: motors, fan, LCD remains ON but unresponsive
- Steps to reproduce:
  - From the LCD panel launch a "print from SD"
  - wait

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1511667/Marlin.zip)

My troubleshooting:
-Formating SD
-Trying different SD
-change the #define LCD_UPDATE_INTERVAL 100 to 1000 (from: https://github.com/MarlinFirmware/Marlin/issues/6992)
-Disconnected the USB cable

Gcode
[smartphone_stand.zip](https://github.com/MarlinFirmware/Marlin/files/1511677/smartphone_stand.zip)

thinkyhead, 2017-11-28:
Have you tested the `bugfix-1.1.x` branch?

apprentice10, 2017-11-28:
Yes! actually I had this problem with bugfix 1.1.x , I come from 1.1.6 witch printed very smoothly.

thinkyhead, 2018-02-09:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to confirm that the issue is fixed. If we already got it then you can close the issue. If you still see the bad behavior then we can investigate further.

jar0fair, 2018-02-12:
Came here to say that I am having this issue too. Suddenly, my printer is shutting off 4 or 5 minutes after engaging. Everything stops. Fans, motors, and everything freezes. It wasn’t happening until I moved the printer to another room.

thinkyhead, 2018-02-20:
@jar0fair — Printing from SD or from a host, or both?

jar0fair, 2018-02-20:
@thinkyhead I haven’t tried printing from my PC. But, I have tried printing from two different SDs. It was working fine though, this was rather sudden, but it is now persistent.

thinkyhead, 2018-02-25:
@jar0fair — Start a new issue and fill out the form. Without details and data it is impossible to move forward.

closed 2018-02-25 by thinkyhead

philippeowagner, 2018-06-06:
Having the same issue as @jar0fair reported. Moved the printer and it crashed after a few minutes of printing.