MarlinFirmware/Marlin #8593: [bugfix-1.1.x] G28 not homing Z axis

pandel, 2017-11-28:
- Description: Today I updated my Anet A8 bord to the current bugfix-1.1.x git revision. After M502 and M500 trying to home all axes doesn't work anymore for Z axis.

- Expected behaviour: G28 executes homing X, Y and Z
- Actual behaviour: G28 homes X and Y, but Z axis raises a bit, comes down fast, goes up a bit, comes down slow, resulting in about the same position as before.
```
SENT: G28
READ: echo:G28
G28
READ: >>> gcode_G28
READ: Machine Type: Cartesian
READ: Probe: FIX_MOUNTED_PROBE
READ: Probe Offset X:0 Y:33 Z:0.00 (Middle-Back & Same Z as Nozzle)
READ: Auto Bed Leveling: BILINEAR (disabled)
READ: 
READ:   current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
READ: > endstops.enable(true)
READ: Raise Z (before homing) to 5.00
READ: >>> do_blocking_move_to(0.00, 0.00, 5.00)
SENT: M105
READ: <<< do_blocking_move_to
READ: >>> homeaxis(X)
READ: Home 1 Fast:
READ: >>> do_homing_move(X, -331.50, 0.00)
READ:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
READ: echo:busy: processing
busy: processing
SENT: M105
READ: <<< do_homing_move(X)
READ: Move Away:
READ: >>> do_homing_move(X, 5.00, 0.00)
READ:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
READ: <<< do_homing_move(X)
READ: Home 2 Slow:
READ: >>> do_homing_move(X, -10.00, 25.00)
READ:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
READ: echo:busy: processing
busy: processing
READ: <<< do_homing_move(X)
READ: >>> set_axis_is_at_home(X)
READ: For X axis:
READ:  home_offset = 0.00
READ:  position_shift = 0.00
READ:  soft_endstop_min = -21.00
READ:  soft_endstop_max = 200.00
READ: > home_offset[X] = 0.00
READ:   current_position=(-21.00, 0.00, 5.00) : 
READ: <<< set_axis_is_at_home(X)
READ:   current_position=(-21.00, 0.00, 5.00) : sync_plan_position
READ:   current_position=(-21.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
READ: <<< homeaxis(X)
READ:   current_position=(-21.00, 0.00, 5.00) : > homeX
READ: >>> homeaxis(Y)
READ: Home 1 Fast:
READ: >>> do_homing_move(Y, -313.50, 0.00)
READ:   current_position=(-21.00, 0.00, 5.00) : sync_plan_position
SENT: M105
READ: <<< do_homing_move(Y)
READ: Move Away:
READ: >>> do_homing_move(Y, 5.00, 0.00)
READ:   current_position=(-21.00, 0.00, 5.00) : sync_plan_position
READ: <<< do_homing_move(Y)
READ: Home 2 Slow:
READ: >>> do_homing_move(Y, -10.00, 25.00)
READ:   current_position=(-21.00, 0.00, 5.00) : sync_plan_position
READ: echo:busy: processing
busy: processing
READ: <<< do_homing_move(Y)
READ: >>> set_axis_is_at_home(Y)
READ: For Y axis:
READ:  home_offset = 0.00
READ:  position_shift = 0.00
READ:  soft_endstop_min = -9.00
READ:  soft_endstop_max = 200.00
READ: > home_offset[Y] = 0.00
READ:   current_position=(-21.00, -9.00, 5.00) : 
READ: <<< set_axis_is_at_home(Y)
READ:   current_position=(-21.00, -9.00, 5.00) : sync_plan_position
READ:   current_position=(-21.00, -9.00, 5.00) : > AFTER set_axis_is_at_home
READ: <<< homeaxis(Y)
READ:   current_position=(-21.00, -9.00, 5.00) : > homeY
READ: Z_SAFE_HOMING >>>
READ:   current_position=(-21.00, -9.00, 5.00) : sync_plan_position
READ:   destination=(100.00, 67.00, 5.00) : Z_SAFE_HOMING
READ: >>> do_blocking_move_to(100.00, 67.00, 5.00)
SENT: M105
READ: <<< do_blocking_move_to
READ: >>> homeaxis(Z)
READ:   current_position=(100.00, 67.00, 5.00) : set_probe_deployed
READ: deploy: 1
READ: do_probe_raise(10.00)
READ: >>> do_blocking_move_to(100.00, 67.00, 10.00)
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to(100.00, 67.00, 10.00)
READ: <<< do_blocking_move_to
READ: Home 1 Fast:
READ: >>> do_homing_move(Z, -330.00, 0.00)
READ:   current_position=(100.00, 67.00, 0.00) : sync_plan_position
SENT: M105
READ: <<< do_homing_move(Z)
READ: Move Away:
READ: >>> do_homing_move(Z, 5.00, 0.00)
READ:   current_position=(100.00, 67.00, 0.00) : sync_plan_position
READ: echo:busy: processing
busy: processing
READ: <<< do_homing_move(Z)
READ: Home 2 Slow:
READ: >>> do_homing_move(Z, -10.00, 1.50)
READ:   current_position=(100.00, 67.00, 0.00) : sync_plan_position
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: <<< do_homing_move(Z)
READ: >>> set_axis_is_at_home(Z)
READ: For Z axis:
READ:  home_offset = 0.00
READ:  position_shift = 0.00
READ:  soft_endstop_min = 0.00
READ:  soft_endstop_max = 220.00
READ: *** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
READ: > zprobe_zoffset = 0.00
READ: > home_offset[Z] = 0.00
READ:   current_position=(100.00, 67.00, 0.00) : 
READ: <<< set_axis_is_at_home(Z)
READ:   current_position=(100.00, 67.00, 0.00) : sync_plan_position
READ:   current_position=(100.00, 67.00, 0.00) : > AFTER set_axis_is_at_home
READ:   current_position=(100.00, 67.00, 0.00) : set_probe_deployed
READ: deploy: 0
READ: do_probe_raise(10.00)
READ: >>> do_blocking_move_to(100.00, 67.00, 10.00)
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to(100.00, 67.00, 10.00)
READ: <<< do_blocking_move_to
READ: <<< homeaxis(Z)
READ: <<< Z_SAFE_HOMING
READ:   current_position=(100.00, 67.00, 10.00) : > (home_all || homeZ) > final
READ:   current_position=(100.00, 67.00, 10.00) : sync_plan_position
READ:   current_position=(100.00, 67.00, 10.00) : clean_up_after_endstop_or_probe_move
READ: X:100.00 Y:67.00 Z:10.00 E:0.00 Count X:10000 Y:6700 Z:4000
READ: <<< gcode_G28
```

- Current config: [config.zip](https://github.com/MarlinFirmware/Marlin/files/1511811/config.zip)

- I am using a IR sensor ABL sensor.

- Also, I had to apply this patch for a successful compilation: https://github.com/MarlinFirmware/Marlin/issues/8577#issuecomment-347304170

I hope, that someone can help me with this!

Regards,
Holger

thinkyhead, 2017-11-28:
Does `M119` show that your Z endstop is working perfectly?

pandel, 2017-11-29:
Jep, it is working correctly. I tried ```M119``` 4 or 5 times with a piece of paper, my hands, some packaging in different colors etc. because I didn't understand what was happening ;-)

I don't need ```M401``` because my IR sensor isn't servo controlled. Never had to use ```M120```, what is it good for? I don't really understand the description in the wiki...

pandel, 2017-12-01:
@thinkyhead Do you still need any other feedback?

AnHardt, 2017-12-01:
The question is not if your sensor is seeing all kinds of materials. The question is what does it see if nothing cloth to it? The symptoms tell a story about seeing false positives.

pandel, 2017-12-04:
Yes, one could think of false positives. But with the old firmware it is seeing nothing when there is nothing.

And I would not speak of false triggering at all! The sensor has a small red led that lights up when the sensor triggers - and this led isn't Marlin controlled ;-)! It does that immediately when I hold a sheet of paper or my finger right below it, but the led doesn't light up during the strange G28 phase I described.

SO, it is obvious that the sensor doesn't trigger at all and the firmware is just using some old values it knows and doesn't take the sensor into account at all. On the contrary, when you disturb the homing process by triggering the sensor it is taking that new information into account immediately. So I think that somehow the G28 doesn't invalidate the current end stop states or something like that before it starts moving...

thinkyhead, 2017-12-05:
Sorry for the long delay. We're trying to get Marlin patched up and stable for the next release.

Please give it another test with today's `bugfix-1.1.x`. Be sure to do `M502` and `M500` before testing. I'd like to see another log output, with `DEBUG_LEVELING_FEATURE` and `M111 S247` to enable maximum logging. If you can post a video that would also help. I'm not able to tell from the previous log where the problem manifests.

pandel, 2017-12-05:
Hey @tinkyhead, I'm totally fine with the delay ;-)!

I'll try to test it this evening. I also thought about adding additional debug output to the endstop triggering routine, something like a live version of M119, so that we can see, if we are alking about false positives or not. Makes that any sense?

pandel, 2017-12-05:
@thinkyhead 
I did try it again with the current git revision and what can I say, IT WORKS 🥇  I'm really happy! And, additionally, I tried bilinear leveling again (which didn't work for me for a very long time) and it also works now! It is absolutely brilliant!

I really don't know what has caused the issue and what has been changed since, but I think I save the current firmware on a USB stick and put it into my bank deposit to have a working copy for all times 👍 

And by the way, did anything change regarding the stepper motor control? It seems as if they work a bit quieter now - but that could also be my ears getting older and older...

Many thanks for your hints!

closed 2017-12-05 by pandel

thinkyhead, 2018-01-26:
Just catching up on issues today…

To answer your question, there have been some changes to stepper control to deal with stuttering and move chaining; nothing that should make it much quieter. But you're not the first person to mention it, so maybe we did something clever without realizing…