MarlinFirmware/Marlin #8594: [FR] LIN ADVANCE only for the outer layers

BeerStein, 2017-11-29:
Would it be at all possible to enable LIN ADVANCE only for the outer layers. I believe that is where one need  quality improvement.  Doing LIN Advance during say a honeycomb infill only adds noise during the print. 
Thanks for all the hard work.

coolio986, 2017-11-29:
I don't know of any way to add this in firmware directly. The gcode parser doesn't know the difference between infill and outer perimeters, only the slicer would know that. 

One could make the case that the slicer could inject M900 Kxx (where xx is the linear advance factor) for perimeters, then M900 K0 to disable it for infill. But that could interfere with the E counts when turned on and off in the motion planner.

thinkyhead, 2017-11-29:
If your slicer can insert G-code at the start of perimeters and different G-code at the start of infill, you can have it insert `M900 K[yourKvalue]` at the start of infill and `M900 K0` at the start of perimeters.

thinkyhead, 2017-11-29:
If your slicer can insert G-code at the start of perimeters and different G-code at the start of infill, you can have it insert `M900 K[yourKvalue]` at the start of perimeters and `M900 K0` at the start of infill.

thinkyhead, 2017-11-29:
> that could interfere with the E counts when turned on and off in the motion planner.

I'm pretty sure there's no problem, since the previous move finishes before the new `M900` takes effect, and the high-level E value works just the same with and without advance.

JimBrown, 2017-11-29:
I can confirm that injecting M900 in the gcode to enable lin advance only for permiters works just fine. Did it for a while when I had a bowden set up.

Using Simplify3D, I added this to my post processing commands:
```
{REPLACE "perimeter\nG" "perimeter\nM900 K400\nG"}
{REPLACE "; solid layer\n" "; solid layer\nM900 K0\n"}
{REPLACE "fill\n" "fill\nM900 K0\n"}
{REPLACE "; skirt\n" "; skirt\nM900 K0\n"}
{REPLACE "; bridge\n" "; bridge\nM900 K0\n"}
```
No idea how to do this with other slicers.

BeerStein, 2017-11-29:
Thanks to all who have responded. 
Jim, I am using slic3r and will certainly investigate the possibility to add some processing commands. Thanks for the tip.

Leaving this one open for a while. Hopefully I could find a solution for slic3r. If so I will post it.

Lonczakt, 2017-12-02:
@JimBrown, that's quite the K value! I think I'm using 15. You running a bowden setup?

coolio986, 2017-12-02:
I use a K of 200-250 on my bowden setup, 20 - 40 on direct.

JimBrown, 2017-12-02:
@Lonczakt, yes, I was. Now using direct drive and K value is 40.

closed 2018-02-10 by thinkyhead