MarlinFirmware/Marlin #8598: [FR] Turn off LCD after some time

apprentice10, 2017-11-29:
The idea came to me after seen this post https://github.com/MarlinFirmware/Marlin/issues/6992 stating that increasing the
#define LCD_UPDATE_INTERVAL 100
would save some resources for the machine. Since I watch my printer occasionally to make sure is printing ok, I don't need the LCD ON all the time. It would be nice to have a feature to keep the LCD ON for a period of time. it would turn on again as soon as you spin or press the wheel.

marcio-ao, 2017-12-05:
Heh, how about adding some [flying toasters](https://en.wikipedia.org/wiki/After_Dark_(software)#/media/File:After_Dark_Flying_Toasters.png)?

apprentice10, 2017-12-06:
The thing is Delta printers need more resources to calculate the movement of the stepper motors than for example cartesian printers, I've been reading also that refreshing the LCD would take some resources from the processor. I've been having problems with my Delta halting while printing after I've done the UBL (witch I think adds more process), So turning off the LCD to save some resources doesn't seem a bad idea.
Not sure how a flying toaster would help thou? ;D

marcio-ao, 2017-12-06:
@apprentice10 : Marlin 2.0 will bring support for 32-bit processors -- I'll toast to that!

greenonline, 2018-09-13:
Isn't this a duplicate of [[FR] Option to turn off lcd when idle #4728](https://github.com/MarlinFirmware/Marlin/issues/4728)

Sangoku, 2019-01-23:
Hmmm... this sounds interesting. Im willing to try to implement this feature. Any ideas where to start? im still new to marlin...

marcio-ao, 2019-01-24:
@Sangoku: AFAIK, the UI consumes no resources when in the menu items. In fact, back when I was using a delta printer, I used to disable the timeout that causes the return to the status screen, so that when printing I could simply go into a sub-menu for more performance. This may be a way to achieve what you are looking for, although it does require the user to take action to go into a submenu at the start of a print.

boelle, 2019-02-19:
@thinkyhead dublicate of https://github.com/MarlinFirmware/Marlin/issues/4728

@apprentice10 how will you turn off the display??

marcio-ao, 2019-02-19:
@boelle: By turn off, I assume he means simply to stop updating it.

boelle, 2019-02-19:
that would not make much sense, not to my small brain :-D

but its an dublicate non the less

marcio-ao, 2019-02-19:
@boelle: On delta printers, the CPU is so busy doing calculations that the act of refreshing the displays slows it down enough to cause stuttering during a print. This causes little dots to appear all over the print -- it looks like chicken pox! I've seen it happen before.

So @apprentice10 wants Marlin to stop writing to the display. One way to do this elegantly would be to clear the display and write a message to the screen like "Push button to wake up", then when the button is pressed, redraw the screen and resume updating.

boelle, 2019-02-19:
but is it not an dub of #4728

marcio-ao, 2019-02-20:
@boelle: Yes, it is a duplicate. A lot of people want this feature, evidently.

boelle, 2019-02-20:
then i suggest that we or @apprentice10 close this so we dont have a dublicate

@thinkyhead agree?

marcio-ao, 2019-02-20:
> then i suggest that we or @apprentice10 close this so we dont have a dublicate

@boelle: If by "we" you mean you or I, I can't close other people's ticket's. I don't have those super-powers :)

boelle, 2019-02-20:
ohh... i meant that @thinkyhead can close it

apprentice10, 2019-02-21:
Sorry I haven't been around:
@marcio-ao Yes the "Push button to wake up" pretty much sums what I had in mind.
@boelle and @greenonline I didn't knew about that post at the time, I see that @thinkyhead has already answer there so I think is ok to close this one.

boelle, 2019-02-21:
@apprentice10 maybe click the green close button below then?

thinkyhead, 2020-07-19:
Re-opening this because it isn't really implemented for our common displays, and it would still be a nice thing to add at some point.

sjasonsmith, 2020-09-11:
Duplicate of #4728

qwewer0, 2021-01-01:
I think this should be closed, as there is already (https://github.com/MarlinFirmware/Marlin/issues/4728) a FR for such feature?

sjasonsmith, 2021-01-01:
Yes, closing as a duplicate. I read through both, and didn't see any real differentiating information in this one.

closed 2021-01-01 by sjasonsmith