MarlinFirmware/Marlin #8606: [FR] disable MIN_STEPS_PER_SEGMENT when calibrating

rszasz, 2017-11-30:
I have a delta printer, and manual calibration was working except for a weird issue where it would jump every few requested steps rather than moving at each requested change in position. 

I finally realized it was the MIN_STEPS_PER_SEGMENT waiting for 6 steps before moving. 

Would it be possible to disable step combining during calibration? And would it be possible to check that the PROBE_MANUALLY_STEP variable is an integer multiple (or a 2^n multiple) of actual stepper motor steps?

thinkyhead, 2018-02-09:
In which procedure does this show up?

Please fill out this template:

---
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

thinkyhead, 2018-02-10:
Note that on delta, segments refers to the movement of the steppers, so it can't be predicted without first doing kinematics. Otherwise, we could easily make sure the minimum movement distance is at least `MIN_STEPS_PER_SEGMENT`.

We can address this by placing `MIN_STEPS_PER_SEGMENT` into a variable and setting it to 0 temporarily. The tricky part is that we don't always know that we are "in a print job" so it may end up being left at 0 in some instances, such as when the nozzle is cold. Also note that this will add slight overhead to the planner.

thinkyhead, 2018-02-19:
You can also address this by placing `MIN_STEPS_PER_SEGMENT` into a variable and setting it to 0 temporarily. For example, in `manage_manual_move`…
```diff
+ min_steps_per_segment = 0;
  processing_manual_move = true;
  prepare_move_to_destination(); // will call set_current_from_destination()
  processing_manual_move = false;
+ min_steps_per_segment = MIN_STEPS_PER_SEGMENT;
```

thinkyhead, 2018-02-19:
It's tricky at the current time to modify the min-steps-per-segment value only for certain LCD screens because the LCD code is written in such a way that we don't have "only on entry" and "only on exit" screen behaviors. If we had those then we could set `planner.min_steps_per_segment` to 0 on entry to a screen and then back to `MIN_STEPS_PER_SEGMENT` on exit. Likewise, if we want to check whether a print job is in progress, we now test that an SD file is open, or that the nozzle has been heated. Since many of us do our bed calibration with a heated nozzle, behavior will be as if a print job is active. The one thing we can do is set it to 0 when inside of blocking procedures like `G29`. Unfortunately that doesn't help with your need to have it set during a certain LCD screen.

When we overhaul the LCD code this is something we want to address. A menu screen should be able to have a behavior that only occurs on first entry, and another when it is exited. Expect us to get into that more after the 2.0 release.

boelle, 2019-02-19:
@rszasz have you tried latest bugfix 2.0?

boelle, 2019-06-22:
wrong label and i think we should even close it

charakaSamawry, 2020-07-25:
Noob here but doing manual ubl and wondering do people just set MIN_STEPS_PER_SEGMENT to 1?? Sorry if this isn't the place for such queries