MarlinFirmware/Marlin #8609: G26 Prints Incomplete Pattern with Bilinear Levelling

benlye, 2017-11-30:
I'm trying out G26 on my printer (Cartesian RAMPS setup) with bilinear levelling and the test pattern is weird.  
* There are lines missing
* Some lines are printed multiple times
* The lines seem out of alignment with the circles (they don't make the little cross-hair pattern)

Config files as well as serial output from the test with G26 debug enabled are attached.  This is with bugfix-2.0.x downloaded and compiled tonight.

The missing lines are not bad adhesion - they were never printed:
[image]

The crosshairs don't line up:
[image]

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1519450/config.zip)
[serial.log](https://github.com/MarlinFirmware/Marlin/files/1519453/serial.log)

If this isn't supposed to fully work with bilinear levelling yet, I'm happy to be told that's the case.

Ben

Roxy-3D, 2017-11-30:
It is supposed to work with Bi-Linear.   But you maybe the first person to use it.

benlye, 2017-11-30:
> It is supposed to work with Bi-Linear. But you maybe the first person to use it.

OK, good to know.

I goofed up the config files in the first post, and posted them while in the middle of a Github merge.  Here are the correct ones.

[config-corrected.zip.zip](https://github.com/MarlinFirmware/Marlin/files/1519521/config-corrected.zip.zip)

Roxy-3D, 2017-12-01:
I'm bringing up a new mesh on a new piece of glass.  The mesh needs to be fine tuned.  But the G26 pattern prints correctly with UBL.   We might need somebody with C coding skills that is running Bi-Linear to take a look at this:

[image]

benlye, 2017-12-01:
I've been trying to wrap my head around this...

The configuration was for a 5x5 bilinear grid, with these boundaries:
```
  #define LEFT_PROBE_BED_POSITION 15
  #define RIGHT_PROBE_BED_POSITION 195
  #define FRONT_PROBE_BED_POSITION 15
  #define BACK_PROBE_BED_POSITION 180
```

So probe point [0,0] is at [15,15] and the distance between horizontal mesh points is 45.

In G26.cpp:
```
#define SIZE_OF_INTERSECTION_CIRCLES 5
#define SIZE_OF_CROSSHAIRS 3
```

So there should be circles at [60,15] and [105,15] and a horizontal line between them from [62,15] to [103,15].

According to the log this is the line which was printed:
`Recv:  Connecting with horizontal line (sx=107.00, sy=15.00) -> (ex=60.00, ey=15.00)`

Still trying to work out why 😕

Roxy-3D, 2017-12-01:
Those probe bed positions are not used by UBL.    You can reserve a border around the bed with INSET.

benlye, 2017-12-02:
I'm not using UBL.  The entire debug log is in the first post.

The nozzle goes to all areas, as all the circles seem to be printed correctly, it's the lines between them that are screwed up.

benlye, 2017-12-02:
A couple more observations:
* Lines printed 'forward' (i.e. low X to higher X) seem OK, only lines printed backwards appear to be too long

e.g. `Recv:  Connecting with horizontal line (sx=17.00, sy=56.00) -> (ex=58.00, ey=56.00)` is OK (17-58 is right)

* There are lots of short backwards lines, e.g. 
`Recv:  Connecting with vertical line (sx=15.00, sy=58.00) -> (ex=15.00, ey=55.00)`

What I don't understand (yet) is why lines are planned backwards at all.  I know that the printing will be reversed if nozzle is nearer the end than the beginning, but that seems to happen after the line is planned and the debug output written, so even for a reversed line it should appear in the log as 'forwards'

I'm suspicious of these lines:
```
sx = _GET_MESH_X(  i  ) + (SIZE_OF_INTERSECTION_CIRCLES - (SIZE_OF_CROSSHAIRS)); // right edge
ex = _GET_MESH_X(i + 1) - (SIZE_OF_INTERSECTION_CIRCLES - (SIZE_OF_CROSSHAIRS)); // left edge
```

Because the maths doesn't correlate to the comments - `GET_MESH_X(i + 1)` should be to the right of `GET_MESH_X( i )` not the left, shouldn't it?

I guess my question is are the lines supposed to be planned for printing left to right or right to left (or both, and I really don't understand what's going on here)?

Roxy-3D, 2017-12-02:
Yeah...  Those comments are wrong.   It scans through each circle and looks to see if the circle next to it exists.  If it does and there isn't a line there, it puts a lint there.    But the left and right comments are flipped.

benlye, 2017-12-02:
Thanks, Roxy, I'll give it a whirl.  I've added a bit more debugging output of my own as well.  I think I'm getting close to figuring it out.

Roxy-3D, 2017-12-02:
What ever is wrong...  It isn't much.    I say that for two reasons.  First, it works with UBL.   But the other thing is you do get a lot of circles and lines.    So what gives?    There is some small logic error that is causing trouble....

It will be interesting if those connecting lines are not being drawn because the 'position is not reachable'.   If that turns out to be the case, we will have to figure out why it is coming up with that idea.

benlye, 2017-12-02:
Just ran the debug code.  There were no 'position not reachable' results.  I'm pretty sure that it's breaking when it tries to find the location of `_GET_MESH_X(i + 1)`.

I modified the debug output to include the index of the mesh point as well as it's coordinates.

Check this out:
`Recv:  Connecting sxi=1 at 60,15 with exi=2 at 61,15 with horizontal line (sx=62.00, sy=15.00) -> (ex=53.00, ey=15.00)`

exi=2 (as in X mesh location 2) should be at 105,15, yet it's given as 61, which is `_GET_MESH_X( i ) + 1` for i = 1.

So perhaps `_GET_MESH` works differently for ABL vs. UBL?

I'm closing in...

Roxy-3D, 2017-12-02:
```
#if ENABLED(AUTO_BED_LEVELING_BILINEAR)
  #define _GET_MESH_X(I) bilinear_start[X_AXIS] + I * bilinear_grid_spacing[X_AXIS]
  #define _GET_MESH_Y(J) bilinear_start[Y_AXIS] + J * bilinear_grid_spacing[Y_AXIS]
#elif ENABLED(AUTO_BED_LEVELING_UBL)
  #define _GET_MESH_X(I) ubl.mesh_index_to_xpos(I)
  #define _GET_MESH_Y(J) ubl.mesh_index_to_ypos(J)
#elif ENABLED(MESH_BED_LEVELING)
  #define _GET_MESH_X(I) mbl.index_to_xpos[I]
  #define _GET_MESH_Y(J) mbl.index_to_ypos[J]
#endif
```
I'm wondering if bilinear_grid_spacing[X_AXIS] & bilinear_grid_spacing[Y_AXIS] are correct?   We may want to print those out too with the error message.   It seems to be coming up with the wrong location.

benlye, 2017-12-02:
Yep, chasing that down now.

benlye, 2017-12-02:
Got it!

The problem is here,
`ex = _GET_MESH_X(i + 1) - (SIZE_OF_INTERSECTION_CIRCLES - (SIZE_OF_CROSSHAIRS));`

in the passing of `i + 1` to the macro:
`#define _GET_MESH_X(I) bilinear_start[X_AXIS] + I * bilinear_grid_spacing[X_AXIS]`

For i=2, the macro expands as:
15 + 2 + 1 * 45 = 63 (wrong) 

Change the macros to evaluate the input parameter by enclosing it in parentheses:
```
#define _GET_MESH_X(I) bilinear_start[X_AXIS] + (I) * bilinear_grid_spacing[X_AXIS]
#define _GET_MESH_Y(J) bilinear_start[Y_AXIS] + (J) * bilinear_grid_spacing[Y_AXIS]
```
And it evaluates as:
15 + (2+1) * 45 = 150 (right)

😄 

I'll put a PR together for that change - it doesn't seem like modifying the macro in that way will impact any other functionality.

Roxy-3D, 2017-12-02:
We need to change both branches...   I can do it easier...  I'll do it right now!

THANKS SO MUCH!!!!    (It did turn out to be very simple logic problem!)

benlye, 2017-12-02:
And here's the proof - it's a bit scruffy, but I think it's beautiful 😉 

[image]

Thanks, Roxy.

benlye, 2017-12-02:
@Roxy-3D just saw your commit in bugfix-2.0.x - you put the parenthesis in the wrong place!

You changed it to `#define _GET_MESH_X(I) (bilinear_start[X_AXIS] + I * bilinear_grid_spacing[X_AXIS])`

But it needs to be `#define _GET_MESH_X(I) bilinear_start[X_AXIS] + (I) * bilinear_grid_spacing[X_AXIS]`

So that the part which gets passed as the parameter is evaluated first.

Ben

benlye, 2017-12-02:
Never mind, fixed now 👍

closed 2017-12-02 by benlye

Roxy-3D, 2017-12-02:
Yeah...   I made a mistake....   But when I went and double checked things I found it.

benlye, 2017-12-02:
BTW, I had to fix M49 to get the G26 debugging to work.  It's not super important, but I opened #8607 to fix it if you want to take a look.  It's a simple serial output fix.

Roxy-3D, 2017-12-02:
That has kind of been on my To Do list.   It was the G26 stuff that necessitated having more debug information on the movement.   But  it was mis-named because it is also used within the UBL code.

I wonder if it would be better to fix this via:
```
 serialprintPGM(g26_debug_flag ? PSTR("On\n") : PSTR("Off\n"));
```
That adds two bytes of text but it elmininates the call to SERIAL_EOL();