MarlinFirmware/Marlin #8617: Filament Width Sensor additional code required to act as RUN-OUT sensor also.

weanything, 2017-12-01:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

weanything, 2017-12-01:
So I know you have done great work incorporating the filament width sensor into the Marlin code so that we can more accurately control extrusion speed relative to volume of filament etc.  And it seems a natural step to me to now enhance this somewhat to also allow the sensor to act as a Filament End sensor with no hardware changes, just a bit of Marlin magic.   I am not a coder so....

empakoso, 2017-12-12:
nice idea.  I would definitely use this.

empakoso, 2017-12-20:
We now have a working MIN and MAX setting along with a setting to control out of tolerance measurements https://github.com/MarlinFirmware/Marlin/issues/8767#issuecomment-352951507 

There has been also been mention from @thinkyhead about your tolerance based action.

I see your idea in the configuration_adv as being something along the lines:

//immediate action on measured result at the sensor
//#define FILAMENT_WIDTH_ACTION M600/nM150 R255 G255 U255/nM117 filament ran out!
define FILAMENT_WIDTH_ACTION_MIN 0.0
define FILAMENT_WIDTH_ACTION_MAX 0.5

I am no help in the coding department, but would definitely test this feature!

thinkyhead, 2017-12-25:
For proper run-out detection you'd want to look for several samples in a row to be out of range before firing the runout action. Keep a counter which increments when a value in the "runout range" is detected, and reset if a normal sample arrives.

boelle, 2019-02-19:
might i add that to make sense the sensor needs to check the filament on 2 axis what if seen from above its one diameter and another seen from the side?

boelle, 2019-06-22:
missing label

ETE-Design, 2020-04-16:
Could you pls. add this... Would be great to make it possible to use Width Sensor for Runout at the same time...

thinkyhead, 2020-04-16:
@ETE-Design — I'll put it on the TODO list and work in it in my copious free time.