MarlinFirmware/Marlin #8618: Marlin 2.0 on MKS sbase

vladimirRUS007, 2017-12-01:
Hello, an error occurs when the installation of Marlin 2.0 mks sbase. Board appears in device Manager as" usb device marlin " My steps for installation.

I downloaded the Marlin 2.0
Set Atom
Downloaded PIO
Changed the #define MOTHERBOARD BOARD_MKS_SBASE
IN the lower left corner clicked on PIO Bulid (дсз1768)
Downloaded firmware.bin from .pioenvs/lpc1768 7) zacamil on the stick
connect the charge mks sbase and it appears as :" marlin usb device "
[image]

What are your thoughts...

AnHardt, 2017-12-01:
Install the Smoothie device driver. See http://smoothieware.org/windows-drivers.

vladimirRUS007, 2017-12-01:
well, now try

vladimirRUS007, 2017-12-01:
driver installed, but nothing has changed

vladimirRUS007, 2017-12-01:
Thank you all, turned out...

closed 2017-12-17 by thinkyhead

madmax2, 2018-04-18:
@vladimirRUS007 how did you fix it?

thinkyhead, 2018-04-20:
@madmax2 — Full instructions for ARM installation are now posted at http://marlinfw.org/docs/basics/install_arm.html