MarlinFirmware/Marlin #8620: Simultaneous extrusion of two extruders filament

vladimirRUS007, 2017-12-01:
Hi all. Required command G code, after which will begin simultaneously push the two plastic extruders.

AnHardt, 2017-12-01:
Likely it need more a new Configuration.h then a g-code.

vladimirRUS007, 2017-12-01:
ok, what do I change in the config?

AnHardt, 2017-12-01:
Impossible to say - without your config - without you stating what variete you want to go to.

vladimirRUS007, 2017-12-01:
I want to use a marlin 2.0, but the configuration is not ready, because I can't make friends marlin 2.0 and MKS sbase

vladimirRUS007, 2017-12-01:
driver installed, but nothing has changed

vladimirRUS007, 2017-12-01:
I don't know what to change in the configuration, I want to command g code like this.
G91 E0 E1
G1 F100 E1xxx E2xxx 
How to implement this?
[image]
the following files of my firmware

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1523003/config.zip)

bjarchi, 2017-12-02:
Is the goal of this to do what's called 'copy printing' or the like? In other words, to have both extruders simultaneously print the same object in a standard dual-extrusion setup? If so, that's already implemented in bugfix-1.1.x and I think also mainline 1.1.6.

You need to enable it in firmware, first, by finding the following define in `Configuration_adv.h`:
`#define DUAL_NOZZLE_DUPLICATION_MODE`
And making sure it is uncommented (therefore enabled)

Then you enable to dual-nozzle simultaneous extrusion mode before a print by issuing a G-code command `M605 S2` to put the machine into simultaneous extrusion mode. See the documentation for `M605` [here](http://marlinfw.org/docs/gcode/M605.html). 

I don't think this will get you independent control over the extrusion length for each extruder; not sure if that's important for your use case. You can enable a define to allow you to set different esteps/mm though - search for `//#define DISTINCT_E_FACTORS` in `Configuration.h`

vladimirRUS007, 2017-12-03:
@bjarchi 
Thank you

bjarchi, 2017-12-04:
> Thank you

You're welcome; I'm glad to help! Does that solve your problem/question? If so you can mark the issue closed via the web interface and it will drop off of the open issues feed.

closed 2017-12-05 by thinkyhead