MarlinFirmware/Marlin #8621: Can't add 2nd Extruder to Firmware without HEADER_1_PIN error on compiling

ElectricCaffeine, 2017-12-01:
### Compile Error
When I compile with second extruder (number of extruders line) ` I get an error:
```
HEATER_1_PIN not defined for this board

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

AnHardt, 2017-12-01:
How about defining a board with two extruders. RAMPS EEB instead of EFB for example.

ElectricCaffeine, 2017-12-01:
How do I do that, @AnHardt ? :) Please tell me how

Bob-the-Kuhn, 2017-12-01:
If you're using a RAMPS board then replace **BOARD_RAMPS_14_EFB** with **BOARD_RAMPS_14_EEB** in the Configuration.h file.

If you're using a different board then please attach your configuration files.

ElectricCaffeine, 2017-12-01:
Thanks Bob! I’ll try it out.
Just a quick question as well: If I am replacing the configuration.h file With the one made for my TiVo tarantula printer, Well that must suck the rest of my marlin firmware that is new?

ElectricCaffeine, 2017-12-02:
Thanks so much! That resolved it. I now have extruder, extruder 1 and Extruder 2 (not sure why a third but progress nonetheless)

Bob-the-Kuhn, 2017-12-02:
If you're converting from TiVo supplied version of Marlin then it's best to manually copy the custom configuration items over to the new config files.  Some vendor supplied versions are quite old and have caused strange problems.

ElectricCaffeine, 2017-12-02:
I was able to successfully get Marlin 1.1.6 flashed onto the RAMPS board of my printer (with large Reprap LCD display).

My current problem of inaccurate thermistor readings still continues, So it doesn’t seem to be a Firmware issue.

See attached photos:

Bob-the-Kuhn, 2017-12-03:
I take it the thermistor was well behaved before changing over to Marlin.

What type of thermistor do you have?

Please attach the TiVo supplied configuration.h file and the one you're using now.  I'll see if I can spot anything.

ElectricCaffeine, 2017-12-03:
Hi Bob..I use the 100K epcos ones I believe..
I've attached my configuration.h file..

Also, how do you change the direction of the menus with the rotary knob? the default is increase values by turning left..which feels unnatural.

Thanks!
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1525222/Configuration.txt)

Bob-the-Kuhn, 2017-12-03:
The config file is setup properly for an Epcos 100K.   Does the temperature reading change when you heat the thermistor?  

If not then I wonder if the expected input pin is different.  Try copying the pins_RAMPS.h file from the TiVo setup to your new and and see if that fixes it.

If yes then either the wrong thermistor selected or there's a hardware problem.

If none of the above fixes it then I'll need a very detailed description of what it's doing wrong.

On the encoder direction, in Configuration .h see the **Encoder Direction Options** section.

ElectricCaffeine, 2017-12-04:
To give you some background, I had been testing my Tarantula in my basement with a thin carpet that seemed to be a strong conductor for electricity. Even if my Tarantula was on or off, I would get zapped just by touching the printer. I noticed sometimes gibberish would show up on the LCD screen while it was on.

Then at some point, the same thing happened with the stock PCB board..all functions would work except the readings of the temperatures would all start off really high (like 300 or so) just when turning on the machine (and not previously heated).

After replacing the PCB with a new one, the problem was gone..and i was getting great heating and correct temperatures on the screen.

Since then, I have now moved the printer to the garage and no more weird electric zaps and gibberish..sometimes the screen just goes blank, but I have to tackle that bear later (and with the new PCB when it arrives).

Hope this helps.

Bob-the-Kuhn, 2017-12-04:
You could try to move the thermistor to a different analog input.  Maybe that pin isn't blown.

I ran for a long time with blown pins on my 2560 based controller until i plugged a bad stepper driver in.

ElectricCaffeine, 2017-12-04:
The rotary encoder adjustment worked, but I can only see motor movement on adding values (turning right). When I turn knob to Left, no x-y-z motor movements.

Also, I can tell PCB is jacked up because even with thermistors removes, it should read -14 for extruders and heat bed. But it doesn’t, it shows these high temps anyway.

bjarchi, 2017-12-08:
> To give you some background, I had been testing my Tarantula in my basement with a thin carpet that seemed to be a strong conductor for electricity. Even if my Tarantula was on or off, I would get [...]

This sounds very much like you have a problem with ground isolation somewhere in your system or in your house wiring - likely causing your power supply case and ground to be floating at 110VAC above earth ground. This will indeed do Bad Things to your low voltage electronics if a path ground opens up - such as through you.

I would submit that you almost certainly did fry your controller/board, as you suspected, and you might continue to do so. It could have been a site wiring fault (or use of a grounded:ungrounded adapter plugged in backwards to a non-grounded outlet). In that case you're probably fine with the printer in the new location and hopefully connected to a grounded circuit. If it wasn't site wiring then I'd worry about something wrong with the supply or its HV (AC input) wiring - which you really want to get fixed.

closed 2018-02-10 by thinkyhead