MarlinFirmware/Marlin #8627: bugfix-2.0.x Re-Arm RAMPS 1.4 EFB no fan output

vilsed, 2017-12-02:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1523927/Configuration.zip)
Good day, everybody. I'm new to github (used to solve my problems silently, on my own), so hoping that I'm asking for help correctly.

- Description: While compiled on PIO LPC1768 (the one that is supposed to work with Re-Arm, as I understand), having correct board selected in Configuration.h (BOARD_RAMPS_14_RE_ARM_EFB), I'm getting no part cooling fan output (D9). It seems like configuration in pins_RAMPS_RE_ARM is incorrect, although it is. To top it all off, I did had output on that pin with smoothieware, with exact same pin configuration. I'm completely lost here, TBH.
- Expected behaviour: Having PWM output on RAMPS output D9
- Actual behaviour: No output on D9
- Steps to reproduce:
  - Compile latest bugfix-2.0.x using my Configuration.h, Configuration_adv.h and pins_RAMPS_RE_ARM.h files for LPC1768;
  - Observe

In advance, thank you all for help.

hg42, 2017-12-03:
may be, my PR #8622 could solve your Problem.

Try changing the definition of _GET_TIMER in Marlin/src/HAL/HAL_LPC1768/fastio.h to this:

`#define _GET_TIMER(IO)        LPC1768_PIN_PWM(IO)`

If it still does not work, try these, too:

```
#define _GET_INPUT(IO)        (LPC1768_PIN_PIN(IO)>=0)
#define _GET_OUTPUT(IO)       (LPC1768_PIN_PIN(IO)>=0)
```
Don't confuse _GET_xxx with GET_xxx later in the file.

vilsed, 2017-12-03:
Unfortunately, none of the changes worked.

hg42, 2017-12-03:
I reviewed my PRs and it's even more important to add my other PR/change too.
You need
#8616
#8622
or manually change in Marlin/src/HAL/HAL_LPC1768/pinmapping.h:
`constexpr int8_t NUM_DIGITAL_PINS = COUNT(pin_map);`
to
`constexpr uint8_t NUM_DIGITAL_PINS = COUNT(pin_map);`

vilsed, 2017-12-03:
It did the trick, thank you :)

Roxy-3D, 2017-12-03:
@hg42    How confident are you on these two Pull Requests?   Should they be merged?

hg42, 2017-12-03:
@Roxy-3D thanks for asking...it's already merged by @Bob-the-Kuhn.
Both were very obvious and extremely simple from my point of view (number overrun or no implementation of a macro that's used for PWM). Bur not exactly easy to find, if you don't expect it :-)

closed 2018-02-10 by thinkyhead