MarlinFirmware/Marlin #8628: Skew Correction: General Questions

Lonczakt, 2017-12-02:
Big thanks to @Paciente8159 and @thinkyhead for implementing this into Marlin. I'm currently testing it out and in the next week or so will be writing up a document for other users (like linear advance has).  I'd like to use this thread as a way to collect general questions which I can then incorporate into the document. 

I'll start with this one:

_How does the Z skew correction interact with UBL, does it override it? Is it an additional layer on top of the mesh (or below it)?_

I programmed in SAS for 10 years so I can get the gist of very basic C/C++ coding but files like the planner are well outside of my comfort zone. 

Thank again for implementing this!

HekTron802, 2017-12-02:
Based on the comment @Roxy-3D  made in #8204, the Z correction acts, in very simple terms, as an overlay on top of the mesh generated by the UBL system. Is that correct? I guess I'm not understanding how this would work as I thought the mesh itself would correct for any skew in the Z axis.

Paciente8159, 2017-12-06:
@HekTron802 The skew compensations is performed before the bed leveling is applied so that the machine compensates any "unsquareness" before performing leveling (or at least that's the objective).
Bed level and skew are two different things. Think like this: For example you can have perfectly leveled bed if both bed and the X axis are tilted at the exact same angle (Every time you move X axis the table will be at the exact same distance). But if you print a cube like this your part will be a deformed (sideway) relative to XZ.

bjarchi, 2017-12-08:
Here's a question to add:

What machine geometries benefit most/least from having skew correction? I assume it's pretty much for cartesian and core machines (with orthogonal axes) - is there any benefit for deltas? 

In the cartesian family, is it more useful for one class of machines vs others - for example, reprap/i3 type machines where x and y move head and bed, vs Replicator/Ultimaker machines where x and y both move the head? Or do all machine classes *potentially* suffer from a degree of misalignment from orthogonal that makes this valuable to apply? 

Also, how about the relative merits of skew correction in x/y only vs also doing it in z? And how do you calibrate the z skews? Really big calibration cube?

HekTron802, 2017-12-08:
Thanks for the replies, good info and questions.

@Paciente8159: for the do this then do that section of the documentation, I moving towards having UBL performed first then skew, if only because my printer's bed has a couple of "dead" spots in it, making printing anything but a very small object in the center risky. Any thoughts on this?

@bjarchi: I see skew correct as an adjustment that would be most useful for home built and low end printers. While it's certainly possible to build a machine with ,for all practical purposes, orthogonal axes (we've been going it for a long time with milling machines or lathes) making skew correction unneeded, my guess is that the majority of home build printers had their frames, linear rails cut with a hacksaw or cheap bandsaw and measured with "ball park" measuring tools. Basically guaranteeing that the printers axes will not be orthogonal. So I don't think the type of machine has as much as an impact on the need for skew correction as the quality of the build does. Of course I've only build one Cart machine, am currently building a coreXY and don't have any experience with a delta, so I'm far from an expert on this!
 #8623 and the links in it, give info on skew correction along with calibration prints (see the config file update) 

Best!

thinkyhead, 2017-12-09:
Skew may be determined without any bed leveling enabled, but it can help to do bed leveling first to get good adhesion. Skew correction is independent of all leveling and can be enabled by itself. It is currently implemented for all leveling except the cartesian non-segmented version of UBL. The description of how to calibrate the skew feature in `Configuration.h` is pretty good, and should be expanded on for a full article on the topic … and also on the `M852` G-code page. There are some web articles and YouTube videos about skew that they could link to.

bjarchi, 2017-12-10:
@HekTron802 

> #8623 and the links in it, give info on skew correction along with calibration prints (see the config file update)

I've looked over that, and the information in the config file - which basically says 'print a square and measure for xy skew correction; repeat for the other axes'.

I suppose my question was really about what would be a minimal calibration object for measuring skew in the [x,y]-z planes. For xy you can print a square, but that doesn't work once you bring z into the picture, and you want to print as large as possible to accurately measure the skew so a cube would eat a lot of material. 

I'm thinking it would be helpful to have a printable object linked somewhere (ideally customizable) that facilitates measurement of xz and yz skew - something that's self-supporting, relatively easy to print even with a marginal printer, uses minimal material, etc. Still trying to wrap my head around how an optimal calibration object would look, though.

Paciente8159, 2017-12-11:
@bjarchi Have you seen the test 2nd object (for the X/Y-Z axis) in the thingiverse page indicated in the configuration.h? You don't have to print a cube. A squared U shape standing up should allow you to measure the skew just as good (I believe).

Has for the why the "skew adjustment before leveling" the reason I implemented like that was to ensure that the X/Y coordinates are not changed after BL compensation is computed (or you you would be mapping a Z height in the wrong X-Y coordinate). But I think this will only make much off a difference if the skew in the machine is very extreme or you printing surface is highly irregular.

bjarchi, 2017-12-11:
> @bjarchi Have you seen the test 2nd object (for the X/Y-Z axis) in the thingiverse page indicated in the configuration.h? You don't have to print a cube. A squared U shape standing up should allow you to measure the skew just as good (I believe).

No... somehow I missed that link when I was looking through the Config comments most recently. That's pretty much exactly what I was thinking about, so excellent 👍

closed 2018-02-10 by thinkyhead