MarlinFirmware/Marlin #8633: Marlin-bugfix-2.0.x Compile “The filename or extension is too long.”

diver197, 2017-12-03:
Linking everything together...
"C:\Users\diver\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc" -mcpu=cortex-m3 -mthumb -Os -Wl,--gc-sections "-TC:\Users\diver\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.10\variants\arduino_due_x/linker_scripts/gcc/flash.ld" "-Wl,-Map,C:\Users\diver\AppData\Local\Temp\arduino_build_512166/Marlin.ino.map"  -o "C:\Users\diver\AppData\Local\Temp\arduino_build_512166/Marlin.ino.elf" "-LC:\Users\diver\AppData\Local\Temp\arduino_build_512166" -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--start-group -u _sbrk -u link -u _close -u _fstat -u _isatty -u _lseek -u _read -u _write -u _exit -u kill -u _getpid "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\Marlin.ino.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\HAL_LCD_I2C_routines.c.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\HAL_LCD_pin_routines.c.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\Marlin.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\I2cEeprom.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\SpiEeprom.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\servo.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_AVR\HAL_AVR.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_AVR\HAL_spi_AVR.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_AVR\MarlinSerial.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_AVR\persistent_store_impl.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_AVR\servo_AVR.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_AVR\watchdog_AVR.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_DUE\HAL_Due.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_DUE\HAL_spi_Due.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_DUE\HAL_timers_Due.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_DUE\InterruptVectors_Due.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_DUE\Servo_Due.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_DUE\persistent_store_impl.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_DUE\watchdog_Due.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\HAL.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\HAL_spi.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\HAL_timers.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\HardwareSerial.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\LPC1768_PWM.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\LPC1768_Servo.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\SoftwareSerial.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\WInterrupts.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\arduino.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\main.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\persistent_store_impl.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\pinmapping.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\u8g_com_HAL_LPC1768_hw_spi.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\u8g_com_HAL_LPC1768_sw_spi.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_LPC1768\watchdog.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_STM32F1\HAL_Servo_Stm32f1.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_STM32F1\HAL_Stm32f1.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_STM32F1\HAL_spi_Stm32f1.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_STM32F1\HAL_timers_Stm32f1.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_STM32F1\persistent_store_impl.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_STM32F1\watchdog_Stm32f1.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_TEENSY35_36\HAL_Servo_Teensy.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_TEENSY35_36\HAL_Teensy.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_TEENSY35_36\HAL_spi_Teensy.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_TEENSY35_36\HAL_timers_Teensy.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_TEENSY35_36\persistent_store_impl.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\HAL\HAL_TEENSY35_36\watchdog_Teensy.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\core\serial.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\core\utility.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\I2CPositionEncoder.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\Max7219_Debug_LEDs.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\baricuda.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\caselight.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\controllerfan.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\fanmux.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\filwidth.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\fwretract.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\mixing.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\pause.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\runout.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\snmm.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\solenoid.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\tmc2130.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\twibus.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\bedlevel\bedlevel.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\bedlevel\abl\abl.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\bedlevel\ubl\G26_Mesh_Validation_Tool.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\bedlevel\ubl\ubl.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\bedlevel\ubl\ubl_G29.cpp.o" "C:\Users\diver\AppData\Local\Temp\arduino_build_512166\sketch\src\feature\bedlevel\ubl\ubl_motion.cpp.o" " [... cut ...] end-group -lm -gcc
"U8glib.h" 对应多个库
 已使用： D:\文档\Arduino\libraries\U8glib
 未使用：C:\Program Files (x86)\Arduino\libraries\U8glib
使用 1.19.1  版本的库 U8glib 在文件夹： D:\文档\Arduino\libraries\U8glib 
使用 1.0  版本的库 Wire 在文件夹： C:\Users\diver\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.10\libraries\Wire 
fork/exec C:\Users\diver\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc.exe: The filename or extension is too long.

How can I fix this problem?

thinkyhead, 2017-12-03:
See this https://github.com/arduino/Arduino/issues/6751#issuecomment-331879017 and this #7707

teemuatlut, 2017-12-03:
Either use Arduino IDE beta version 1.90 or [edit your IDE's build path](https://github.com/MarlinFirmware/Marlin/issues/7076#issuecomment-345670043).

diver197, 2017-12-03:
The problem has been dealt with, remove the IDE's preferences.txt file line 54 # to build.path = F: \ tmp
thank you all

Roxy-3D, 2017-12-03:
I believe the beta or latest version of Arduino can handle the long file names.   Alternatively, maybe in the #7076 thread or #3851 somebody gave directions to make the Arduino temporary directory a much shorter name.   Please try those suggestions...

closed 2018-02-10 by thinkyhead

josocjo, 2020-04-25:
same problem

thinkyhead, 2020-04-25:
@josocjo — It is now recommended to use PlatformIO with the Auto Build Marlin extension. https://marlinfw.org/docs/basics/install_platformio.html

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y