MarlinFirmware/Marlin #8639: Marlin 2.x Compile error

pbrgle1, 2017-12-03:
I'm getting the following compile error with marlin 2.x

Arduino: 1.8.5 (Windows 10), Board: "Arduino Due (Programming Port)"

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp: In function 'void store_rxd_char()':

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:201:29: error: 'struct Usart' has no member named 'UART_RHR'

   const uint8_t c = HWUART->UART_RHR;

                             ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp: In function 'void UART_ISR()':

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:300:29: error: 'struct Usart' has no member named 'UART_SR'

   uint32_t status = HWUART->UART_SR;

                             ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:315:13: error: 'struct Usart' has no member named 'UART_CR'

     HWUART->UART_CR = UART_CR_RSTSTA;

             ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp: In static member function 'static void MarlinSerial::begin(long int)':

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:333:11: error: 'struct Usart' has no member named 'UART_PTCR'

   HWUART->UART_PTCR = UART_PTCR_RXTDIS | UART_PTCR_TXTDIS;

           ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:336:11: error: 'struct Usart' has no member named 'UART_CR'

   HWUART->UART_CR = UART_CR_RSTRX | UART_CR_RSTTX | UART_CR_RXDIS | UART_CR_TXDIS;

           ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:339:11: error: 'struct Usart' has no member named 'UART_MR'

   HWUART->UART_MR = UART_MR_CHMODE_NORMAL | US_MR_CHRL_8_BIT | US_MR_NBSTOP_1_BIT | UART_MR_PAR_NO;

           ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:342:11: error: 'struct Usart' has no member named 'UART_BRGR'

   HWUART->UART_BRGR = (SystemCoreClock / (baud_setting << 4));

           ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:345:11: error: 'struct Usart' has no member named 'UART_IDR'

   HWUART->UART_IDR = 0xFFFFFFFF;

           ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:346:11: error: 'struct Usart' has no member named 'UART_IER'

   HWUART->UART_IER = UART_IER_RXRDY | UART_IER_OVRE | UART_IER_FRAME;

           ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:355:11: error: 'struct Usart' has no member named 'UART_CR'

   HWUART->UART_CR = UART_CR_RXEN | UART_CR_TXEN;

           ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp: In static member function 'static void MarlinSerial::checkRx()':

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:370:15: error: 'struct Usart' has no member named 'UART_SR'

   if (HWUART->UART_SR & UART_SR_RXRDY) {

               ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp: In static member function 'static void MarlinSerial::writeNoHandshake(uint8_t)':

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:532:22: error: 'struct Usart' has no member named 'UART_SR'

     while (!(HWUART->UART_SR & UART_SR_TXRDY)) { sw_barrier(); };

                      ^

sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp:533:13: error: 'struct Usart' has no member named 'UART_THR'

     HWUART->UART_THR = c;

             ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1524888/Configuration.zip)

cjsoong, 2017-12-04:
MarlinSerial_Due.cpp

// Based on selected port, use the proper configuration
#if SERIAL_PORT == 0
  #define HWUART UART
  #define HWUART_IRQ UART_IRQn
  #define HWUART_IRQ_ID ID_UART
//#elif SERIAL_PORT == 0
//  #define HWUART USART0
//  #define HWUART_IRQ USART0_IRQn
//  #define HWUART_IRQ_ID ID_USART0
#elif SERIAL_PORT == 1
  #define HWUART USART1
  #define HWUART_IRQ USART1_IRQn
  #define HWUART_IRQ_ID ID_USART1

pbrgle1, 2017-12-04:
so are you saying i should choose a different serial port?

cjsoong, 2017-12-04:
#define SERIAL_PORT 0.....compile ok & uart connect ok 
#define SERIAL_PORT -1.....compile ok & uart connect fail

diver197, 2017-12-05:
I compile also appeared the same problem, do not know how to deal with

cjsoong, 2017-12-05:
Today I re-test and found that only need to set SERIAL_PORT -1, connected to the Arduino DUE's "NATIVE USB PORT" ....... compile pass & PC online ok. (But does not show a connection message, you need Execute "Gcode command" to confirm)

Solution:
1.#define SERIAL_PORT -1
2. PC USB <--> Arduino DUE  " NATIVE USB PORT"

thinkyhead, 2018-02-10:
Presuming solved.

closed 2018-02-10 by thinkyhead