MarlinFirmware/Marlin #8641: [Regression] `G29 J` fails after boot since commits of 2 Dec.

bjarchi, 2017-12-03:
### Bug Report
- Description: ...

I've run into a strange bug with the `G29 L[1]; G29 J` sequence since pulling down recent bugfix-1.1.x commits yesterday. In summary, I generally use `G29 L1; G29 J` in my startup code to load a dense mesh and then correct for bed tilt. This has worked perfectly since bringing up UBL a couple weeks back. 
Since applying the commits of the past few days (week maybe?) this no longer works after a cold system boot. If I load a mesh - probed using the same firmware build - and then issue `G29 J` the system probes the first two points, and then after probing the second point it emits the following error:
```
Recv: ? in get_z_correction(rx0=241.00, ry0=28.00)
```
And starts a move toward Z+ (bed down) that doesn't stop until the controller is reset (or maybe it hits the soft endstop).
I noticed this yesterday, but didn't report it since it stopped happening after I used `G29 P1` to re-probe my dense mesh (and then saved the completed mesh). However after restarting my machine today (it was off overnight) the error has returned. Most likely it will go away again after `G29 P1` until the next power cycle. 

From the error, it looks like the system is trying to get a correction value for X=241, which is outside of the mesh bounds - but that is not the X value of my 3rd triangle probe point; I have the points defined as below. However my probe x-offset is -71, so 241 would be the *nozzle* coordinate of the 3rd probe point...

```
  #define UBL_PROBE_PT_1_X 30       // Probing points for 3-Point leveling of the mesh
  #define UBL_PROBE_PT_1_Y 170
  #define UBL_PROBE_PT_2_X 30
  #define UBL_PROBE_PT_2_Y 30
  #define UBL_PROBE_PT_3_X 170
  #define UBL_PROBE_PT_3_Y 30
```

Any help would be appreciated; this is probably above my pay grade in terms of trying to find a fix (or even know where to start looking). 

- Expected behaviour: `G29 L1; G29 J` should load a stored mesh, and do a 3-point probe to 'tilt' the loaded mesh according to differences in bed position from the original probing. 
- Actual behaviour: `G29 J` fails after the second point is probed, with the error shown above.
- Steps to reproduce:
  - Seen after fresh build of yesterday's tree, with `M502; M500; M501` to initialize EEPROM.
  - `G29 P1; G29 P3; G29 S1` to probe a new mesh and save it in slot 1
  - `G29 L1; G29 J` to load mesh 1 and do 3-point probe - this should work
  - Power cycle the machine (or, probably, just reset the controller?)
  - `G29 L1; G29 J` again - this time it should fail

Note that I saved the same (newly probed) mesh in slot 2, and `G29 L2; G29 J` fails identically. 

Output of `G29 W` (after reset)
```
Send: G29 W
Recv: echo:Home XYZ first
Recv: Unified Bed Leveling System v1.01 active.
Recv: Mesh 1 Loaded.
Recv: UBL object count: 1
Recv: planner.z_fade_height : 10.0000
Recv: # of samples: 49
Recv: Mean Mesh Height: -0.060185
Recv: Standard Deviation: 0.091768
Recv: zprobe_zoffset: -1.7599999
Recv: MESH_MIN_X  5=5
Recv: MESH_MIN_Y  5=5
Recv: MESH_MAX_X  214 - (5)=209
Recv: MESH_MAX_Y  186 - (5)=181
Recv: GRID_MAX_POINTS_X  7
Recv: GRID_MAX_POINTS_Y  7
Recv: MESH_X_DIST  34.00
Recv: MESH_Y_DIST  29.33
Recv: X-Axis Mesh Points at: 5.000  39.000  73.000  107.000  141.000  175.000  209.000
Recv: Y-Axis Mesh Points at: 5.000  34.333  63.667  93.000  122.333  151.667  181.000
Recv: Kill pin on :41  state:1
Recv: 
Recv: ubl_state_at_invocation :0
Recv: 
Recv: ubl_state_recursion_chk :0
Recv: 
Recv: Meshes go from 0x02F0 to 0x0F7F
Recv: sizeof(ubl) :  1
Recv: 
Recv: z_value[][] size: 196
Recv: 
Recv: EEPROM free for UBL: 0x0C8F
Recv: EEPROM can hold 16 meshes.
Recv: 
Recv: Unified Bed Leveling sanity checks passed.
Recv: ok
```

[config-20171202.zip](https://github.com/MarlinFirmware/Marlin/files/1525199/config-20171202.zip)

bjarchi, 2017-12-04:
Quick update: I've applied the most recent commits and re-flashed, and `G29 L[n]; G29 J` is still failing with the same error. I now have level debugging built in, so I will be able to provide more detailed debug information soon.

@Roxy-3D - Any thoughts on whether this is an issue with UBL or another issue with stored settings?

Roxy-3D, 2017-12-04:
I don't know the answer...   But I have a branch with G29 J improvements here: https://github.com/Roxy-3D/Fast_G26_work

It has the 3 point probing using the normal grid based Least Squares Fit algorithm.   It has debug code (that is currently commented out) that proves the sampled points are in the tilted plane (assuming you are tilting a G29 P0 mesh).    I'm trying to decide how to handle the home point.  It seems to me the Z value for the home location should be 0.000 mm.    But by doing that the 3 points are shifted out of the plane.   I'm thinking a user controlled option is the best way to handle that.

My suggestion is download this branch and load it into your controller board.    This branch was forked before any of the Planner:: changes happened.  So, if the root of the problem is in those changes, things will start working again for you.      And for sure...   I trust the 3-Point leveling in that branch.  I've done a fair amount of analysis to make sure it was working right.    The one outstanding item is I want it to be able to repetive 3-point (or grid) tilting and have the results converge.   Right now, the results do not converge.   But a single 3-point tilt **_DOES_** do the 'right' thing.   (Uncomment the verification code starting at roughly line 1525 and generate a zero mesh with G29 P0.   When you tilt it using G29 J it **_DOES_** do the right thing!)

bjarchi, 2017-12-04:
OK, I will try out your branch when I'm able to - probably later today - and report back with the results. Is there any specific debug logging that would help track down this bug while I have bugfix-1.1.x/HEAD installed on my controller? 

I'm wondering if it might be another issue with persistent storage - maybe some UBL constant incorrectly saved - since the problem goes away after using `G29 P1` to re-build a mesh and presumably re-initialize some of the UBL system... but I'm not sure I have the skill or familiarity to debug that yet. 

I guess it might be time to make friends with `git bisect` and see if I can track down the exact commit that introduced the regression, too :)

bjarchi, 2017-12-05:
@Roxy-3D and @thinkyhead I was able to test out your (Roxy's) Fast_G26_work branch, and the `G29 J` is not present there, so we can be sure that it was introduced somewhere between where you forked that branch and HEAD (as of this morning)... looks like you forked at 82df656 if I'm reading the history correctly.

Do you have any specific tips for tracking down the cause of this bug? My approach for now will be to start looking at debug logs for the leveling system - which might help - and using git bisect to try to track down the commit that introduced the bug... but I suspect there are a lot of commits over the past 8 days so that may take a while (I've never actually used bisect before). Any tips for more targeted debugging would be helpful. Seems like I may be the only one affected by this bug right now so hopefully I can track it down myself!

Roxy-3D, 2017-12-05:
Those changes to the Planner:: are very low level.   And UBL has its own motion routine to break up moves at mesh cell boundaries.   There is no telling what subtle variable got introduced.  

With that said...  I would argue any changes to the Planner:: should not break G29 P1; G29 P3; G29 S[n] or G29 P0; G29 J   !!!   

What ever changes are made should be tested and fully backward compatible.

bjarchi, 2017-12-05:
> With that said... I would argue any changes to the Planner:: should not break G29 P1; G29 P3; G29 S[n] or G29 P0; G29 J !!!

Agreed that they shouldn't break anything in UBL... and I'm still not completely sure they did, as opposed to it being another bug in how things are persisted/loaded/initialized on startup, like that DELTA bug. 

I haven't seen breakage of `G29 P1; G29 P3; G29 S[n]` - they always seem to run fine - the only thing that consistently breaks is `G29 L[n]; G29 J` and (so far) only when there has been a controller reset since the last execution of `G29 P1; G29 P3...`. It's very strange. 

I haven't tried `G29 P0; G29 J` after cold boot, but I'll put that on my debug list. Unfortunately the sequence that I want to have work is `G29 L1; G29 J`. I suppose I can just omit the `G29 J` when printing, but I worry about corruption of the leveling system while printing in that case. 

2 updates: 
* `G29 J2` breaks similarly to `G29 J` - I thought maybe having UBL determine the probe points instead of using the compiled-in triangle might help, but it throws errors too - this time on two of the four points, but it seemed to finish probing (didn't go to the endless z+ move). Log snippet (no debug on this one):
```
READ: ? in get_z_correction(rx0=6.00, ry0=180.00)
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: ? in get_z_correction(rx0=178.00, ry0=180.00)
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: ? in get_z_correction(rx0=249.00, ry0=4.00)
READ: ok
```

* I was able to capture logs of `G29 J` both working and failing with DEBUG_LEVEL activated, and other than slight z-coord differences due to the new mesh they look **exactly the same** up to the point where the error occurs. Logs - failure followed by success; note that all the shared x & y coordinates are the same:

```
SENT: G29 J
READ: >>> probe_pt(30.00, 170.00, no stow)
READ:   current_position=(101.00, 87.00, 11.76) : 
READ: >>> do_blocking_move_to(101.00, 168.00, 11.76)
READ: <<< do_blocking_move_to
READ:   current_position=(101.00, 168.00, 11.76) : set_probe_deployed
READ: deploy: 1
READ: do_probe_raise(10.00)
READ: >>> do_blocking_move_to(101.00, 168.00, 11.76)
READ: <<< do_blocking_move_to
READ:   current_position=(101.00, 168.00, 11.76) : >>> run_z_probe
READ:   current_position=(101.00, 168.00, 11.76) : >>> do_probe_move
READ: set_bltouch_deployed(1)
READ: >>> do_blocking_move_to(101.00, 168.00, -10.00)
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: set_bltouch_deployed(0)
READ:   current_position=(101.00, 168.00, 1.81) : sync_plan_position
READ:   current_position=(101.00, 168.00, 1.81) : <<< do_probe_move
READ:   current_position=(101.00, 168.00, 1.81) : <<< run_z_probe
READ: >>> do_blocking_move_to(101.00, 168.00, 7.81)
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: <<< probe_pt
READ: >>> probe_pt(30.00, 30.00, no stow)
READ:   current_position=(101.00, 168.00, 7.81) : 
READ: >>> do_blocking_move_to(101.00, 28.00, 7.81)
READ: <<< do_blocking_move_to
READ:   current_position=(101.00, 28.00, 7.81) : set_probe_deployed
READ: deploy: 1
READ:   current_position=(101.00, 28.00, 7.81) : >>> run_z_probe
READ:   current_position=(101.00, 28.00, 7.81) : >>> do_probe_move
READ: set_bltouch_deployed(1)
READ: >>> do_blocking_move_to(101.00, 28.00, -10.00)
READ: echo:busy: processing
busy: processing
[... 130 lines cut ...]
READ: before rotation = [107.0000000,151.0000000,-0.0559081]   ---> after rotation = [107.0000000,151.0000152,-0.0230932]
READ: before rotation = [107.0000000,180.0000000,-0.1245208]   ---> after rotation = [107.0000076,180.0000152,-0.0863969]
READ: before rotation = [140.6666717,6.0000000,-0.0733731]   ---> after rotation = [140.6666717,6.0000133,-0.0654761]
READ: before rotation = [140.6666717,35.0000000,-0.0322055]   ---> after rotation = [140.6666717,35.0000076,-0.0189994]
READ: before rotation = [140.6666717,64.0000000,0.0077147]   ---> after rotation = [140.6666717,64.0000000,0.0262298]
READ: before rotation = [140.6666717,93.0000000,0.0164472]   ---> after rotation = [140.6666717,93.0000000,0.0402712]
READ: before rotation = [140.6666717,122.0000000,0.0014771]   ---> after rotation = [140.6666717,122.0000000,0.0306102]
READ: before rotation = [140.6666717,151.0000000,-0.0783632]   ---> after rotation = [140.6666717,151.0000152,-0.0439211]
READ: before rotation = [140.6666717,180.0000000,-0.0783632]   ---> after rotation = [140.6666717,180.0000152,-0.0386121]
READ: before rotation = [174.3333435,6.0000000,-0.1008183]   ---> after rotation = [174.3333435,6.0000185,-0.0912941]
READ: before rotation = [174.3333435,35.0000000,-0.0259680]   ---> after rotation = [174.3333435,35.0000038,-0.0111348]
READ: before rotation = [174.3333435,64.0000000,-0.0783632]   ---> after rotation = [174.3333435,64.0000152,-0.0582210]
READ: before rotation = [174.3333435,93.0000000,-0.0396905]   ---> after rotation = [174.3333435,93.0000076,-0.0142393]
READ: before rotation = [174.3333435,122.0000000,-0.0671356]   ---> after rotation = [174.3333435,122.0000152,-0.0363754]
READ: before rotation = [174.3333435,151.0000000,0.0114572]   ---> after rotation = [174.3333435,151.0000000,0.0475264]
[... cut ...]
READ: before rotation = [208.0000000,64.0000000,-0.0783632]   ---> after rotation = [208.0000000,64.0000152,-0.0565938]
READ: before rotation = [208.0000000,93.0000000,-0.0396905]   ---> after rotation = [208.0000000,93.0000076,-0.0126121]
READ: before rotation = [208.0000000,122.0000000,-0.0671356]   ---> after rotation = [208.0000000,122.0000152,-0.0347482]
READ: before rotation = [208.0000000,151.0000000,0.1012776]   ---> after rotation = [208.0000000,150.9999847,0.1389740]
READ: before rotation = [208.0000000,180.0000000,-0.0384428]   ---> after rotation = [208.0000000,180.0000000,0.0045625]
READ: ok
```

bjarchi, 2017-12-05:
I guess `git bisect` is the next step to see if I can track down the commit, but it will be time consuming. I need to figure out a way to stash and pop my configuration changes between bisect iterations, since I need to actually flash and test on hardware.

Roxy-3D, 2017-12-05:
My guess is these lines are the culprit:  

-  if (moves_queued > 1 && !UNEAR_ZERO(previous_nominal_speed)) {
+  if (moves_queued && !UNEAR_ZERO(previous_nominal_speed)) {

bjarchi, 2017-12-05:
Interesting... and easy to test. Do you have the commit SHA for that change?

bjarchi, 2017-12-05:
OK. I'm at a loss to explain this. `git bisect` fixed my bug.

The only thing I can propose is that the EEPROM bug I found yesterday introduced some kind of corruption in my EEPROM that wasn't fixed by the usual `M502; M500; M501` cycle, but was fixed when the EEPROM settings version was changed (v46->v45->v46) in the course of bisecting through commits. Maybe? 

Sorry to waste peoples' time on this. It really was acting like a reproducible bug, but I'm back at HEAD and I still can't trigger it. I'll do some more testing to be sure, at which point I'll close the report.

closed 2017-12-05 by thinkyhead