MarlinFirmware/Marlin #8642: Not an issue but a tip: Compiling Marlin with Arduino in Windows 10 ?

Spawn32, 2017-12-03:
Turn of Windows 10 real-time virus protection, will give you about 10 times the compiling speed.
Please remember to turn it on again when you are finished :)

To bad you can't make a real-time protection exception rule in Windows :/

JimBrown, 2017-12-04:
But you can make real-time exclusions in Windows Defender... https://www.tenforums.com/tutorials/5924-add-remove-windows-defender-exclusions-windows-10-a.html

Are you sure this is a thing though? I just tried it with Defender on and off. I did not see any appreciable difference in speed. I did clear the Arduino cache before each run. Is it possible you are seeing the speed increase due to the second run using cached objects?

Spawn32, 2017-12-04:
Did you try to turn the part called realtime protection off ? inside defender ? makes a hugh difference at least on the Insider releases, goes form taking 0.5 to 1 second for every compiler line to just flying through them..

Will have a look at that link, thanks :)

Spawn32, 2017-12-04:
added an exception for the arduino folder and also the arduino-builder process, diden't help, turn'd off realtime protection and it was flying again , typical Microsoft, LOL

Tannoo, 2017-12-04:
I don't have an issue with PlatformIO flying through the compilation process. Is this just an Arduino IDE issue?

JimBrown, 2017-12-04:
@Spawn32, yep, turned off realtime protection as you describe. Made no difference here. Typical Microsoft. ;-)

Spawn32, 2017-12-04:
@JimBrown Have a look her, no ider whats going on, but it's crazy :)

https://www.youtube.com/watch?v=ynwIXLhSONo&feature=youtu.be

fiveangle, 2017-12-09:
My 4yr old Macbook Pro compiles literally 8 times faster than my brand new , with Arduino and PlatformIO temp and working directories whitelisted.

Late 2013 1TB SSD-equipped 16GB DDR3 RAM MacBook Pro (22nm "Haswell/Crystalwell" 2.6 GHz Intel "Core i7" processor (4960HQ)):
```
AVR Memory Usage
----------------
Device: at90usb1286
Program:  116928 bytes (89.2% Full)
(.text + .data + .bootloader)
Data:       3973 bytes (48.5% Full)
(.data + .bss + .noinit)
========================= [SUCCESS] Took 10.21 seconds =========================
```

2 month old Dell Latitude 7280 with 512GB NVMe-equipped 16GB DDR4 RAM Dell Latitude with Symantec Endpoint Protection (14nm "Kaby Lake/Coffee Lake" 2.8 GHz Intel "Core i7-7600U" processor):
```
AVR Memory Usage
----------------
Device: at90usb1286
Program:  117014 bytes (89.3% Full)
(.text + .data + .bootloader)
Data:       3981 bytes (48.6% Full)
(.data + .bss + .noinit)
 [SUCCESS] Took 143.58 seconds 
```

To add insult to injury, it's quite odd that resulting binary compiled with PIO on Windows 10 system is slightly larger in both PROGMEM and SRAM usage to same compiled on Mac OS.

Not trying to start a platform war, just reporting facts * shrug *

I'd quote the Arduino, but Arduino won't even compile on Windows 10 w/o hacks (and I haven't re-installed the latest 1.9 beta yet) but I can tell you that it didn't perform any miracles.

-=dave

fiveangle, 2017-12-09:
I'll bet that if you load a Debian VM into Hyper-V and run PIO, it will run closer to the 11 second compile that it does on Mac OS (or faster !)

So the real tip sould be: don't use Windows to develop on with PlatformIO unless you're a masochist.

-=dave

TadyTheFish, 2018-05-17:
can i ask a noob question.. how do you compile Marli 2.0 with arduino ?

JimBrown, 2018-05-18:
With Arduino IDE version 1.9.0-Beta (or newer.)

TadyTheFish, 2018-05-18:
Does 1.9.0 need the teensyduino install? I'm asking because i's not supported... I want no compile for teensy 3.5
I was reading the latest release of teensyduino and it says that support was added for 1.9.0 beta but it expects 1.8.5

thinkyhead, 2018-05-22:
If you're feeling adventurous, try PlatformIO. It might be simpler in the long run. Instructions are posted at http://marlinfw.org/ for ARM-based builds.