MarlinFirmware/Marlin #8643: Nozzle fall down when finished

jowa01, 2017-12-03:
Nozzle fall down when finished.

When V1.1.6 is in use.
In V1.1.5 it's OK.

#define DISABLE_INACTIVE_Z is set to false.

bjarchi, 2017-12-04:
1. Is it still an issue in bugfix-1.1.x? 

2. There are separate defines you might want to check for disabling axes when not active (as opposed to when the machine is idle). Found in `Configuration.h`, look for the following block:
```
// Disables axis stepper immediately when it's not being used.
// WARNING: When motors turn off there is a chance of losing position accuracy!
#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false
```
And make sure that DISABLE_Z is false there too. I'm not sure whether DISABLE_INACTIVE_Z would override this setting; it's worth checking.

jowa01, 2017-12-04:
I tried it 3 times.
New Download V1.1.6, and all Settings new.
The settings for my issue with the nozzle:

In Configuration.h
731  #define DISABLE_X false
732  #define DISABLE_Y false
733  #define DISABLE_Z false

In Configuration_adv.h
371  // Default stepper release if idle. Set to 0 to deactivate.
372  // Steppers will shut down DEFAULT_STEPPER_DEACTIVE_TIME seconds after the last move when DISABLE_INACTIVE_? is true.
373  // Time can be set by M18 and M84.
374  #define DEFAULT_STEPPER_DEACTIVE_TIME 120
375  #define DISABLE_INACTIVE_X true
376  #define DISABLE_INACTIVE_Y true
377  #define DISABLE_INACTIVE_Z false  // set to false if the nozzle will fall down on your printed part when print has finished.
378  #define DISABLE_INACTIVE_E true

End of Gcode:
G92 E0
G1 E-1.4000 F1800
M104 S0  
M140 S0  
G1 Z85
G1 X85 Y85

When the printer has finished, all driver will be deactivated immediately.
So i think it's a bug-

AnHardt, 2017-12-04:
From SD or via USB?

revilor, 2017-12-05:
Could also be just a slicer setting like M84 in the end g-code.

Roxy-3D, 2017-12-05:
If the nozzle is falling down...  It kind of sounds like the motors are being turned off.

Maybe the 'right' answer is for your end time Gcode to move the nozzle to a 'safe' corner and then turn off the motors?

bjarchi, 2017-12-05:
> Could also be the slicer adding M84 at the end of the g-code.

@jowa01 you said your end gcode was this:
>End of Gcode:
G92 E0
G1 E-1.4000 F1800
M104 S0
M140 S0
G1 Z85
G1 X85 Y85

Is that what you told your slicer to do, or have you checked a sliced file itself? Might be worth looking to make sure some revision of the slicer didn't start adding M84 regardless of whether it's in your custom postamble.

jowa01, 2017-12-05:
Printed by sd card.
No safe corner on this little printer.
Gcode is from a sliced file itself.

This line in Configuration_adv.h was the Problem.
#define SD_FINISHED_RELEASECOMMAND "M84 X Y Z E"
I changed it to
#define SD_FINISHED_RELEASECOMMAND "M84 X Y E"
and now the nozzle stay forever on position.

So it's not a bug but a new feature in V1.1.6

Thanks a lot!

bjarchi, 2017-12-06:
So... seems like this is resolved and not a bug? If so, could you close the issue? 

Thanks.

closed 2017-12-08 by jowa01

thinkyhead, 2017-12-16:
> So it's not a bug but a new feature in V1.1.6

Nope. `SD_FINISHED_RELEASECOMMAND` has been in Marlin for many years, since "1.0" before it started getting real version numbers.