MarlinFirmware/Marlin #8644: [2.x] Possible compile error when LCD_SET_PROGRESS_MANUALLY is defined

GMagician, 2017-12-03:
'ultralcd_impl_hd44780.h':
`[cut]
   #if DISABLED(LCD_SET_PROGRESS_MANUALLY)
      const uint8_t progress_bar_percent = card.percentDone();
    #endif
    if (progress_bar_percent > 2 && (ELAPSED(millis(), progress_bar_ms + PROGRESS_BAR_MSG_TIME) || !lcd_status_message[0]))
      return lcd_draw_progress_bar(progress_bar_percent);
[cut]`

'ultralcd_impl_dog.h':
`[cut]
    #if DISABLED(LCD_SET_PROGRESS_MANUALLY)
      const uint8_t progress_bar_percent = card.percentDone();
    #endif

    if (progress_bar_percent > 1) {
   [cut]`

variable is not defined but used...

thinkyhead, 2017-12-05:
You can use three-backticks for multi-line plaintext or code. See https://guides.github.com/features/mastering-markdown/

thinkyhead, 2017-12-05:
Are you getting a real compile error or just scanning the code for logic errors?
```
#if DISABLED(LCD_SET_PROGRESS_MANUALLY)
  const uint8_t progress_bar_percent = card.percentDone();
#endif
```
When `LCD_SET_PROGRESS_MANUALLY` is disabled there is no global variable named `progress_bar_percent` so the code defines a local variable with that name.

GMagician, 2017-12-05:
Well, I was looking into code to know what was good to include and what wasn't useless.
I saw that part of code and I saw that it's true you don't have any 'progress_bar_percent' variable defined but in the same time you use it...
Check 'ultralcd_impl_hd44780.h'. After #ifdef #endif you have an if clause that uses 'progress_bar_percent' and that can't be possible since you don't have any variable with such name

```#if DISABLED(LCD_SET_PROGRESS_MANUALLY)
  const uint8_t progress_bar_percent = card.percentDone();
#endif
if (progress_bar_percent > 2 && (ELAPSED(millis(), progress_bar_ms + PROGRESS_BAR_MSG_TIME) || !lcd_status_message[0]))
  return lcd_draw_progress_bar(progress_bar_percent);
```

GMagician, 2017-12-05:
@thinkyhead sorry, I tryed to use markdown but it seems not to work as expected. Example uses 3 ' symbols ad start and 3 at end. If I do the same I got....normal text

GMagician, 2017-12-06:
@thinkyhead re-reading your post I now understood what you said...but trying to compile I really get an error

closed 2017-12-06 by GMagician