MarlinFirmware/Marlin #8645: Delta homing broken in latest bugfix-2.0.x

revilor, 2017-12-03:
After updating to the latest bugfix-2.0.x, homing my delta is no longer working. 

Start positions: X below endstop, Y and Z hitting endstop

When I issue G28, X moves up until the endstop is hit. Then there should be a 5mm re-bump, but instead the X carriage moves down about half the delta heigth, does the slow 5mm re-bump there, and because no endstop hits in the middle of nowhere, moves down again.

Looks like the 5mm bump is not interpreted as a relative move down from top position but as a move to absolute -5.

The homing debug info:
23:45:52.940 : Machine Type: Delta
23:45:52.940 : Probe: FIX_MOUNTED_PROBE
23:45:52.940 : Probe Offset X:0 Y:0 Z:-0.80 (Aligned With & Below Nozzle)
23:45:52.940 : Auto Bed Leveling: UBL (disabled)
23:45:52.940 : current_position=(0.00, 0.00, 0.00) : bracket_probe_move
23:45:52.940 : > endstops.enable(true)
23:45:52.940 : current_position=(0.00, 0.00, 0.00) : >>> home_delta
23:45:52.956 : current_position=(0.00, 0.00, 0.00) : sync_plan_position
23:45:52.956 : >>> homeaxis(X)
23:45:52.956 : Home 1 Fast:
23:45:52.956 : >>> do_homing_move(X, 600.00, 0.00)
23:45:52.956 : current_position=(0.00, 418.35, 418.35) : sync_plan_position
23:45:54.999 : <<< do_homing_move(X)
23:45:54.999 : Move Away:
23:45:54.999 : >>> do_homing_move(X, -5.00, 0.00)
23:45:54.999 : current_position=(0.00, 418.35, 418.35) : sync_plan_position
23:45:57.558 : start
23:45:57.558 : echo: External Reset

What looks strange here is

23:45:52.956 : current_position=(0.00, 418.35, 418.35) : sync_plan_position

with X=0.0.0 which I suppose should also be 418.35.

revilor, 2017-12-03:
I just had a look at motion.cpp, and in lin 980 there is

  // Tell the planner the axis is at 0
  current_position[axis] = 0;

I guess this should not be done for deltas?

HekTron802, 2017-12-04:
Perhaps this issue is related to  #8638, which resulted is a couple of reverts.

thinkyhead, 2017-12-04:
Retest with current code, which has been reverted. If it fixes the issue, that would be good news!

Tannoo, 2017-12-04:
I had issues as well. Homing was acting as if my probe was active most of the time and when homing was successful, ubl probing of the bed was not.
The latest revert has corrected it for me.

revilor, 2017-12-04:
I can also confirm that behavior is back to normal with latest bugfix-2.0.x (cff886ba120b57da82b54869b444f98f8acab94f). Thankx.

closed 2017-12-04 by revilor