MarlinFirmware/Marlin #8654: Not a issue but a question about thermal setting

budi67, 2017-12-04:
Hello

I have modified my CR10 with an Aero Titan and a 50mm blower.
But the blower is to strong, when it engages, he lowers the temperature of almost 8 ° and a few second later the printer stops.

It's there a regulation in modifying to leave the time in the nozzle to go back up in temperature ?

A big thanks
Budi67

Deneteus, 2017-12-04:
It will only do that if you have not insulated your heat block with ceramic heater block cotton. Too much air is hitting the heater block rather than past the tip of the nozzle and the part. [https://images-na.ssl-images-amazon.com/images/I/61-QZSRFP3L._SL1000_.jpg](url)

budi67, 2017-12-04:
Thanks for your answer Deneteus, your link don't work.
I go even for this cotton

Deneteus, 2017-12-04:
Check this thread out.

http://forums.reprap.org/read.php?1,656519

budi67, 2017-12-04:
Thank you, I am going to read this thread.

bjarchi, 2017-12-04:
While I agree with @Deneteus about the root cause here - you should insulate your heat block and/or get your cooling air directed off of the block and onto the part - there is also some firmware configuration that might help get you up and running. I assume when the printer stops it's doing so because of a thermal runaway error?

budi67, 2017-12-04:
I am really stupid, I have one e3dv6 and I had not even put the shell in silicone, it is now done and the printings it passes marvelously.
Without you, I would not really have thought of it.
I am also going to revise the airflow

A big thanks !!

bjarchi, 2017-12-09:
@budi67 Sounds like everything is sorted then? If so, can you close the issue since it's resolved?

closed 2017-12-10 by budi67