MarlinFirmware/Marlin #8656: I'm having a compiling error.

mclemens18, 2017-12-04:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`
This is my error im having with marlin 1.0.0 Im using this because Robo 3D uses this firmware 

fastio.h:45: error: pasting "DIO" and "-" does not give a valid preprocessing token

#define _WRITE(IO, v) do { if (&(DIO ## IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Bob-the-Kuhn, 2017-12-04:
It's trying to say that the pin number "-" is invalid.  Not a very descriptive error message.

Did you recently change something in a configuration file or a pins_YOUR_BOARD.h file?

mclemens18, 2017-12-04:
@Bob-the-Kuhn I did not it just randomly gives me that error. What i did do was desolder and replace the blown voltage regulator on the arduino. Could that be doing something?

Bob-the-Kuhn, 2017-12-04:
??  That's a compile time error so it won't be affected by the presence/absence of the Arduino.

mclemens18, 2017-12-04:
@Bob-the-Kuhn OK, what should i check even if i didnt change that?

Bob-the-Kuhn, 2017-12-04:
What happens if you just do a verify (the check mark icon) rather than an upload (the right arrow icon)?

mclemens18, 2017-12-04:
@bob-the-kuhn it just does the same thing.

Bob-the-Kuhn, 2017-12-04:
I think you have two different problems.

Your Marlin files are either corrupted or missing.  I hope you have a backup.  If not we could try to build you a version of the latest Marlin using your old Configuration.g, Configuration_adv.h and pins_YOUR_BOARD.h files.

Your Arduino board has/had a hardware issue.  Is there any signs of life in it?

mclemens18, 2017-12-04:
@Bob-the-Kuhn Everything is there. I changed all of the pins to the same thing (just randomized) and it uploaded fine.

Bob-the-Kuhn, 2017-12-05:
Sounds like the board is functional.  Good news!

Were you able to recover your configuration/setup?

mclemens18, 2017-12-05:
@Bob-the-Kuhn  All that stuff was never deleted. I have everything. Are you fimiliar with Marlin firmware. Robo 3d twisted it for their printers and im having pin error

Bob-the-Kuhn, 2017-12-05:
I'm very familiar with the standard Marlin.

I think I'm missing something and/or making a wrong assumption.

You shouldn't be having a pin error if you're using the original Robo 3d files exclusively.  If it compiled cleanly before it should still compile cleanly.

Are you trying to convert from Robo 3d to the current Marlin?

I suggest ZIPing up everything and then dropping it on your reply.  That way I'll be on the same page as you.

Please include a copy of the Arduino error log in the ZIP file.

thinkyhead, 2017-12-05:
We don't really support Marlin 1.0.0 at this point.

The compile error is because you have a bad pin definition.

If you ZIP up and drop your whole Marlin folder, maybe we can get you past the compile error.

mclemens18, 2017-12-05:
Ok, it’s a really crappy and un Stable version and robo doesn’t seem to want to change it but I’ll send it to you when i have a minute. Because sometimes the max pin for the z is set to -1 and a couple are set to 19 and another is set to 18. The problem I’m having is just the randomized PIN numbers

mclemens18, 2017-12-05:
@Bob-the-Kuhn The only problem is that the pins are set to the wrong pinnumber. Some Z max is set to -1 and the same pin max is set to 19 on others. Thats the only problem

Bob-the-Kuhn, 2017-12-05:
Z max isn't used in a lot of printers.  

When you home the Z axis, does it go down to the minimum height or up to the maximum height?

If it goes to the minimum height just set Z max to -1.  That's how you tell the machine that Z max isn't used.

mclemens18, 2017-12-06:
Ok so change it all to -1? and There is also a problem with X max too which is a bunch of different values. What should i change that too?

mclemens18, 2017-12-06:
Arduino: 1.8.5 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch/Marlin.h:23:0,
                 from sketch/BlinkM.cpp:5:
sketch/pins.h:505:0: warning: "CONTROLLERFAN_PIN" redefined
     #define CONTROLLERFAN_PIN  10 //Pin used for the fan to cool controller
 ^
In file included from sketch/Configuration.h:709:0,
                 from sketch/Marlin.h:22,
                 from sketch/BlinkM.cpp:5:
sketch/Configuration_adv.h:64:0: note: this is the location of the previous definition
 #define CONTROLLERFAN_PIN -1 //Pin used for the fan to cool controller (-1 to disable)
 ^
In file included from sketch/Marlin.h:23:0,
                 from sketch/ConfigurationStore.cpp:1:
sketch/pins.h:505:0: warning: "CONTROLLERFAN_PIN" redefined
     #define CONTROLLERFAN_PIN  10 //Pin used for the fan to cool controller
 ^
In file included from sketch/Configuration.h:709:0,
                 from sketch/Marlin.h:22,
                 from sketch/ConfigurationStore.cpp:1:
sketch/Configuration_adv.h:64:0: note: this is the location of the previous definition
 #define CONTROLLERFAN_PIN -1 //Pin used for the fan to cool controller (-1 to disable)
 ^
In file included from sketch/Marlin.h:23:0,
                 from sketch/MarlinSerial.cpp:23:
sketch/pins.h:505:0: warning: "CONTROLLERFAN_PIN" redefined
     #define CONTROLLERFAN_PIN  10 //Pin used for the fan to cool controller
 ^
In file included from sketch/Configuration.h:709:0,
                 from sketch/Marlin.h:22,
                 from sketch/MarlinSerial.cpp:23:
sketch/Configuration_adv.h:64:0: note: this is the location of the previous definition
 #define CONTROLLERFAN_PIN -1 //Pin used for the fan to cool controller (-1 to disable)
 ^
In file included from sketch/Marlin.h:23:0,
                 from sketch/Marlin_main.cpp:30:
sketch/pins.h:505:0: warning: "CONTROLLERFAN_PIN" redefined
     #define CONTROLLERFAN_PIN  10 //Pin used for the fan to cool controller
 ^
In file included from sketch/Configuration.h:709:0,
                 from sketch/Marlin.h:22,
                 from sketch/Marlin_main.cpp:30:
sketch/Configuration_adv.h:64:0: note: this is the location of the previous definition
 #define CONTROLLERFAN_PIN -1 //Pin used for the fan to cool controller (-1 to disable)
 ^
sketch/Marlin_main.cpp:2192:36: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
         LCD_MESSAGEPGM(MACHINE_NAME" "MSG_OFF".");
                                    ^
sketch/Marlin_main.cpp: In function 'void loop()':
sketch/Marlin_main.cpp:588:49: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
                  card.openFile("robo~1.gco",true);
                                                 ^
sketch/Marlin_main.cpp: In function 'void set_bed_level_equation_lsq(double*)':
sketch/Marlin_main.cpp:884:36: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     planeNormal.debug("planeNormal");
                                    ^
In file included from sketch/Marlin.h:23:0,
                 from sketch/thermistortables.h:4,
                 from sketch/Configuration.h:710,
                 from /Users/iLab/Downloads/ROBO3DR1PLUSV2/ROBO3DR1PLUSV2.ino:33:
sketch/pins.h:505:0: warning: "CONTROLLERFAN_PIN" redefined
     #define CONTROLLERFAN_PIN  10 //Pin used for the fan to cool controller
 ^
In file included from sketch/Configuration.h:709:0,
                 from /Users/iLab/Downloads/ROBO3DR1PLUSV2/ROBO3DR1PLUSV2.ino:33:
sketch/Configuration_adv.h:64:0: note: this is the location of the previous definition
 #define CONTROLLERFAN_PIN -1 //Pin used for the fan to cool controller (-1 to disable)
 ^
In file included from sketch/Marlin.h:23:0,
[... 399 lines cut ...]
               ^
pins.h:557: error: unable to find numeric literal operator 'operator""_PIN'
     #define HEATER_BED_PIN     -1    // NO BED
                                 ^
sketch/fastio.h:17:26: note: in definition of macro 'MASK'
 #define MASK(PIN)  (1 << PIN)
                          ^
sketch/fastio.h:45:105: note: in expansion of macro '_WRITE_NC'
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                                                                                         ^
sketch/fastio.h:68:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
sketch/temperature.cpp:591:9: note: in expansion of macro 'WRITE'
         WRITE(HEATER_BED_PIN,LOW);
         ^
sketch/temperature.cpp:591:15: note: in expansion of macro 'HEATER_BED_PIN'
         WRITE(HEATER_BED_PIN,LOW);
               ^
exit status 1
pasting "DIO" and "-" does not give a valid preprocessing token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mclemens18, 2017-12-06:
@Bob-the-Kuhn This is the error

bjarchi, 2017-12-06:
> In file included from sketch/Marlin.h:23:0,
from sketch/SdVolume.cpp:20:
sketch/pins.h:505:0: warning: "CONTROLLERFAN_PIN" redefined
#define CONTROLLERFAN_PIN 10 //Pin used for the fan to cool controller
^
In file included from sketch/Configuration.h:709:0,
from sketch/Marlin.h:22,
from sketch/SdVolume.cpp:20:
sketch/Configuration_adv.h:64:0: note: this is the location of the previous definition
#define CONTROLLERFAN_PIN -1 //Pin used for the fan to cool controller (-1 to disable)

It seems like you're trying to redefine a pin that was already specified in pins.h elsewhere (in Configuration_adv.h?) - which is both throwing warnings and somehow causing the -1 (not assigned) identifier to propagate through into compilation.

Is this trying to compile the source code distributed from Robo *as-is* without modifications? @Bob-the-Kuhn  is it possible this is an old enough fork that it needs to be compiled with an older Arduino release? I seem to recall that being an issue for some mfr's source archives in my readings.

mclemens18, 2017-12-06:
@bjarchi I changed all the Max pins to -1 because they arnt being used.

bjarchi, 2017-12-06:
>  I changed all the Max pins to -1 because they arnt being used.

That makes sense - assuming that syntax was supported in 1.0.x - as long as redefining those pins is supported in the Configuration_adv.h instead of directly in the boards/pins files. I was more worried about `CONTROLLERFAN_PIN` but I have a 1.0 era archive from my printer's mfr and that does seem to be supported. 

You might try removing the redefines for the Max pins in `Configuration_adv.h` and instead enabling this define in `Configuration.h`:
```
//#define DISABLE_MAX_ENDSTOPS
```
(uncomment it), assuming that define exists in the version you're using.

At this point it does seem like it would be useful to have your entire Marlin distribution zipped and attached to the thread... ideally both with the original configuration files (from Robo) as well as your modifications.

Bob-the-Kuhn, 2017-12-07:
**AMEN**

We really need those files if we're going to make progress.

Sometimes GitHub gits finicky about uploading files.  In that case i like to use a free service like DropBox and then post the link.

Exactly which Robo printer are you using?

mclemens18, 2017-12-07:
@bob-the-Kuhn I have fixed my issue, something went wrong. I reset everything and it all I stalled. Sorry for using your time but it magically uploaded and worked. Thank you for your help

bjarchi, 2017-12-07:
So... close?

mclemens18, 2017-12-07:
@bjarchi sorry for the lack of response. I have figured out the error. I don’t know what happened but I full another computer and uploaded and it worked.

closed 2017-12-08 by Bob-the-Kuhn