MarlinFirmware/Marlin #8657: Creality Voltage Regulator Burned Out

Deneteus, 2017-12-04:
With the Bugfix version of Marlin installed I have had 3 Creality printer boards where there have been Temperature Runaway errors and then the voltage regulator failed. The first time it happened I thought it was a just a faulty board but I found if you powered the board via ICSP port the board would boot. The screen lights up faintly because but you can see Marlin boot. 

This is how I identified the process. I had just printed 2 parts using PETG where the temp was between 240 to 255. I had noticed during the second print that the temp was set to 240 but it drifted to between 242 to 250. I figured this was because the PID settings needed to be modified.  I was using Astroprint and when I went to enter the GCODE in to PIDAUTOTUNE was when I r

Tannoo, 2017-12-04:
Are you positive there isn't a short somewhere?
I don't have your board, but all prints fine for me.

Tannoo, 2017-12-04:
I would say that the general setup for temp control is fine. Mine has no issues there.

Deneteus, 2017-12-04:
I have 3 different printers with the same issue. No shorts. I literally just got this one in the mail this week. The only change was firmware.

revilor, 2017-12-05:
If the temperature went up to 255 degrees or maybe above, the hot end thermistor might have died causing the thermal runaway. I don't know the specific thermistor Creality uses, but lots of the cheaper ones are specified for no more than 250 or 260 degrees.

I don't know if this is the case here, but the most likely cause to kill a 5V regulator is to power it with 24V. This would drive any 5V regulator I know far beyond it's specification, which for the supply voltage usually ends at 18V. The other way would be to draw more than the 1 to 2A of current the voltage regulator is able to provide. Thermal runaway on the hot end or not, I don't see this to happen for the 5V circuit in a 3D printer without a production of layout issue on the board itself.

Deneteus, 2017-12-05:
@revilor The PSU in the CR10 is a 12v but unfortunately the AMS1117-5.0v can be killed at +15V. I looked at the data sheets before I ordered a replacement because I believe it could have been a voltage spike from the power supply due to the fluctuating power draw from the bed+hot end. I ordered +18V regulators this time since they were not that much more than the 15v versions in bulk.

www.advanced-monolithic.com/pdf/ds1117.pdf

I am well aware of PTFE degrading due to temps at 250C if used over the period of a month. I learned about that from my experience with the Wanhao i3 Plus. On my other printers that is why I switched to the E3D V6 and the Microswiss CR10. I actually planned on swapping the stock PTFE lining with Capricorn XS which can survive up to 320C. 

I used this for reference before I made the decision to upgrade.
http://www.rjchase.com/ptfe_handbook.pdf

Deneteus, 2017-12-05:
This is what the screen looks like once the VR fails.

[image]

revilor, 2017-12-05:
@Deneteus I'm not talking about the PTFE tube inside the hot end which is another issue when printing at 250° and above. I'm talking about the temperature-dependent resistor or NTC placed on the heating block to measure the temperature of the nozzle. And looking at your screenshot I'm sure that the thermistors on your printers is broken because it can't handle more than 250°. Because when you check the nozzle temperature on the display it says -14° which is exactly the value you get if there is no thermistor connected. Or a broken one in your case. So that is the reason for the thermal runaways you got. Still no explanation for the blown voltage regulator, but if you plan to print at 250° or above in the future you should replace the thermistor with a model rated up to 290° or more like the Semitec 104GT2 used by E3D. Have a look at http://reprap.org/wiki/Thermistor to find a suitable one.

Deneteus, 2017-12-05:
@Revilor the focus of that picture is not the thermistor. That board is not even in the printer. You should be able to tell that because the LCD is not in its casing.

The first printer that it failed on with the CR-10S 2560 has a brand new M3 NTC 3950.
https://reprapchampion.com/pages/firmware-configuration-for-ntc-3950-100k-thermistor

My I3 and one other CR10 are currently using that same thermistor with non-stock hot ends. 

The second printer it failed on was a stock Hictop Creality with the stock thermistor. It also managed to kill the board from the first printer.

The third printer it failed on was also a stock Creality that I literally just received.

You aren't going to sit here and tell me that the thermistors were all faulty when I was able to get prints out without the temps failing.  Does this pic look like a print with a failing thermistor? The printer was idling at temp for about 15 min when it failed.

[image]

Tannoo, 2017-12-05:
That display picture looks like mine when I have too much current hogging the 5v regulator.
If you connect a PC / tablet / raspi / whatever via a USB cable (if you have a usb port on that board), does the display brighten?

Tannoo, 2017-12-05:
Try unplugging different things from the board.. could also have a bad or misajusted driver.

Tannoo, 2017-12-05:
Maybe just a bad pin assignment for your board?

If it worked with a previous version of Marlin, check that old pin file against the current one.

revilor, 2017-12-05:
@Deneteus Sorry for trying to be helpful. Won't happen again.

Deneteus, 2017-12-06:
Thanks @Tannoo  I thought the same thing as well so I tested using the first board that had failed with nothing else connected. When the 12V from the PSU is connected it would shut off. You could see the power come on and one of the leds flash and go back off again.

I'll check the pin assignments next. I kept my builds seperate so I'll do a compare and check it out. On the stock CR-10 Melzi builds I used the example files but the Pins I never changed. The only thing I changed was the LCD being used.

@revilor The stock Creality supplied thermistor is good to 270C. You also assumed I didn't read the manual and had no idea what could cause a Thermal Runaway normally. You only have yourself to blame. Next time you try to help someone, please make sure you do some research before throwing out incorrect information about someones hardware.

https://groups.google.com/forum/#!topic/wanhao-printer-3d/cnsSfpuED9o

Tannoo, 2017-12-06:
If it falls on it's face with nothing connected to 12v, is that board supposed to be power by 12v or 24v? I still think there is a hardware issue. If you can, check the current of the 12v supply with nothing connected but power. It should be next to nothing.

Another thought is to try an ATMega2560 / Ramps combo and see what that does for you. Works fine for me.

Deneteus, 2017-12-06:
Printer #1 | CR-10S 2560 PSU measures 14.3V and it should be 12V. Regulator would fail over 15V. Should be +/-2.5% @ 12.3V max. [image]

I turned it down to 12.3V and power cycled but nothing changed. 

Printer #2 | CR-10 Stock measures 12.4V. And when I was taking the picture I realized that now the display is working again.
. 
[image]

It is getting cold downstairs in the dining room so the temps right.

[image]

So at least one of the 3 boards is working now and it resolved itself LOL. The other 2 I won't be able to fix till the replacement regulators get here.

Tannoo, 2017-12-06:
Install a heatsink on the voltage regulator. Check the voltage of your 5v rail (endstop or something).

Deneteus, 2017-12-06:
@Tanoo There is already a heatsink on the voltage regulator on the CR10S board. I actually had to take the heatsink off of the stock board to find out what chip they used to get the replacement. As I said, in the beginning, the 5V Regulators are toast. How they got there because of an idle Temperature Runaway was the question.

What I am going to do is repair the boards and then hook up some multimeters and see if I can get it to happen again. because any issue you can consistently repeat is a defect. I'm not the only one that has had the issue in the past year otherwise I wouldn't have posted about it.

Tannoo, 2017-12-06:
Have you chosen a more robust regulator?

bjarchi, 2017-12-08:
I have a fair bit of EE experience, and this really feels like a hardware problem - chronic overcurrent on the 5V rail, poor board design causing intermittent short (possibly with thermal changes), bad batch of 5V regulators (esp. if they were marginally spec'd to begin with)... that kind of thing. 

Or possibly a problem with the board file, if that has changed recently. @Deneteus you say this started after you pulled bugfix-1.1.x earlier this year... how old was the firmware you were using before that? 

> Printer #1 | CR-10S 2560 PSU measures 14.3V and it should be 12V. Regulator would fail over 15V. Should be +/-2.5% @ 12.3V max.

I don't suppose all of these boards were at some point installed in printer 1? That could certainly be the failure cause right there. 

> So at least one of the 3 boards is working now and it resolved itself LOL. The other 2 I won't be able to fix till the replacement regulators get here.

Also suggests hardware issues as the most likely cause of failure. How long between when that board failed and it then resolved itself? Was it powered in between (either intermittently or continuously)?

Tannoo has made a couple of good suggestions for troubleshooting possible hardware issues - such as measuring the *current* the board is drawing from the 12V supply, disconnecting various connected devices, etc.

Roxy-3D, 2017-12-08:
I laughed when I saw this meter:  

https://github.com/MarlinFirmware/Marlin/issues/8657#issuecomment-349552325

Two years ago I was in school and dirt poor.   The only thing I could afford was a Harbor Freight volt meter.   I have that exact meter and it cost me $3.95 because I got it as a weekly special.   That meter never failed me.   And now I have 3 of those exact same meters.  (Except one of them I got for $2.95 because it didn't have the transistor testing socket on it.)

fiveangle, 2017-12-08:
I gave one of those meters to my 2yr old niece so she could touch the leads with her fingers and watch the display change, as that’s about all it’s good for 😂

-=dave

Tannoo, 2017-12-08:
> How they got there because of an idle Temperature Runaway was the question.

Well, if the 5v is dropping too far (usually due to overloading the regulator), then the thermo input would drop also...causing the fault.

bjarchi, 2017-12-08:
>> How they got there because of an idle Temperature Runaway was the question.

>Well, if the 5v is dropping too far (usually due to overloading the regulator), then the thermo input would drop also...causing the fault.

Ah yes that makes complete sense. I figured the temp runaway was related, if at all, due to the printer running near its temp limit, and thus at high power draw exposing a board / component fault. Your explanation is much more satisfying.

Deneteus, 2017-12-08:
@bjarchi The firmware that was on there before was the stock CR10 firmware.  If I remember correctly it's from 2016. I figured it was a bad batch of regulators as well so I ordered some better ones. Alot of cheap Melzi designs have design problems.

Actually, the first failure was in one of the 2 Hictops that were Amazon returns so I replaced the Hictop board with the one from my original OEM CR10 that I have had since September. The CR10S 2560 board was running a Bugfix build from early October and I just upgraded it after I got the other 2 printers so this all went down within a week. It was running fine before the upgrade. 

So right now 3 out of the 4 CR-10s  are working. I kept a couple stock so the prints would be consistent if I needed to break a model up.  The one that fixed itself was basically just sitting there for a couple days. The only thing that did change was the temperature of the room it was in. 

@Roxy-3D I got this meter for free with a coupon and I use it on all kinds of stuff. They give 2 away a year it seems. I have a couple of Fluke meters as well but they are on my bench.

bjarchi, 2017-12-11:
> The one that fixed itself was basically just sitting there for a couple days. The only thing that did change was the temperature of the room it was in.

The reason I asked about this is that often these boards are built with some kind of 'polyfuse' that acts like a fuse - opens - if it sees too much current (gets too hot?), but 'resets' some time after cooling down. It's possible that something overloaded the 5V bus, blowing the regulator, and also or subsequently tripped the polyfuse making the board seem dead until it sat unpowered for long enough. 

> So right now 3 out of the 4 CR-10s are working.

How? Did you replace regulators or did others spontaneously resurrect?

The consensus here seems to be that this was a hardware issue, and not directly caused by the change in firmware - or at least that we're not looking at a firmware bug. I think the relationship between the 5V reg. and the thermal runaway condition have also been explained - and if the previous firmware was really old it may not have had runaway detection. 

Are you satisfied with this explanation?

closed 2017-12-14 by thinkyhead

Deneteus, 2017-12-23:
@bjarchi I got the regulators in the mail but I haven't changed them out yet on the boards that aren't working. Creality sent me another board to replace the CR-10S board that failed and I just got it a day ago. It is a completely different board with V2.0 marking on it.

As far as the polyfuse explanation it could be but the other 2 boards are still non-operational. I just checked the one I am about to replace. I'm not done with this, holiday is slowing me down. I have a couple other projects I have to get done before I can get back to the boards that failed.

Deneteus, 2018-02-06:
FYI - https://www.th3dstudio.com/cr-10s-v2-0-board-temperature-issues/

thinkyhead, 2018-02-06:
I've got my capacitor at the ready. Just need to solder it in.

Deneteus, 2018-06-14:
@bjarchi @thinkyhead I got the new more robust regulator installed and it's back to working correctly now.

Sk7drive, 2018-06-15:
Hello, I think I have the same problem with a cr10, you could take pictures, I can not figure out where the voltage regulator is.
Thanks in advance

thinkyhead, 2018-06-15:
Can't explain it better than https://www.th3dstudio.com/cr-10s-v2-0-board-temperature-issues/