MarlinFirmware/Marlin #8669: Z_HOMING_HEIGHT ignored when Z_HOME_DIR > 0

marcio-ao, 2017-12-05:
- Description: Z_HOMING_HEIGHT ignored when Z_HOME_DIR > 0
- Expected behaviour: Z_HOMING_HEIGHT should be respected
- Actual behaviour: Homing just Y or X (G28 X0 or G29 Y0) ignores Z_HOMING_HEIGHT and may cause the toolhead to crash.
- Steps to reproduce:
  - Configure printer to home to the top 
  - Set a Z_HOMING_HEIGHT
  - Home all axis (G28)
  - Make nozzle touch print bed (G0 Z0)
  - Home just X or Y (G28 Y0 or G28 X0)

The reason for this bug lies in the following [code](https://github.com/MarlinFirmware/Marlin/blob/4d3c3d0b7384038e12c6eddb14bf717985b8dd7a/Marlin/src/gcode/calibrate/G28.cpp#L199):

```
    #if Z_HOME_DIR > 0  // If homing away from BED do Z first

      if (home_all || homeZ) {
        HOMEAXIS(Z);
        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (DEBUGGING(LEVELING)) DEBUG_POS("> HOMEAXIS(Z)", current_position);
        #endif
      }

    #else

      if (home_all || homeX || homeY) {
        // Raise Z before homing any other axes and z is not already high enough (never lower z)
        destination[Z_AXIS] = LOGICAL_Z_POSITION(Z_HOMING_HEIGHT);
        if (destination[Z_AXIS] > current_position[Z_AXIS]) {

          #if ENABLED(DEBUG_LEVELING_FEATURE)
            if (DEBUGGING(LEVELING))
              SERIAL_ECHOLNPAIR("Raise Z (before homing) to ", destination[Z_AXIS]);
          #endif

          LULZBOT_ADJUST_Z_HOMING_CURRENT(true)
          do_blocking_move_to_z(destination[Z_AXIS]);
          LULZBOT_ADJUST_Z_HOMING_CURRENT(false)
        }
      }

    #endif
```

The problem here is that the "#else" part needs to always execute so it can take care of the case in which only X or Y was being homed. The fix is to simply replace the #else with #endif and get rid of the trailing #endif.

GMagician, 2017-12-05:
That's sound strange. If you home to reverse direction you are moving far from bed when you do home. Why should you crash?

revilor, 2017-12-05:
Z is not in home position in the described scenario but at Z0. When homing X or Y in this case, Z should move up to Z_HOMING_HEIGHT first.

But from my understanding the code shown does exactly that.

revilor, 2017-12-05:
Ok, I got it. I took pre-processor #if and C-level if for the same. So the error is that the pre-processor #else should be a C-level else.

marcio-ao, 2017-12-05:
@revilor: Yes. Exactly.

@GMagician: You don't crash if you home all, but if you home X or Y (and not Z) then you can potentially crash. The reason this comes up is that Cura has home buttons for the individual axis. So a user could try to home just one of the axis while their nozzle is touching the bed. In that case, lifting to Z_HOMING_HEIGHT is crucial.