MarlinFirmware/Marlin #8670: G28 X0 Y0 disables legacy auto-leveling when not using AUTO_BED_LEVELING_UBL

marcio-ao, 2017-12-05:
- Description: Running G28 will cause auto-leveling to be disabled when AUTO_BED_LEVELING_UBL is not enabled.
- Expected behaviour: Running "G28 X0 Y0" should retain the autoleveling state.
- Actual behaviour: Running "G28 X0 Y0" will turn off autoleveling, unless using UBL
- Steps to reproduce:
  - Enable some form of auto-leveling that is not AUTO_BED_LEVELING_UBL
  - Do "G28" to home
  - Do "G29" to autolevel
  - Use "M420" to verify that autoleveling is enabled
  - Do "G28 X0 Y0" (after pausing a print, for example)
  - Use "M420" to see that autoleveling is disabled.

The cause of this bug is the check [here](https://github.com/MarlinFirmware/Marlin/blob/4d3c3d0b7384038e12c6eddb14bf717985b8dd7a/Marlin/src/gcode/calibrate/G28.cpp#L158) and [here](https://github.com/MarlinFirmware/Marlin/blob/4d3c3d0b7384038e12c6eddb14bf717985b8dd7a/Marlin/src/gcode/calibrate/G28.cpp#L311):

```
void GcodeSuite::G28(const bool always_home_all) {
  ...
  // Disable the leveling matrix before homing
  #if HAS_LEVELING
    #if ENABLED(AUTO_BED_LEVELING_UBL)
      const bool ubl_state_at_entry = planner.leveling_active;
    #endif
    set_bed_leveling_enabled(false);
  #endif

  ...

    #if ENABLED(AUTO_BED_LEVELING_UBL)
    set_bed_leveling_enabled(ubl_state_at_entry);
    #endif
}
```

The state of leveling is saved and restored only if AUTO_BED_LEVELING_UBL is enabled, but in truth it should be saved and restored for the legacy leveling modes as well.

thinkyhead, 2018-02-10:
`G28` is specified to always disable bed leveling. UBL marches to a different drummer.

Roxy-3D, 2018-02-14:
Just to be clear and make sure nobody jumps to the wrong conclusion....

With UBL enabled...  G28 still disables leveling during its operation.   It needs to do that so it can sense the machine's true position.    But after the homing operation, it restores the current bed leveling state (whether enabled or disabled).

marcio-ao, 2018-02-15:
@Roxy-3D: When not using UBL, the bed leveling state is not re-enabled, this was the nature of my ticket. I think it should be re-enabled for all leveling modes.

marcio-ao, 2018-02-15:
@thinkyhead : The reason we added re-homing after a pause is that users sometimes mess with their print heads during a pause and this was causing the head to shift and the resume to fail. I don't think it matters whether bed leveling is on *during* the G28, but re-enabling it after a "G28 X0 Y0" would be useful.

thinkyhead, 2018-02-23:
Some would obviously like to leave bed leveling on after `G28` if it was on before `G28`. Others, like myself, very much prefer that `G28` leave the machine at 0, 0 with leveling off. So I propose that this be added as an optional behavior, and that it be exposed as a `Cap:` capabilities string so that the host software can tell the difference between machines that do this and machines that don't.

I'm just proposing this as a config option that would make all `G28` restore leveling to its prior state for the given machine. I'm not proposing an extra `L` or `S` parameter to `G28` to apply this behavior, but I'd be open to that. While `G28 X Y\nM420 S1` accomplishes almost the same thing, it's either on or off, and there's no option to bring it back to its previous state.

If that's cool, I can implement it and apply it in `G28` for all forms of bed leveling. This would have the added benefit for fools like me to have UBL always be disabled after `G28` too.

closed 2018-02-23 by thinkyhead