MarlinFirmware/Marlin #8675: Anycubic Delta auto-level error "ZPROBE"

FlailAway, 2017-12-05:
Not sure if it is a bug, or something I have, or have-not done.

Anycubic-Linear-Plus, Delta used Anycubic's version of Marlin 1.1.0, dated 07-12-2017 but the auto-level never worked. 

I downloaded Marlin 1.1.6 (stable) and then using the Generic Delta example went through and changed the 1.1.6 configuration.h and configuration_adv.h to match the Anycubic 1.1.0 versions.

All look good, correct Z-height 297.4 and paper-clearance at Z0. Tried the Bed Level and it probed a bunch of spots that looked consistent and regular positions. After what looked like a uniform covering of probing, it stopped at the rear and raised to about Z-20. It then moved to what would be something like X-110/Y0 and moved toward X0. Allmost at the center of the bed, it declared "Err: ZPROBE" and stopped.

I cannot easily get the two configuration files to post here as they are on Ubuntu-PC for the printer at a friend's house. 

What or where should I start looking to fix this error?

Thanks

Swwils, 2017-12-10:
I have also had this error but it is most likely a bad config, I should be able to iron it out tomorrow and then I will PR a example config for the anycubic kossel with trigorilla board for use into bug fix.

laveur, 2018-01-26:
Is there any update on this?

thinkyhead, 2018-02-14:
If we can get some tuned AnyCubic configs, they would be most welcome.

Swwils, 2018-02-14:
My Linear Plus in the office is pretty close, but I have had it better before; will re-tune and send over. 

I believe the stock config from anycubic has several elements wrong... especially the pid values for temps.

maz3max, 2018-02-18:
this would be awesome!

boelle, 2019-02-19:
@FlailAway Please test with the latest `bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-06 by shitcreek