MarlinFirmware/Marlin #8676: LEVEL_BED_CORNERS menu not displayed with UBL enabled

bjarchi, 2017-12-05:
This is low-priority, and something I suspect I can fix, but I want to document it so there's an issue to reference in subsequent PR(s).

As a somewhat related question - I've committed at this point to running a bugfix release on my printer so I can help with finding and fixing bugs. From the standpoint of the developers, is it preferable for me to run bugfix-1.1.x or bugfix-2.0.x on an 8-bit controller? Is the 2.0.x bugfix branch sufficiently stable (and feature-matched) compared to 1.1.x to run on a machine I expect to use for printing things?

### Bug Report
- Description: LEVEL_BED_CORNERS is supposed to create an LCD menu item to easily visit the four bed corners to assist manual leveling of the bed. From reading through `ultralcd.cpp` it seems this menu item is not created when UBL is selected for automatic leveling - only when LCD_BED_LEVELING and/or a non-UBL system is selected for ABL. 
- Expected behaviour: defining `#LEVEL_BED_CORNERS` should cause the menu item to appear, even when compiled with UBL. 
- Actual behaviour: When compiled with the attached configuration, the corner leveling menu does not appear anywhere in the menu system. 
- Steps to reproduce:
  - Compile (current bugfix-1.1.x) with attached config files
  - Search for corner leveling menu

[config-20171205.zip](https://github.com/MarlinFirmware/Marlin/files/1532782/config-20171205.zip)

Roxy-3D, 2017-12-05:
> As a somewhat related question - I've committed at this point to running a bugfix release on my printer so I can help with finding and fixing bugs. From the standpoint of the developers, is it preferable for me to run bugfix-1.1.x or bugfix-2.0.x on an 8-bit controller? 

In theory...  either one should be fine for the 8-bit AVR's.    But I think you will find less problems for your development if you do the work in the bugfix-v1.1.x branch.   With that said, doing the work (and hitting problems) in the bugfix-v2.0.0 is valuable in its own right.

> Is the 2.0.x bugfix branch sufficiently stable (and feature-matched) compared to 1.1.x to run on a machine I expect to use for printing things?

Yes.   The functionality is identical.   If you find a difference, we need to know about it and get it fixed.

It looks to me like the bulk of the problem is here:
```
    //
    // Level Bed
    //
    #if ENABLED(AUTO_BED_LEVELING_UBL)
      MENU_ITEM(submenu, MSG_UBL_LEVEL_BED, _lcd_ubl_level_bed);
    #elif ENABLED(LCD_BED_LEVELING)
      #if ENABLED(PROBE_MANUALLY)
        if (!g29_in_progress)
      #endif
          MENU_ITEM(submenu, MSG_BED_LEVELING,
            #if ENABLED(ENABLE_LEVELING_FADE_HEIGHT)
              _lcd_goto_bed_leveling
            #else
              lcd_bed_leveling
            #endif
          );
    #else
      #if PLANNER_LEVELING
        MENU_ITEM(gcode, MSG_BED_LEVELING, PSTR("G28\nG29"));
      #endif
      #if ENABLED(LEVEL_BED_CORNERS)
        if (axis_homed[X_AXIS] && axis_homed[Y_AXIS] && axis_homed[Z_AXIS])
          MENU_ITEM(function, MSG_LEVEL_CORNERS, _lcd_level_bed_corners);
      #endif
    #endif
```
It might be as simple as changing to something like this:
```
    //
    // Level Bed
    //
    #if ENABLED(AUTO_BED_LEVELING_UBL)
      MENU_ITEM(submenu, MSG_UBL_LEVEL_BED, _lcd_ubl_level_bed);
    #endif

    #if ENABLED(LCD_BED_LEVELING)
      #if ENABLED(PROBE_MANUALLY)
        if (!g29_in_progress)
      #endif
          MENU_ITEM(submenu, MSG_BED_LEVELING,
            #if ENABLED(ENABLE_LEVELING_FADE_HEIGHT)
              _lcd_goto_bed_leveling
            #else
              lcd_bed_leveling
            #endif
          );
    #else
      #if PLANNER_LEVELING
        MENU_ITEM(gcode, MSG_BED_LEVELING, PSTR("G28\nG29"));
      #endif
      #if ENABLED(LEVEL_BED_CORNERS)
        if (axis_homed[X_AXIS] && axis_homed[Y_AXIS] && axis_homed[Z_AXIS])
          MENU_ITEM(function, MSG_LEVEL_CORNERS, _lcd_level_bed_corners);
      #endif
    #endif
```

bjarchi, 2017-12-05:
> In theory... either one should be fine for the 8-bit AVR's. But I think you will find less problems for your development if you do the work in the bugfix-v1.1.x branch. With that said, doing the work (and hitting problems) in the bugfix-v2.0.0 is valuable in its own right

Thanks! I'll stick with 1.1.x for now, but probably will move to 2.0.x before too long so I can start testing there instead. It's good to know that features are at parity (or should be). Am I correct in thinking that the project is trying to move toward deprecation of the 1.x versions in favor of 2.x?

As for the code suggestion - that's a great idea! I was looking at that section of code as well, but your solution is more elegant than anything that occurred to me. I'll try it out and see if there are any unwanted effects this evening, and put together PRs if it looks good. Thanks!!

Roxy-3D, 2017-12-05:
> Am I correct in thinking that the project is trying to move toward deprecation of the 1.x versions in favor of 2.x?

Yes.   Depreciation wasn't the word I would choose.  But that is what is being done.   v2.0.0 has two major changes.  It has the files moved into a non-flat hierarchy.  And it provides support for some 32-bit platforms.   Both of those changes pretty much work.  But there are still issues.   When the issues are put to bed, bugvis-v1.1.x will stop being a development branch and most activity will shift to bugfix-v2.0.0.   Right now, we are doing extra work to keep both branches in lock step.

Roxy-3D, 2017-12-05:
> I'll try it out and see if there are any unwanted effects this evening, and put together PRs if it looks good. Thanks!!

This isn't tested other than to get the menu items to show up.  There is a problem with these changes.  Telling it to probe the corners caused my Z-Axis to try to go up to infinity.   (Now...  I need to get my X-carriage leveled again!)    But it will give you a good starting point for what you are trying to do.

[Marlin-bugfix-1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/1533150/Marlin-bugfix-1.1.x.zip)

bjarchi, 2017-12-06:
Sweet! I'll play around with it when I can. 

At the moment I'm dealing with the fact that the `G29 J` bug has returned... somehow... and I no longer have any idea how to reliably trigger it :|

I was able to watch the error message scrolled on the LCD... 

To the terminal it just writes 
```
Send: N15 G29 J*76
[...]
Recv: ? in get_z_correction(rx0=241.00, ry0=28.00)
```
As before...

The LCD message was an `index out of range` error in get_z_correction. What's the protocol here? Re-open the old issue, or re-create a new one? (I'm going to pull down commits from today before I do either, in case it got incidentally fixed in the  meantime). Grr. Heisenbug.

Roxy-3D, 2017-12-06:
> Recv: ? in get_z_correction(rx0=241.00, ry0=28.00)

That message is from ubl_motion.cpp.   It detected you trying to move the nozzle off of the mesh.   In a perfect world, your mesh should cover everywhere the nozzle is going to try to print.   You can go off the mesh, but you get no correction when you do that.

bjarchi, 2017-12-06:
Yeah the problem is that that point should be well within my mesh, and the same exact `G29 J` works most of the time (with the same probe points). I've also seen the same error happen with `G29 J2`, where UBL is picking the probe points itself rather than using the defined 3-point probes. 

This is #8641 that we were talking about yesterday. I thought it arose due to some subtle EEPROM corruption that was fixed in the course of trying to track it down, but apparently not. I just didn't have the conditions to trigger it nailed down as well as I thought. It may be that it only occurs on `G29 L[n]; G29 J` after a *cold* boot, rather than after any controller reset. 

I'll keep poking at it. For now it seems I can work around it by just not using `G29 J` and working directly from the loaded mesh. I will also try pulling in the 3-point probe points, but I don't think that's the root cause since those points have been working fine for me since I started using UBL.

Roxy-3D, 2017-12-06:
The specified probe points are corrected for the X & Y PROBE_OFFSET_FROM_EXTRUDER.     The specified location is what is probed.   That implies the nozzle will not be at that location.  The nozzle location will have the X & Y PROBE_OFFSET_FROM_EXTRUDER subtracted from it.

This is the 3-Point tilting code in UBL.   It uses the defined points exactly as specified.
```
  else { // grid_size == 0 : A 3-Point leveling has been requested
          float z3, z2, z1 = probe_pt(UBL_PROBE_PT_1_X, UBL_PROBE_PT_1_Y, false, g29_verbose_level);
          if (!isnan(z1)) {
            z2 = probe_pt(UBL_PROBE_PT_2_X, UBL_PROBE_PT_2_Y, false, g29_verbose_level);
            if (!isnan(z2))
              z3 = probe_pt(UBL_PROBE_PT_3_X, UBL_PROBE_PT_3_Y, true, g29_verbose_level);
          }

          if (isnan(z1) || isnan(z2) || isnan(z3)) { // probe_pt will return NAN if unreachable
            SERIAL_ERROR_START();
            SERIAL_ERRORLNPGM("Attempt to probe off the bed.");
            goto LEAVE;
          }
```

And within the probe_pt() routine the correction for the Probe Offset happens:
```
  float probe_pt(const float &rx, const float &ry, const bool stow, const uint8_t verbose_level, const bool probe_relative=true) {
...
    float nx = rx, ny = ry;
    if (probe_relative) {
      if (!position_is_reachable_by_probe(rx, ry)) return NAN;  // The given position is in terms of the probe
      nx -= (X_PROBE_OFFSET_FROM_EXTRUDER);                     // Get the nozzle position
      ny -= (Y_PROBE_OFFSET_FROM_EXTRUDER);
    }
    else if (!position_is_reachable(nx, ny)) return NAN;        // The given position is in terms of the nozzle
...
    const float old_feedrate_mm_s = feedrate_mm_s;
    feedrate_mm_s = XY_PROBE_FEEDRATE_MM_S;

    // Move the probe to the starting XYZ
    do_blocking_move_to(nx, ny, nz);
```

bjarchi, 2017-12-06:
Hmm, that's not what's being reported by DEBUG_LEVEL though... 

Take a look at #8641 again. My 3-point leveling point defines are in the first post there, and in one of the near-to-last posts you can see a couple examples of debug output with `M111 S32` turned on for verbose leveling output. All of the coordinates reported in the leveling output clearly have the `PROBE_OFFSET_...` values baked in (e.g. x=170 -> x=241, x=30 -> x=101). 

Maybe this points to an issue with the machine space coordinate system? 

It still doesn't explain why I only see the `G29 J` problem some of the time though (e.g., not after re-initializing the UBL system with `G29 P1; G29 P3; G29 S1`. 

Should we re-open #8641 for this, or should I create a new issue? (provided it hasn't been magically fixed by the past day's commits - I haven't been able to check yet) It's definitely not related to the LEVEL_BED_CORNERS thing, and it looks like Scott already fixed that one.

Roxy-3D, 2017-12-06:
If you X_Max is less than 241...  We should try shifting all X probe locations to a smaller number so that doesn't happen any more.  (165 ???)

> It still doesn't explain why I only see the G29 J problem some of the time though (e.g., not after re-initializing the UBL system with G29 P1; G29 P3; G29 S1.

Agreed.   We may need to investigate further.  But let's get the simple stuff taken care of first.   And then see if anything else is still mis-behaving.

bjarchi, 2017-12-06:
> If your X_Max is less than 241... We should try shifting all X probe locations to a smaller number so that doesn't happen any more. (165 ???)

 X_MAX (assuming this is global X_MAX?) is 249 in my system - so there should be no problem there. My G27 park_head point is (244, 181) and that doesn't cause any problems (except for a previously reported bug with T1 activated). 

I will try moving the probe point in a bit - I think I already updated in my latest config.h to 160,30 for that last point, but still need to flash it. I'll report back when I do so (should be this evening).

Roxy-3D, 2017-12-06:
Also...   Your Mesh Inset (if being used) will shrink the usable space too...

bjarchi, 2017-12-06:
Yeah, but it's only 6mm (previously 5). I think I posted an output from `G29 W` in the original bug too, but the mesh_max_x = 208 that I quoted above was with the mesh_inset taken into account (BED_SIZE_X = 214, MESH_INSET=6).

I'll try the more conservative 3-point definitions when I'm home, but my gut feeling is that it won't solve the problem (esp. since `G29 J2` also threw errors, and UBL generates those probe points from the mesh boundaries.). 

It's a little confusing to me that `G29 P1` never runs into errors - I suppose that's because it's not trying to get correction values while it probes?

bjarchi, 2017-12-06:
Since #8641 is closed, and the original subject of this bug seems to have been patched, do you think I should edit the title to reflect current discussion of the `G29 J` issue?

Roxy-3D, 2017-12-06:
> It's a little confusing to me that G29 P1 never runs into errors - I suppose that's because it's not trying to get correction values while it probes?

That is true.  
```
      /**
       * This is the generic Z-Correction. It works anywhere within a Mesh Cell. It first
       * does a linear interpolation along both of the bounding X-Mesh-Lines to find the
       * Z-Height at both ends. Then it does a linear interpolation of these heights based
       * on the Y position within the cell.
       */
      static float get_z_correction(const float &rx0, const float &ry0) {
        const int8_t cx = get_cell_index_x(rx0),
                     cy = get_cell_index_y(ry0);

        if (!WITHIN(cx, 0, GRID_MAX_POINTS_X - 2) || !WITHIN(cy, 0, GRID_MAX_POINTS_Y - 2)) {

          SERIAL_ECHOPAIR("? in get_z_correction(rx0=", rx0);
          SERIAL_ECHOPAIR(", ry0=", ry0);
          SERIAL_CHAR(')');
          SERIAL_EOL();
```
This looks wrong.   It would seem GRID_MAX_POINTS_X-1 and GRID_MAX_POINTS_Y-1 are what should be used here.   Please replace this function in ubl.h and see if things behave better:
```

      /**
       * This is the generic Z-Correction. It works anywhere within a Mesh Cell. It first
       * does a linear interpolation along both of the bounding X-Mesh-Lines to find the
       * Z-Height at both ends. Then it does a linear interpolation of these heights based
       * on the Y position within the cell.
       */
      static float get_z_correction(const float &rx0, const float &ry0) {
        const int8_t cx = get_cell_index_x(rx0),
                     cy = get_cell_index_y(ry0);

        if (!WITHIN(cx, 0, GRID_MAX_POINTS_X - 1) || !WITHIN(cy, 0, GRID_MAX_POINTS_Y - 1)) {

          SERIAL_ECHOPAIR("? in get_z_correction(rx0=", rx0);
          SERIAL_ECHOPAIR(", ry0=", ry0);
          SERIAL_CHAR(')');
          SERIAL_EOL();

          #if ENABLED(ULTRA_LCD)
            strcpy(lcd_status_message, "get_z_correction() indexes out of range.");
            lcd_quick_feedback();
          #endif
          return NAN;
        }

        const float z1 = calc_z0(rx0,
                                 mesh_index_to_xpos(cx), z_values[cx][cy],
                                 mesh_index_to_xpos(cx + 1), z_values[cx<GRID_MAX_POINTS_X-1 ? cx + 1 : cx][cy]);

        const float z2 = calc_z0(rx0,
                                 mesh_index_to_xpos(cx), z_values[cx][cy<GRID_MAX_POINTS_Y-1 ? cy + 1 : cy],
                                 mesh_index_to_xpos(cx + 1), z_values[cx<GRID_MAX_POINTS_X-1 ? cx + 1 : cx][cy<GRID_MAX_POINTS_Y-1 ? cy + 1 : cy]);

        float z0 = calc_z0(ry0,
                           mesh_index_to_ypos(cy), z1,
                           mesh_index_to_ypos(cy + 1), z2);

        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (DEBUGGING(MESH_ADJUST)) {
            SERIAL_ECHOPAIR(" raw get_z_correction(", rx0);
            SERIAL_CHAR(',');
            SERIAL_ECHO(ry0);
            SERIAL_ECHOPGM(") = ");
            SERIAL_ECHO_F(z0, 6);
          }
        #endif

        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (DEBUGGING(MESH_ADJUST)) {
            SERIAL_ECHOPGM(" >>>---> ");
            SERIAL_ECHO_F(z0, 6);
            SERIAL_EOL();
          }
        #endif

        if (isnan(z0)) { // if part of the Mesh is undefined, it will show up as NAN
          z0 = 0.0;      // in ubl.z_values[][] and propagate through the
                         // calculations. If our correction is NAN, we throw it out
                         // because part of the Mesh is undefined and we don't have the
                         // information we need to complete the height correction.

          #if ENABLED(DEBUG_LEVELING_FEATURE)
            if (DEBUGGING(MESH_ADJUST)) {
              SERIAL_ECHOPAIR("??? Yikes!  NAN in get_z_correction(", rx0);
              SERIAL_CHAR(',');
              SERIAL_ECHO(ry0);
              SERIAL_CHAR(')');
              SERIAL_EOL();
            }
          #endif
        }
        return z0;
      }
```

bjarchi, 2017-12-06:
That does indeed look questionable... and parts of that function were changed recently enough (for machine space) to be plausible as a root cause :) 

I'll try it out as soon as I get home - probably around 19:00 PST

Roxy-3D, 2017-12-06:
The -2 on those values is attempting to keep the mesh cell index numbers in range.   The arrays go from 0 to GRID_MAX_POINTS_X-1.   So -2 lets it use the correct indexes to cover the whole mesh.   To do a Z correction you need both the left and right (and top and bottom) mesh points.   

The problem with the way it was is if the mesh goes from 0 to 200 and you try to position the nozzle at 200, there are no mesh points further out to help with the extrapolations.   Those ternary operators I put in place should just use the edge values if the nozzle gets to the edge or even past it.

bjarchi, 2017-12-07:
> The -2 on those values is attempting to keep the mesh cell index numbers in range. The arrays go from 0 to GRID_MAX_POINTS_X-1. So -2 lets it use the correct indexes to cover the whole mesh. To do a Z correction you need both the left and right (and top and bottom) mesh points.

Oh, yeah, I remember you explaining this logic when we were talking about UBL documentation now.  

>  Those ternary operators I put in place should just use the edge values if the nozzle gets to the edge or even past it.

So does this mean it's safe to use ...-1 now? 

Still doesn't make sense to me that this would have suddenly become a problem in the past week, but I'll try it out and see if it addresses the problem at least. Might have to wait until tomorrow, nightmare traffic and I just got home from work.

Roxy-3D, 2017-12-07:
> So does this mean it's safe to use ...-1 now?

A better way to word it would be..  With -1 we can allow the nozzle to go right up to the very edge of the mesh.   And with the ternary operators, we won't violate the subscript bounds.   We will just use the same correction that would be applied at the edge of the mesh if the nozzle keeps moving outward.

> Still doesn't make sense to me that this would have suddenly become a problem in the past week, but I'll try it out and see if it addresses the problem at least. Might have to wait until tomorrow, nightmare traffic and I just got home from work.

None of this is simple and easy.  You are doing something most people are not doing and that is triggering the problem.   But...  it doesn't matter if hardly anybody else sees the problem.  If there is a bug...  we need to fix it.

I suspect you will see better behavior when you get the changes in and try printing something.   I have the modified firmware loaded and a fresh mesh generated.   Maybe tomorrow I'll have time to kick off a print.   If things work better for you and I can get a successful print using the changed firmware...   I'll merge the changes.    And we can start working on the next difficulty you find...

bjarchi, 2017-12-07:
> A better way to word it would be.. With -1 we can allow the nozzle to go right up to the very edge of the mesh. And with the ternary operators, we won't violate the subscript bounds. We will just [...]

That makes sense, and it seems like the behavior that most users would expect as well. 

> None of this is simple and easy. You are doing something most people are not doing and that is triggering the problem. But... it doesn't matter if hardly anybody else sees the problem. If there is a bug... we need to fix it.

I'm glad we're on the same page there. I was a little worried that I was stealing cycles from something more important since I'm the only one apparently seeing this issue - even if I am also trying to track it down. 

I'll be able to try changing the subscript bounds today and see if that fixes things. I'm able to print by omitting the `G29 J` in my preamble, but I think I can still reliably trigger the bug when I do use it. The only confusing thing is how it seems to come and go depending on how the system was initialized. 

I was reading through some of the core code yesterday to start getting a feel for how the project is structured, and I think I found a bug in `conditionals_post.h`. I'm going to set up a PR for that soon, and I'll reference it here. Would appreciate if you could take a look since it relates to mesh setup.

bjarchi, 2017-12-09:
I see you made the change to GRID_MAX... - 1 while I was waiting for the homing fixes to get merged (and work and stuff) :) 

I was about to start working on this when I noticed your changes, so I have some time to test now. I will close this issue if I'm unable to reproduce the `G29 J` issue - although I do want to test enough to be sure I haven't missed it again. 

BTW I think this -1 change makes sense regardless, since it brings the handling near MAX in parity with how things are handled near 0 (min_index). Do you agree / am I understanding things correctly?

bjarchi, 2017-12-09:
Update: So far I've seen no failures of `G19 L[n]; G29 J` commands since flashing the latest commits. I'd like to do a bit more testing to be sure, so maybe leave the issue open for another day or so? I'll close it tomorrow if I don't see any more failures. 

THANK YOU for all of your help!

bjarchi, 2017-12-09:
Further update:

`LEVEL_BED_CORNERS` menu shows up with UBL enabled, and does the right thing - at least, what it's programmed to do. Might look into having it keep both/all extruders in a multi- setup over the bed at some point, but that's a new FR/PR. 

`G29 L[n]; G29 J` has yet to fail for me, even when I reset my largest X coordinate back to 170 (the original value that exposed this problem). Closing this issue now; I will re-open if the problem returns but I doubt it will.

closed 2017-12-09 by bjarchi

thinkyhead, 2017-12-10:
> the right thing - at least, _what it's programmed to do_

🤣

Roxy-3D, 2017-12-10:
The full Marlin feature set should be available and working...   No argument there...

But the truth is I've never used the 4-corner leveling because G29 P1 T U actually does a better job.   You should give that a try just to see what it does.   And you don't have to run to completion.   When you have enough points, stop it and make what ever change makes sense.    

And then do a G28 and run it again...     If you only are going to probe 4 points, the corners do make sense.   The problem is the edges (and especially the corners) have 'edge effects'.   You want to see some number of other points so you know how much to discount the values you are seeing in the corners.

With that said....   Use what ever works for you!!!!!

thinkyhead, 2017-12-16:
> I've never used the 4-corner leveling

It isn't a leveling procedure. It's just for physically adjusting the corner screws.