MarlinFirmware/Marlin #8678: [FR] chromium based configuration tool for the firmware

Swwils, 2017-12-05:
With 32bit boards coming, why don't we look at making a chromium based configuration tool for the firmware; that can flash our required targets and then configure them in a GUI rather than having to edit inside the arduino IDE.

There is already a large and well developed OS project with similar goals; cleanflight and its configurator

https://github.com/cleanflight/cleanflight-configurator

This most likely would also simplify firmware uploads for 32bit boards in the future.

Spawn32, 2017-12-05:
You have seen this ? dosent flash but it's @thinkyhead 's gui Configurator..

http://rawgit.com/thinkyhead/MarlinConfigurator/master/src/index.html

i think i have seen another one as well, but dont have the link...

Swwils, 2017-12-05:
Yes I have seen the configurator, its great!

But as far as I am aware such a tool becomes complex with 32bit targets because of the tool chain involved.

bruce356, 2017-12-05:
A configurator like Repetier's would be the go, its simple to use for novices and pros, works all the time, I have never had a problem with it.
https://www.repetier.com/firmware/dev/index.php.

Visted http://rawgit.com/thinkyhead/MarlinConfigurator/master/src/index.html , loaded my existing config files and have no idea how this is supposed to work, for me its easier to edit my files directly.

regards - bruce

Sineos, 2017-12-05:
There is also: https://github.com/akaJes/marlin-config by @akaJes 
Really nice and well done

bruce356, 2017-12-05:
Its ok if you can get it to work on your computer I could not.
too complicated to install

thinkyhead, 2017-12-06:
It's always easier to edit your files directly, or use a diff tool as most text editors now provide. The main request has been a way to convert older configs into newer ones, and that is one thing that marlin-config provides.

As for my old configurator, it's well out of date and needs to be completed. The idea of my configurator, starting out, was to provide the "expert" view on the configs showing every option within categories, and to base the forms directly on the configuration files. Later on I was planning to produce a more chopped-down version of the interface with the core options up front and the rest hidden away.

All-in-all it's a daunting project. It should be possible to produce a tool that converts any config between versions of Marlin, since the configurations are well-defined for each tagged release. The marlin-config tool by @akaJes does handle that. As a macOS user I can't evaluate that tool. I feel strongly that all such tools should be written in Javascript so they can run anywhere, locally or remotely, and that's the ethos I employed with my configurator.

Someone should really finish that thing….

bjarchi, 2017-12-06:
> It's always easier to edit your files directly, or use a diff tool as most text editors now provide. The main request has been a way to convert older configs into newer ones, and that is one thing that marlin-config provides.

`git stash` and `git stash pop` work pretty well for this too. Maybe not great for very different releases, but in general they do the right thing most of the time - and leave behind an easily interpreted merge conflict when they don't. Not for the beginning user unless they already work with git, but I've found it super helpful for tracking the development branch.

akaJes, 2017-12-06:
> too complicated to install

download and install **TWO** files - I agree - it is very complicated! :)
https://github.com/akaJes/marlin-config/releases

Swwils, 2017-12-07:
> download and install TWO files - I agree - it is very complicated! :)
https://github.com/akaJes/marlin-config/releases

I have made a PR for your readme that should help with this!

closed 2017-12-10 by Swwils

thinkyhead, 2017-12-16:
> git stash and git stash pop work pretty well for this too. 

Well, in that sense so does `git reset my_old_version`. After that command you can see the diffs quite clearly in Github Desktop, for example.