MarlinFirmware/Marlin #8682: bugfix-1.1.x Compile Error

JimBrown, 2017-12-06:
### Compile Error
When I compile I get an error:
```
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Build\sketch\planner.cpp: In static member function 'static void Planner::_buffer_line(const float&, const float&, const float&, const float&, const float&, uint8_t)':

planner.cpp:1439: error: 'DISABLE_STEPPER_DRIVER_INTERRUPT' was not declared in this scope

     DISABLE_STEPPER_DRIVER_INTERRUPT();

                                      ^

planner.cpp:1442: error: 'ENABLE_STEPPER_DRIVER_INTERRUPT' was not declared in this scope

     ENABLE_STEPPER_DRIVER_INTERRUPT();

                                     ^

Using library LiquidCrystal at version 1.0.7 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
'DISABLE_STEPPER_DRIVER_INTERRUPT' was not declared in this scope
```

thinkyhead, 2017-12-06:
I think that's already been fixed. Please try the latest code.

Swopno, 2017-12-06:
Thinkyhead, I tried to compile 1.1.6 and get compile errors in marlinserial.h so where would you say the latest code is? thanks

JimBrown, 2017-12-06:
Just updated a minute ago. Now getting this:
```
planner.cpp:1425: error: prototype for 'void Planner::_buffer_line(const float&, const float&, const float&, const float&, const float&, uint8_t)' does not match any in class 'Planner'

 void Planner::_buffer_line(const float &a, const float &b, const float &c, const float &e, const float &fr_mm_s, const uint8_t extruder) {

      ^

planner.cpp:698: error: candidate is: static void Planner::_buffer_line(const float&, const float&, const float&, const float&, float, uint8_t)

 void Planner::_buffer_line(const float &a, const float &b, const float &c, const float &e, float fr_mm_s, const uint8_t extruder) {
```

Thanks.

fiveangle, 2017-12-06:
> thinkyhead : Please try the latest code.

> Swopno : I tried to compile 1.1.6 and get compile errors in marlinserial.h so where would you say the latest code is? thanks

Latest is always the bugfix* branch of the major release set you are running.

-=dave

JimBrown, 2017-12-06:
Might be a git issue. I have a fork of Marlin, which is cloned to my desktop. When I got the first error above, I reverted the merge commits I had just received. If I then do a new merge will I get the commits I reverted again, plus any changes since those commits?

JimBrown, 2017-12-06:
@thinkyhead 

git merge upstream/bugfix-1.1.x tells me it is "Already up to date". Still getting the error in my original post.

Thanks.

fiveangle, 2017-12-06:
```
avr-objcopy -O ihex -R .eeprom .pioenvs\megaatmega2560\firmware.elf .pioenvs\megaatmega2560\firmware.hex
AVR Memory Usage
----------------
Device: atmega2560
Program:   54178 bytes (20.7% Full)
(.text + .data + .bootloader)
Data:       2619 bytes (32.0% Full)
(.data + .bss + .noinit)
 [SUCCESS] Took 172.34 seconds 
 [SUMMARY] 
Environment megaatmega2560      [SUCCESS]
Environment megaatmega1280      [SKIP]
Environment printrboard         [SKIP]
Environment printrboard_revf    [SKIP]
Environment brainwavepro        [SKIP]
Environment rambo               [SKIP]
Environment anet10              [SKIP]
Environment sanguino_atmega644p [SKIP]
 [SUCCESS] Took 172.36 seconds 
```

fiveangle, 2017-12-06:
```
Linking everything together...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc"  -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections,--relax -mmcu=atmega2560  -o "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032/Marlin.ino.elf" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\G26_Mesh_Validation_Tool.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\twibus.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\servo.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\serial.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\M100_Free_Mem_Chk.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\stepper_indirection.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\digipot_mcp4451.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\SdFatUtil.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\gcode.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\hex_print_routines.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\leds.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\nozzle.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\printcounter.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\stepper_dac.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\Marlin.ino.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\planner.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\MarlinSerial.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\pca9632.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\mesh_bed_leveling.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\temperature.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\ubl.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\Marlin_main.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\ubl_G29.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\stepper.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\Max7219_Debug_LEDs.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\ultralcd.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\utility.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\ubl_motion.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\Sd2Card.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\planner_bezier.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\vector_3.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\SdBaseFile.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\cardreader.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\watchdog.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\SdVolume.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\digipot_mcp4018.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\dac_mcp4728.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\I2CPositionEncoder.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\configuration_store.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\least_squares_fit.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\blinkm.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\stopwatch.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\endstops.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\neopixel.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032\sketch\SdFile.cpp.o" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032/core\core.a" "-LC:\Users\djohnson\AppData\Local\Temp\arduino_build_622032" -lm
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0  "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032/Marlin.ino.elf" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032/Marlin.ino.eep"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-objcopy" -O ihex -R .eeprom  "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032/Marlin.ino.elf" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_622032/Marlin.ino.hex"
Sketch uses 53602 bytes (21%) of program storage space. Maximum is 253952 bytes.
Global variables use 2618 bytes (31%) of dynamic memory, leaving 5574 bytes for local variables. Maximum is 8192 bytes.
```

JimBrown, 2017-12-07:
Fixed.

But I have no idea what happened. I fixed it by downloading a new zip of the bugfix-1.1.x branch and overwriting the existing files in my local repository with the downloaded files. About 60 files (mostly example configs, but also several .h and .cpp files) showed up as changed in my local repository. Committed them and pushed all that to my github repository. All appears to be good now. Pretty sure I messed things up somewhere along the way.

Thanks for looking.

closed 2017-12-07 by JimBrown