MarlinFirmware/Marlin #8683: Babystepping not compiling for Sbase in 2.0

Sto-li, 2017-12-06:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `#define BABYSTEPPING` I get an error:
```
Paste_the_error_text_here
```Marlin\src\module\stepper.cpp:1322:7: error: '_SAVE_START' was not declared in this scope
Marlin\src\module\stepper.cpp:1324:7: error: '_PULSE_WAIT' was not declared in this scope

------------------------------------------------------------
### Bug Report
- Description: --- babystepping
- Expected behaviour: --- compile
- Actual behaviour: --- errors out
- Steps to reproduce:
  - Do this uncoment babystepping, select sbase board 
  - Do that--- I think it is because it is trying to use TCNT0 but is not described or used by 32 bit boards? 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1534390/Configuration.zip)

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

coolio986, 2017-12-06:
The conditionals that cause this compile error are when `MINIMUM_STEPPER_PULSE` is larger than 0 and `#define BABYSTEPPING`. The calculation in stepper.cpp is `#define EXTRA_CYCLES_BABYSTEP (STEP_PULSE_CYCLES - (CYCLES_EATEN_BABYSTEP))`

When `EXTRA_CYCLES_BABYSTEP > 20` it runs code that is specific to AVR processors. TCNT0 is the faulting condition since TCNT0 (timer) doesn't exist in the LPC1768.

```
// Stepper pulse duration, in cycles
#define STEP_PULSE_CYCLES ((MINIMUM_STEPPER_PULSE) * CYCLES_PER_MICROSECOND)
```

The above code block in Conditionals_post.h is used in stepper.cpp to calculate the `EXTRA_CYCLES_BABYSTEP`

I've changed the code block from:
`#if EXTRA_CYCLES_BABYSTEP > 20`

to

`#if EXTRA_CYCLES_BABYSTEP > 20 && !defined(CPU_32_BIT)`  on Line 1305

and it compiles ok.

I cannot say this is the right fix, but at least we've identified the block of code causing the issue.

Hopefully Thomas or Scott can give it a look and figure out if this is the corrective action.

coolio986, 2017-12-06:
The better way to solve this is to get the timer from the LPC and if defined around `_SAVE_START` and `_PULSE_WAIT` and switch between 8 bit and 32 bit timers.

coolio986, 2017-12-06:
I think this is the code we're looking for:

```
#if EXTRA_CYCLES_BABYSTEP > 20 
    #ifdef CPU_32_BIT
      #define _SAVE_START const uint32_t pulse_start = HAL_timer_get_count(STEP_TIMER_NUM)
      #define _PULSE_WAIT while (EXTRA_CYCLES_BABYSTEP > (uint32_t)(HAL_timer_get_count(STEP_TIMER_NUM) - pulse_start) * (INT0_PRESCALER)) { /* nada */ }
    #else
      #define _SAVE_START const uint32_t pulse_start = TCNT0
      #define _PULSE_WAIT while (EXTRA_CYCLES_BABYSTEP > (uint32_t)(TCNT0 - pulse_start) * (INT0_PRESCALER)) { /* nada */ }
    #endif
  #else
    #define _SAVE_START NOOP
    #if EXTRA_CYCLES_BABYSTEP > 0
      #define _PULSE_WAIT DELAY_NOPS(EXTRA_CYCLES_BABYSTEP)
    #elif STEP_PULSE_CYCLES > 0
      #define _PULSE_WAIT NOOP
    #elif ENABLED(DELTA)
      #define _PULSE_WAIT delayMicroseconds(2);
    #else
      #define _PULSE_WAIT delayMicroseconds(4);
    #endif
  #endif
```

p3p, 2017-12-06:
I remember fixing this before when I went through looking for all mentions of AVR registers.. but apparently not? you're nearly right @coolio986 but It should be `HAL_timer_get_current_count` to return the timer value, `HAL_timer_get_count` returns the match register value, (I didn't choose the naming convention), also if you find yourself needing an `#ifdef CPU_32_BIT` it reveals an issue in the HAL implementation (AVR is missing the HAL_timer_get_count implementation).

coolio986, 2017-12-06:
Ah, a generic function call across all HAL's. I suppose a pull request is in order.

thinkyhead, 2018-02-10:
Presuming the original issue is resolved.

closed 2018-02-10 by thinkyhead