MarlinFirmware/Marlin #8692: [FR] Raise3D Printer/Touchscreen support

Firesped, 2017-12-06:
### Feature Request
When marlin 1.1.0 came out and worked on and created a version that works with the Raise3D controller and touchscreen. This firmware works and I am not looking to get the firmware working with Marlin. This request is to work on integrating the processes for that printer/touchscreen into Marlin officially.

Mostly it was modifying the outputs to be compatible with the pcDuino Touchscreen's expected formatting. 

I created it under my own repository so that changes to the Marlin repository did not start throwing things off. So far it has looked that none of the updates have affected anything but I am not sure.

I can see that options could be added to a special file for the raise3d printer for configurations it needs. 
The issues are that it requires special processes under a number of places and that would likely mean that official support for it would need to be added so that those processes can be subbed out for the ones that the raise3d touchscreen needs. 

I am not officially affiliated with Raise3D. so modifying the format of the closed sourced Touchscreen's expectations are not tasks I can do.

Of the top of my head, as I have not looked at the firmware recently, the processes include

a modification to the printer parameter startup report to include SD INIT FAILED message. The motion controller does not have an SD CARD but the touchscreen expects the message as part of it's handshake

a modification to the OK output for communication feedback

a modification to the M105 output to report the expected format the touchscreen reads.

I think there is also a modification to the PID tuning for reporting for a PID autotune feature recently added to the touchscreen

There are some custom functions that were added for a custom Filament Runout Sensor Feature added to the touchscreen.

a custom function I added in relation to the FRS feature to configure it.

This is what I can think of off the top of my head but there might be more. 

Mostly looking for feedback on to wither this is something that can be done or if I should just continue to keep my repository as a separate entity. 

To Note: Raise3D will not sell their touchscreen as a separate unit from their printers.

My repository is located here. https://github.com/Firesped/Marlin-Raise3D-N-Series-Unofficial
-Firesped

fiveangle, 2017-12-08:
Surely can be done. Just port your changes to the bugfix2.0.x branch, and the new feature to Configuration*.h (or pins_[BOARD].h if the touchscreen can’t be made to work on any other printer), and put as many of the logic for it into the Conditional*.h files, wrapping any inline defines for the feature in the rest of the code where absolutely required, then once you have it working, submit the PR.

bjarchi, 2017-12-09:
I looked through the repo when this was first posted, and it seems like it would be difficult at best to merge in as-is. First off, there are changes to how the base code works (e.g. fake 'SD Init fail' message to trick the display) just to support a closed-source display...

But my biggest concern, @Firesped, is that you seem to have made all of your 'changes' in git by simply uploading new changed files in full. Git is made to track *changes* (not *files*) - which makes the process of merging (and seeing what's being merged) pretty simple. The way you made your changes, though, it's nearly impossible for me to tell what changed compared to base Marlin. 

A more experienced contributor will have to confirm, but I suspect that if you really want to submit this as a PR you'll need to start with a clean repo based on bugfix-2.0.x and then re-commit your changes on top of that - properly using git to commit *only changes* - so we can see what is changed and how.

Firesped, 2017-12-09:
at the time I could not understand how to get the software to work to upload the files from my computer. also the repository is not uploaded to github as clean marlin 1.1.0 but as the changed version itself.

at the moment I would need to have the time to work on it. It will likely be to the level of starting over with 2.0.x, so it is not being so critical in changing core code. To note, that core code was not exactly changed but added onto so that with a #define, it would alter the output for the code. 

It looks like I have a clean copy of 1.1.0, so I could upload that to a repository and then fork off the altered version. I have yet to look at the 2.0.x code yet.

bjarchi, 2017-12-09:
As I said, github has really good articles on things like setting up / forking repos, tracking upstream changes, etc. They really need a button to 'pull changes from upstream' on a forked repo, but they don't have it so you have to do a little more manual work with `git` e.g.:
[Setting up your upstream remote](https://help.github.com/articles/configuring-a-remote-for-a-fork/)
[Merging changes from upstream](https://help.github.com/articles/syncing-a-fork/)
For the latter you have to change branch names as appropriate, but the process is straightforward. 

Github and [git](https://git-scm.com/book/en/v2) both provide substantial tutorial material to help you understand how git works and what workflows are common. 

The basic process for github is that you want to fork a repo (Marlin), clone it locally, checkout the branch of interest, create a new branch for your feature work - based on branch of interest, make your changes - committing logical chunks as you go, [periodically merge in upstream changes as necessary, or rebase], push your *changes* back to your fork in github, then create a PR against the original base branch (e.g. MarlinFirmware/Marlin/bugfix-2.0.x). 

This sounds complicated but it really isn't once you get used to the git paradigm. I think it took me about 5 minutes the first time I wanted to set up a PR to fork, clone, checkout, etc. - but I was experienced with git and only new to GitHub's particular workflow.

Firesped, 2017-12-10:
my brother recommended TortoiseGIT, so I got that. I had a local copy of marlin 1.1.0. I uploaded that to a new repository, noting it was a clone. Then I created a branch of for raise3d version 1.2.1.1 (they created their own version numbers.) That is located here. https://github.com/Firesped/Marlin-1.1.x/tree/raise3D-1.2.1.1

raise3d version 1.2.2 is located here. But I need to create the local branch and add the modified files into it still. then push the files up.
https://github.com/Firesped/Marlin-1.1.x/tree/raise3d-1.2.2

bjarchi, 2017-12-10:
> my brother recommended TortoiseGIT, so I got that.

TortoiseSVN was a pretty good solution for svn, but I use the command line for git so I can't comment on the GUI solutions. IMO command line is really the way to go for git usage - worth learning, and will make following tutorials much easier. 

The key is to *fork* marlin in github, which automatically sets up a lot of the tracking you'll want for PRs. Then clone it locally, and make changes to the files as needed for the Raise3d support. Don't copy over entire files from their repo; just change what needs to be different (use diff or a diff tool). Then commit those *changes* to your forked repo, and submit a PR. 

This is the zen of git. Unlike other vc systems, it tracks changes (not whole files). Makes working on a distributed project much easier, but you need to use it correctly. Read some of those docs I posted above - the beginning chapters of the git book are a really good overview. It takes time to learn, but it's time well spent.

Firesped, 2017-12-11:
My issue is, if I fork marlin now. I have to spend a ton of time redoing all the work I have in the current version. then I have to test it to make sure it works. Which I can't do at this time because my printer needs parts.
This was built on Marlin 1.1.0, marlin is now on Marlin 1.1.6. That is why I took the copy I downloaded back on 5/5/2017 and used tortoisegit to push it to github. Then I made a fork of that and put the files I changed, created a commit and pushed it to the fork.

configuration changes, which are likely normal for that file to be customized. 
I did a modification on the Version only so the firmware would show the that it was marlin 1.1.0 since raise started using their own version numbering. I think there firmware was based on 1.0.1 or earlier. 

in configurationstore marlinsettings::reset() I had some code at the end if the N-series protocle was set and SDsuport was disabled to do SERIAL_ECHOLNPGM(MSG_SD_INIT_FAIL).

The touchscreen was using message to discover if the firmware was working on the controller board. 
They were turning on support for the reprap discount full graphics screen to get it to run SDSUPPORT to get an SD INIT FAIL message. by adding that little bit of code, I was able to save a ton of memory allocation that was not needed. 

MSG_B and MSG_T in language had to be adjusted back to using "ok"s

a number of PIN adjustments in the RUMBA pin file.

The added PID Autotuning to the touchscreen. I had to do a dirty patch to make it look like the older firmware formats the touch screen was looking for.

ultimately a lot of my changes were making the firmware display stuff like it did back in marlin 1.0.1 or earlier. I had to make print_heaterstates display correctly. A lot of the work I did used a control #define to turn it on. 

stuff in marlin main touched was print_heaterstates, M105, M109 and ok_to_send.

ok to send was sending back feedback to the touchscreen either 0 or planner.block_buffer_head - planner.block_buffer_tail + BLOCK_BUFFER_SIZE)%BLOCK_BUFFER_SIZE

bjarchi, 2017-12-11:
> My issue is, if I fork marlin now. I have to spend a ton of time redoing all the work I have in the current version. then I have to test it to make sure it works.

This is true, but unfortunately it's also probably the only option to get a PR approved. I'm not a core developer so I can't say for sure, but I can say this...

First, if you submit a PR based on your current code and repo, which is in turn based on Marlin 1.1.0, that's kind of like submitting a PR to roll Marlin back to 1.1.0. That's not going to happen. 

Basically, submitting a PR that isn't tracked as changes - to say nothing of one based on so old of a Marlin fork - would be asking the core developers to go through and figure all this out for you. Basically, asking them to do the work that you would otherwise do getting your PR properly set up. Again, I don't see that happening - especially when everyone is currently tracking two versions evolving simultaneously until development switches to 2.0.x. 

I'm not trying to be harsh here, but this is how I see things going given the state of the repo I saw (and that's without realizing it was based on 1.1.0). A major benefit of git is that it allows you to fork a repository, make changes, and continue to pull in the upstream updates (from the main project) without losing the changes that you've made. That's what you want to be doing, and the documentation I linked will help you get to that point, but you will have to replay your changes on top of an up-to-date fork of Marlin (or try to merge the upstream changes into your repo). Either way it's going to take some work, although the merge and diff tools available in git (and in general) will help a lot.

Firesped, 2018-01-30:
I'm finally working on this however, it looks like can't use the FIlament Runout Sensor code base that already exists. the first reason being that I have potentially 2 runout sensors and the second reason is that it puts a code in the serial log and the raisetouch actually handles the process. 

The code base already in Marlin has marlin handle it using advanced pause and other functions.

I have some code that I setup to control the sensors. I created a new gcode but I want to move that into something else. I guess I could modify M120 and M121 to allow for additional input to turn them on and off.  Not sure what to use to configure them. my command allowed for the sensors to be configured to be normally open or normally closed.

Firesped, 2018-01-30:
Now to figure out how to properly do this since I now have the code ready.

https://github.com/Firesped/Marlin/tree/raisetouch-master

thinkyhead, 2018-02-14:
If you know exactly the changes you made, create a new branch based on `upstream/bugfix-1.1.x` (not your stale copy of `bugfix-1.1.x`) and apply equivalent changes to that. Commit them, push to your fork, and then you can make a PR targeting `bugfix-1.1.x`.

From the command-line:
```
git fetch upstream
git checkout upstream/bugfix-1.1.x -b raisetouch_for_bugfix11x
```
(You may need to create `upstream` first.)
```
git remote add upstream git@github.com:MarlinFirmware/Marlin.git
```

Apply your changes, commit them, and push to your fork:
```
git push --set-upstream origin raisetouch_for_bugfix11x
```
Go to your fork page on Github, and you should see an option to "compare" (and create a PR). Do that, and be careful to target `bugfix-1.1.x`.

boelle, 2019-02-19:
@Firesped had time to look at thinky's instructions?

boelle, 2019-06-22:
missing label and i think we can close as no activity

InsanityAutomation, 2019-06-22:
No, keep this one. It basically requires 1.1.0 protocols plus a few custom commands.... Can possibly be emulated with an ExtUI module at this point. I have one here but its a machine I hate with a passion so ill get to it when my open project list gets low....