MarlinFirmware/Marlin #8699: 1.1.6 issue with servos

locutus40, 2017-12-07:
i get this error..... static_assert(COUNT(servo_delay) == NUM_SERVOS, "SERVO_DELAY must be an array NUM_SERVOS long.");

i download it from the repository... dont edit anything just remove the // from the number of servos to test its a good copy and it spits out the error above far back ones dont do this... the versions but they dont have bl touch support.. so i need to find one that will stop giving me this.... error even the fix does this...

makes no sense... the 2.0 there isnt any thing in the files no config.h nothing just a patch.. so i cant even download that...

you guys really need to straighten out the releases have one stable release and go up from there from what i see its 1.0.2.0 then 1.1.6 im confused the first one says latest yet its older .. so which one is the one that works  or fixes this issue?

GMagician, 2017-12-07:
You didn't attached your configuration but I bet you didn't defined servo delay as an array...
when you have 3 servos it should looks like: SERVO_DELAY { 300, 300, 300 }

locutus40, 2017-12-07:
and your right lol just tried it and it works.. would be nice for us noobs if that was already done... but i guess its a learning process it now compiles fine

bjarchi, 2017-12-07:
> and your right lol just tried it and it works.. would be nice for us noobs if that was already done... but i guess its a learning process it now compiles fine

It can't really be 'already done' - in the default configuration NUM_SERVOS is 1, and SERVO_DELAY is defined as { 300 } - so an array of length 1. I suppose there could be a comment instructing the user to edit SERVO_DELAYS as appropriate (near NUM_SERVOS), but the compile error is pretty descriptive.

bjarchi, 2017-12-08:
>you guys really need to straighten out the releases have one stable release and go up from there from what i see its 1.0.2.0 then 1.1.6 im confused the first one says latest yet its older .. so which one is the one that works or fixes this issue?

Regarding versions, 1.1.x is the current 'stable' version, while the 2.0.x is the development version and the direction for new work (not in releases, only in git). I'm not sure what that 1.0.2-2 release is, but my guess would be that it's a small bugfix release for people who are for some reason stuck using 1.0.x with their hardware. 

The version numbers mean exactly what you'd expect - 1.1.x is newer than 1.0.x (which AFAIK is basically not worked on anymore). Just because the release bundle was wrapped up and posted later does not necessarily mean it was newer code

fiveangle, 2017-12-08:
> you guys need to straighten out the releases

Maybe read the readme ? 😂

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/README.md

-=dave

bjarchi, 2017-12-08:
> Maybe read the readme ? 😂

I wasn't going to point that out 😛

bjarchi, 2017-12-09:
@locutus40 Can you close this issue please?

locutus40, 2017-12-09:
why what if someone else has the same issue.. they might want to comment

locutus40, 2017-12-09:
i sure do love marlin posters there condecending tones and there rudeness...

closed 2017-12-09 by locutus40

thinkyhead, 2017-12-10:
> i sure do love marlin posters…

Apologies, @locutus40 — Recently I posted a short paragraph on [Community Guidelines](http://marlinfw.org/docs/development/contributing.html#community-guidelines), but I admit it's hard to find.

> from what i see its 1.0.2.0 then 1.1.6 im confused the first one says latest yet its older

Yes, that suddenly occurred, and I'm not sure why! Maybe a commit was made to 1.0.x accidentally, then reverted, then fixed, or something, causing 1.0.2-2 to pop up to the top of the Releases page. Anyway, when 1.1.7 comes out it will get moved to the top — hopefully soon.

Cheers, and I hope nothing is too broken for you during this very active development period.