MarlinFirmware/Marlin #8700: [2.0.x] Homing broken by recent commit

cg9999, 2017-12-07:
On my delta kossel the recent commit
"Use block cleaning instead of split flag" (commit 840289e7ccaa79846acfdceee63ffc1baa77168c)
broke the homing function.

Easiest to see if I manually move each carriage down, but position X carriage the lowest.
Now when I do G28, Y and Z seem to home but X stops as soon as other carriage hits the end stop.
X stays halfway, never hitting the endstop and machine thinks it's homed. Starting a print now will cause the head to hit the bed :(

AnHardt, 2017-12-07:
Do you get a message "Homing failed"?
Do B and C tower home normally - with backing up from endstops?

cg9999, 2017-12-07:
No fail message, and B and C do not back up like normally.
I can post a video if it helps

thinkyhead, 2017-12-08:
Thanks for the report! I believe it's patched now. We forgot we had to deal with interrupted split moves. Now we have a better way to deal with interrupted moves in the future, so, cheers.

Please give the latest bugfix a try and let us know if we got it.

cg9999, 2017-12-08:
Thank you, the issue is fixed now!

closed 2017-12-08 by cg9999