MarlinFirmware/Marlin #8701: UBL offsets Z incorrectly.

zarthcode, 2017-12-07:
### Bug Report
- Description: Upon creating a bed mesh, I cannot get Z to offset correctly. Then after using the prepare->move axis to move just 0.1mm in Z, it snaps back to place.  My Z probe offset is *perfect* after G28. (nozzle is 11.13mm from the table, in my case, probe offset is 1.13)

Maybe I'm missing something important?

- Expected behaviour: Compensation should return nozzle to correct height after G29 UBL probing.
- Actual behaviour: 
- Steps to reproduce:
```
G28   ; Home
G0 Z1;  jogging from here to 1mm over the table is correct.
G29 L1  ; UBL load profile
G29 J  ; UBL probing
G29 Z1; Nozzle Touches table!
```
What's interesting, jogging the Z axis from the menu at this point, up +0.1mm, moves the nozzle to 1.1mm over the table!  I've added my bed mesh, for reference (400x250 glass). I'm setting for homing again, but I've no way to know that UBL is working/active.

What am I doing wrong?

Bed Topography Report:
(0,11)                                                                                   (11,11)
(10,240)                                                                                (340,240)
 -0.970   -0.788   -0.507   -0.466   -0.264   -0.123   -0.082   -0.060   -0.079    0.042    0.044   -0.015

 -0.934   -0.687   -0.584   -0.423   -0.329   -0.291   -0.180   -0.121   -0.028    0.025    0.072    0.115

 -0.977   -0.730   -0.584   -0.641   -0.336   -0.351   -0.320   -0.214   -0.094    0.002    0.015    0.048

 -0.961   -0.687   -0.638   -0.421   -0.250   -0.199   -0.262   -0.225   -0.160   -0.034   -0.171    0.029

 -0.945   -0.641   -0.668   -0.299   -0.288   -0.248   -0.066   -0.183   -0.104   -0.037   -0.175    0.015

 -0.948   -0.664   -0.676   -0.336   -0.406  [-0.303]  -0.061   -0.228    0.030   -0.197   -0.198   -0.007

 -0.952   -0.729   -0.607   -0.406   -0.261   -0.194   -0.261   -0.184   -0.012   -0.021   -0.162   -0.019

 -0.896   -0.764   -0.608   -0.507   -0.253   -0.196   -0.210   -0.206   -0.078   -0.039   -0.148   -0.025

 -0.899   -0.562   -0.430   -0.442   -0.389   -0.364   -0.205   -0.033   -0.144   -0.047   -0.013   -0.004

 -0.883   -0.768   -0.624   -0.332   -0.307echo:busy: processing
   -0.315   -0.081   -0.080   -0.180   -0.250   -0.039   -0.025

 -0.887   -0.699   -0.671   -0.287   -0.423   -0.429   -0.062   -0.252   -0.062   -0.084   -0.035   -0.030

 -0.850   -0.685   -0.488   -0.441   -0.403   -0.340   -0.245   -0.124   -0.192   -0.030   -0.017   -0.015
(10,10)                                                                                    (340,10)
(0,0)                                                                                     (11,0)

This is on 1.1.x bugfix, last updated yesterday
Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1540109/Marlin.zip)

Roxy-3D, 2017-12-07:
The Z offset command has been removed from UBL for a while now....
```
   *   P6    Phase 6    Shift Mesh height. The entire Mesh's height is adjusted by the height specified
   *                    with the C parameter. Being able to adjust the height of a Mesh is useful tool. It
   *                    can be used to compensate for poorly calibrated Z-Probes and other errors. Ideally,
   *                    you should have the Mesh adjusted for a Mean Height of 0.00 and the Z-Probe measuring
   *                    0.000 at the Z Home location.
   *
   *   Q     Test       Load specified Test Pattern to assist in checking correct operation of system. This
   *                    command is not anticipated to be of much value to the typical user. It is intended
   *                    for developers to help them verify correct operation of the Unified Bed Leveling System.
   *
   *   R #   Repeat     Repeat this command the specified number of times. If no number is specified the
   *                    command will be repeated GRID_MAX_POINTS_X * GRID_MAX_POINTS_Y times.
   *
   *   S     Store      Store the current Mesh in the Activated area of the EEPROM. It will also store the
   *                    current state of the Unified Bed Leveling system in the EEPROM.
   *
   *   S #   Store      Store the current Mesh at the specified location in EEPROM. Activate this location
   *                    for subsequent Load and Store operations. Valid storage slot numbers begin at 0 and
   *                    extend to a limit related to the available EEPROM storage.
   *
   *   S -1  Store      Store the current Mesh as a print out that is suitable to be feed back into the system
   *                    at a later date. The GCode output can be saved and later replayed by the host software
   *                    to reconstruct the current mesh on another machine.
   *
   *   T     Topology   Display the Mesh Map Topology.
   *                    'T' can be used alone (e.g., G29 T) or in combination with most of the other commands.
   *                    This option works with all Phase commands (e.g., G29 P4 R 5 T X 50 Y100 C -.1 O)
   *                    This parameter can also specify a Map Type. T0 (the default) is user-readable. T1 can
   *                    is suitable to paste into a spreadsheet for a 3D graph of the mesh.
   *
   *   U     Unlevel    Perform a probe of the outer perimeter to assist in physically leveling unlevel beds.
   *                    Only used for G29 P1 T U. This speeds up the probing of the edge of the bed. Useful
   *                    when the entire bed doesn't need to be probed because it will be adjusted.
   *
   *   V #   Verbosity  Set the verbosity level (0-4) for extra details. (Default 0)
   *
   *   W     What?      Display valuable Unified Bed Leveling System data.
   *
   *   X #              X Location for this command
   *
   *   Y #              Y Location for this command
   *
   *
   *   Release Notes:
   *   You MUST do M502, M500 to initialize the storage. Failure to do this will cause all
   *   kinds of problems. 
```
May I ask what the date of your firmware is?  It would be helpful for you to switch to the latest bugfix-v1.1.x

zarthcode, 2017-12-07:
Just compiled from git yesterday from that very branch.

I haven't tried the mesh shift, I followed the guide here: (http://marlinfw.org/docs/features/unified_bed_leveling.html) as it was the best I could find.

Roxy-3D, 2017-12-07:
Thanks!!!   I'll have to get the updated.

bjarchi, 2017-12-07:
Is there something in the docs that no longer reflects the current state of development? Since I wrote the latest revisions to that page, still even have an issue open for it, it would be very quick for me to update any incorrect information.

Roxy-3D, 2017-12-07:
It was decided that the G29 Z command was just duplicate functionality and should be removed.

By setting the Z_PROBE_OFFSET everything could be accomplished that way.   (And as a fall back... The G29 P6 mesh adjust is still in place.)

The big problem is we don't want too many ways to accomplish the same thing.  It just causes confusion and makes testing that much harder.

bjarchi, 2017-12-07:
Ok I think we're fine - I don't mention `G29 Z` in that page anywhere that I can find. 

@zarthcode The last line of your code to reproduce is given as this:
`G29 Z1; Nozzle Touches table!` Did you mean `G1 Z1` (or, equivalently, `G0 Z1`) to move to z=1?

Roxy-3D, 2017-12-07:
@bjarchi   Can we change the table slightly?  It might require adding a second line but change this:

Command | Description
-- | --
G29 P1 | Phase 1 – Automatically probe the bed.
G29 P2 | Phase 2 – Manually probe points that automated probing couldn’t reach.  
                               Usually not needed.    Usually G29 P3 is sufficient to fill the mesh.

bjarchi, 2017-12-07:
> Can we change the table slightly? It might require adding a second line but change this:

Certainly! Just to check - you're referring to the actual table of commands, not the 'quick start' list right? (Either way the answer is yes - I can also clarify the quick-start snippets a bit too). 

I was hoping to make some progress on #8676 first, but the most recent commits on bugfix seem to have broken homing so I'm kind of stuck there.

Also can you take a look at #8702 quickly? I think it should be a trivial change to merge, given that it's already in bugfix-2.0.x.

Roxy-3D, 2017-12-07:
> Also can you take a look at #8702 quickly? I think it should be a trivial change to merge, given that it's already in bugfix-2.0.x.

I don't understand what the intent of the _MIN_PROBE_X/Y numbers are.    It seems like the version without the underscore should be sufficient.   I better wait for ThinkyHead to merge it.

bjarchi, 2017-12-07:
My understanding was that the _... versions were to define the defaults, and then the following #ifdef blocks set the main variables to the defaults if (and only if) the user didn't override these defines in their configuration files. It's bringing the behavior into line with how things are done in 2.0.x (at least bugfix 2.0.x) and in the following block with _MESH_... 

Not pushing you to merge it, just wanted to explain my thinking :)

zarthcode, 2017-12-08:
@bjarchi  Yeah, I meant G0 Z1.

closed 2018-02-10 by thinkyhead