MarlinFirmware/Marlin #8703: Unable to resume print when pausing from LCD if PARK_HEAD_ON_PAUSE enabled with PAUSE_PARK_RETRACT_LENGTH > 0

marcio-ao, 2017-12-07:
- Description: Unable to resume print when pausing from LCD if PARK_HEAD_ON_PAUSE enabled with PAUSE_PARK_RETRACT_LENGTH > 0
- Expected behaviour: After starting a print from the LCD, it should be possible to pause it and resume the print from the LCD.
- Actual behaviour: The LCD shows "PRINT PAUSED; Heating nozzle; Please wait..." and there is no way to return to the menu to resume the print.
- Steps to reproduce:
  - Enable PARK_HEAD_ON_PAUSE
  - Leave PAUSE_PARK_RETRACT_LENGTH at default value of 2
  - Start print from SD card
  - Pause print from LCD menu

I've traced the sequence of events that leads to this error:
  - lcd_sdcard_pause() enqueues as "M125" (see code [here](https://github.com/MarlinFirmware/Marlin/blob/86b65e52c4dd72cf55a3f943fd861602d8421772/Marlin/ultralcd.cpp#L780))
  - gcode_M125() calls pause_print() with a value of negative PAUSE_PARK_RETRACT_LENGTH (see code [here](https://github.com/MarlinFirmware/Marlin/blob/86b65e52c4dd72cf55a3f943fd861602d8421772/Marlin/Marlin_main.cpp#L8537))
  - pause_print() calls ensure_safe_temperature() (see code [here](https://github.com/MarlinFirmware/Marlin/blob/86b65e52c4dd72cf55a3f943fd861602d8421772/Marlin/Marlin_main.cpp#L6466))
  - ensure_safe_temperature() calls lcd_advanced_pause_show_message() (see code [here](https://github.com/MarlinFirmware/Marlin/blob/86b65e52c4dd72cf55a3f943fd861602d8421772/Marlin/Marlin_main.cpp#L6430))

The lcd_advanced_pause_show_message is a modal dialog box that cannot be dismissed by the user, hence the user cannot recover from the pause.

jaysonkelly, 2018-01-24:
I'm having the same issue. What do you think needs to be done to correct the issue? Can the temperature message be a simple status message on the lcd instead of a modal dialog box? Will it work correctly if I change the retract length to 0?

jaysonkelly, 2018-01-24:
It does work correctly if I change PAUSE_PARK_RETRACT_LENGTH to 0

jaysonkelly, 2018-01-24:
It looks like it wouldn't enter the heating screen if my temp is within the TEMP_HYSTERESIS zone. I'm going to try increasing my TEMP_HYSTERESIS value to better match my extruder temp variation.

jaysonkelly, 2018-01-24:
That worked as well. Seems like I need to work on my PID tuning. Regardless, there is a bug here because if it does get stuck in the waiting to heat screen, the heat turns off and it just gets stuck there.

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

jaysonkelly, 2018-02-14:
I no longer get stuck in the modal heating window when pausing outside my hysteresis zone but it seems to under extrude for a time after the mandatory purge that happens upon resume. Is this related?

jaysonkelly, 2018-02-14:
Dialing the purge back to 2mm made a big difference but when printing a small tower the under-extrusion layer is still very visible. Is it just my hotend mechanics or is it a software issue?

jaysonkelly, 2018-02-14:
It is probably bad etiquette to mention other bugs while in this thread, but when I pause the print then cancel it then start a new one, the printer returns to the paused point before beginning to heat up. There's probably some variables that need to be cleared when you cancel a print while it is paused.

marcio-ao, 2018-02-15:
@thinkyhead : I have confirmed that the issue with not being able to resume a print is fixed in 1.1.8.

thinkyhead, 2018-02-23:
> pause the print then cancel it then start a new one, the printer returns to the paused point

@jaysonkelly — I'll look into that now…

thinkyhead, 2018-02-23:
@jaysonkelly Do you have `SDSUPPORT` enabled? Are you pausing an SD print or pausing a print from the host? Do both host and SD printing exhibit the same behavior on starting the next print?

jaysonkelly, 2018-02-23:
I am pausing an sd print from the LCD controller. I haven't tried pausing
from the host.

thinkyhead, 2018-02-23:
See if these PRs help.

jaysonkelly, 2018-02-23:
Ok. I'll test when I can.

PR #9780 (2018-02-23): [2.0.x] Un-pause print on cancellation
Fix #8703

Clear `did_pause_print` when a print is canceled so that `resume_print` won't move the head back on the next print.
PR #9779 (2018-02-23): [1.1.x] Un-pause on print cancellation
Fix #8703

Clear `did_pause_print` when a print is canceled so that `resume_print` won't move the head back on the next print.
closed 2018-02-23 by thinkyhead