MarlinFirmware/Marlin #8704: [bugfix-1.1.6] Homing broken by latest commits

bjarchi, 2017-12-07:
Probably similar to #8700 but I'm on a cartesian machine and I don't know how much difference there is in terms of homing behavior.

### Bug Report
- Description: After pulling latest commits on bugfix-1.1.x, homing (w/ Z_SAFE_HOMING, though I don't think it matters) no longer works properly
- Expected behaviour: on `G28` machine homes X and Y, then moves to center of bed and homes Z using bed probe
- Actual behaviour: Depends on starting location. If starting from the endstops, the machine moves immediately to bed center (no homing re-bump) and seems to kind of do the right thing. If not starting from endstops, machine starts homing normally but doesn't re-bump after hitting endstops, doesn't always hit both endstops, and tries to home Z wherever it ends up. Repeated `G28`s seem to produce additive error in home reference frame. 
- Steps to reproduce: 
  - Flash latest commits
  - Initialize EEPROM (M502; M500; M501)
  - `G28` and watch behavior

[config-20171207.zip](https://github.com/MarlinFirmware/Marlin/files/1540520/config-20171207.zip)

Tannoo, 2017-12-07:
I can't find `bugfix-1.1.6` to check this out.

bjarchi, 2017-12-07:
My bad, bugfix-1.1.x

The main bugfix branch for 1.1. development.

thinkyhead, 2017-12-08:
We can have one topic for both 1.1 and 2.0. Make it #8700.

closed 2017-12-08 by thinkyhead