MarlinFirmware/Marlin #8709: Compile error with SKEW_CORRECTION

DaveAhrendt, 2017-12-08:
Compiling with Arduino IDE 1.9.0-beta and Marlin bugfix 2.0 from 12/6.
Was running well and decided to see what skew correction was about.  Just enabling #define SKEW_CORRECTION with default values causes a compile error or more likely a linker error.

C:\Users\Dave\AppData\Local\Temp\ccWPadut.ltrans8.ltrans.o: In function `MarlinSettings::save()':
ccWPadut.ltrans8.o:(.text+0x706): undefined reference to `Planner::xy_skew_factor'
ccWPadut.ltrans8.o:(.text+0x708): undefined reference to `Planner::xy_skew_factor'
ccWPadut.ltrans8.o:(.text+0x71e): undefined reference to `Planner::xz_skew_factor'
ccWPadut.ltrans8.o:(.text+0x720): undefined reference to `Planner::xz_skew_factor'
ccWPadut.ltrans8.o:(.text+0x736): undefined reference to `Planner::yz_skew_factor'
ccWPadut.ltrans8.o:(.text+0x738): undefined reference to `Planner::yz_skew_factor'
collect2.exe: error: ld returned 1 exit status
Using library U8glib at version 1.19.1 in folder: C:\Users\Dave\Documents\Arduino\libraries\U8glib 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

HekTron802, 2017-12-08:
Just a heads up that Skew isn't working with UBL enabled (I'm using 1.1.X_bugfix and it's compiling fine). 
See #8632

thinkyhead, 2017-12-09:
The issue shown here should now be fixed. A fix for skew with UBL is pending at #8729

closed 2017-12-09 by thinkyhead