MarlinFirmware/Marlin #8723: 1.1.6 Nozzle hits bed after moving outside of mesh area

Festivejelly, 2017-12-09:
------------------------------------------------------------
### Bug Report
- Description: Nozzle hits the bed after telling it to move outside of the probable area
- Expected behaviour: nozzle should stay at the same height
- Actual behaviour: nozzle hits the bed
- Steps to reproduce:
  - G29 P1 
  - Should yeild a mesh (I cant probe my full bed so I end up with a row of NAN's
 - Do G29 P3 to get rid of the nans
- Now you should have a mesh
- Home
- move nozzle outside of probed mesh area
- nozzle hits the bed
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1544161/Marlin.zip)

Here is my initial report after levelling

 Bed Topography Report for CSV:
 0.764,0.815,0.747,0.696,0.661,0.630,0.664,0.624,0.687,0.787
 0.775,0.821,0.753,0.715,0.678,0.639,0.662,0.622,0.668,0.753
 0.762,0.822,0.747,0.726,0.687,0.641,0.662,0.638,0.667,0.748
 0.673,0.734,0.678,0.669,0.632,0.598,0.630,0.597,0.624,0.712
 0.653,0.728,0.668,0.655,0.606,0.584,0.606,0.581,0.600,0.688
 0.613,0.706,0.652,0.638,0.607,0.576,0.607,0.582,0.625,0.698
 0.640,0.723,0.676,0.666,0.627,0.610,0.637,0.600,0.632,0.696
 0.652,0.753,0.712,0.707,0.671,0.645,0.675,0.637,0.678,0.743
 0.696,0.796,0.763,0.747,0.723,0.697,0.717,0.683,0.736,0.798
 0.718,0.834,0.791,0.779,0.752,0.727,0.749,0.725,0.778,0.868

Here is the UBL status:

2:45:54.761: Unified Bed Leveling System v1.01 active.
2:45:54.811: Mesh 1 Loaded.
2:45:54.864: UBL object count: 1
2:45:54.864: planner.z_fade_height : 10.0000
2:45:54.872: # of samples: 100
2:45:54.872: Mean Mesh Height: 0.689300
2:45:54.876: Standard Deviation: 0.064814
2:45:54.876: zprobe_zoffset: -0.6000000
2:45:54.880: UBL_MESH_MIN_X 10=10
2:45:54.880: UBL_MESH_MIN_Y 10=10
2:45:54.909: UBL_MESH_MAX_X (300 - 50)=250
2:45:54.913: UBL_MESH_MAX_Y (300 - (10+5))=285
2:45:54.937: GRID_MAX_POINTS_X 10
2:45:54.941: GRID_MAX_POINTS_Y 10
2:45:54.966: MESH_X_DIST 26.67
2:45:54.970: MESH_Y_DIST 30.56
2:45:55.254: X-Axis Mesh Points at: 10.000 36.667 63.333 90.000 116.667 143.333 170.000 196.667 223.333 250.000
2:45:55.516: Y-Axis Mesh Points at: 10.000 40.556 71.111 101.667 132.222 162.778 193.333 223.889 254.444 285.000
2:45:55.569: ubl_state_at_invocation :0
2:45:55.573: ubl_state_recursion_chk :0
2:45:55.622: Meshes go from 0x0270 to 0x0F7F
2:45:55.675: sizeof(ubl) : 1
2:45:55.675: z_value[][] size: 400
2:45:55.704: EEPROM free for UBL: 0x0D0F
2:45:55.757: EEPROM can hold 8 meshes.
2:45:55.786: Unified Bed Leveling sanity checks passed.

One of the things I know is my mesh is smaller than my bed. I had to do that else I was getting NAN on the edges so I figured if I reduce the size of the mesh the probe will probe as far as it can.

My assumption is that when it moves outside the mesh this causes an issue. But then again I might be wrong.
If I am correct, what is the best solution? Keep the mesh size 300 x 300 and just do G29 P3 to fill the values? I guess Thought I was being smart by stretching the mesh.

I've attached my configs

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1544293/Marlin.zip)

```

thinkyhead, 2017-12-09:
Do you see the same issue when running `bugfix-1.1.x`?
Please test in case we fixed the issue since 1.1.6 was released.

Festivejelly, 2017-12-09:
I will test the bugfix when i get up in the morning :)

Roxy-3D, 2017-12-09:
Some changes to UBL's mesh motion were made yesterday.   The purpose of the changes were to make the behavior better when the nozzle is going off the mesh.  It is very possible those changes will help you.

> If I am correct, what is the best solution? Keep the mesh size 300 x 300 and just do G29 P3 to fill the values? I guess Thought I was being smart by stretching the mesh.

You do wan't the mesh to cover the entire area that you will be printing on.   And doing a fill with G29 P3 is fine.   That is what it is there for.    But you should expect the filled in mesh points to need a little more 'editing' to be correct than the auto-probed points when you do a G26 and G29 P4 R cycle.

PS.  Your bed is remarkably flat.

bjarchi, 2017-12-09:
I made the same mistake when I first set up UBL - it's smarter than I am :) 

You definitely want your mesh to cover the printable area, as Roxy said. The suite of tools to fill in and edit the mesh from where you can probe makes it pretty easy to get a good mesh even in non-probable areas. 

> PS. Your bed is remarkably flat.

It is, but I'm a bit worried about the consistent mesh offset of about 0.7 mm - shouldn't the mean mesh height be near zero with properly configured z-probe offset and etc? I know mine usually is, and the one time I misconfigured something and had a large mesh mean offset it caused strange problems.

Roxy-3D, 2017-12-09:
Yes...   This is coming from an error in the Z_PROBE_OFFSET_FROM_EXTRUDER number.   Ideally, where ever you home should have those mesh points very close to 0.000 mm.   And I prefer to home in the center of the bed, so with that at 0.000  you can get a good idea of what is happening at various edges or corners.

bjarchi, 2017-12-09:
> Yes... This is coming from an error in the Z_PROBE_OFFSET_FROM_EXTRUDER number.

Yeah, so this is probably *not* a bug then - the observed behavior comes from UBL 'hiding' the incorrect probe offset over its active area, and once the tool head goes outside of the mesh area the incorrect probe height is no longer compensated and results in a head crash. 

I wonder if it would be useful to have a warning printed to the LCD/console during mesh generation if UBL detects a grid mean height that's too far from zero - something to alert the user that they may have an incorrect probe offset (or etc.). Unfortunately it might be difficult to define 'too far' from zero - since a bed that has a bowl- or dome-shaped contour centered around (0,0) will also have a net negative or positive mean mesh height.

Festivejelly, 2017-12-09:
I download the bugfix and i'm currently doing some probing. It cant help but feel would it be better for the nozzle to keep its current height while moving out of the mesh instead of instantly hitting the bed?
I mean I wouldnt have seen the behaviour I guess if my mesh wasnt so inset. I was just trying to get rid of the NAN values, but I now know its better to do a G29 P3 to get rid.

Festivejelly, 2017-12-09:
I suppose ultimately I'll want to change my probe to a Piezoelectric so that I dont have any offsets.

Festivejelly, 2017-12-09:
I wondered about why the bed was always showing higher than 0. But i set the offset to 0 for Z height then changed it using M851 Z{value} by doing the ole paper test. Its currently set to Z-0,55

bjarchi, 2017-12-09:
> I download the bugfix and i'm currently doing some probing. It cant help but feel would it be better for the nozzle to keep its current height while moving out of the mesh instead of instantly hitting the bed?

That may be the behavior now; the idea though is that if UBL can't correct an area (outside the mesh) then correction defaults to 0 / is not applied. This makes sense, but it is a problem if for some reason your mesh has a large constant offset that is masking another problem.

>I wondered about why the bed was always showing higher than 0. But i set the offset to 0 for Z height then changed it using M851 Z{value} by doing the ole paper test. Its currently set to Z-0,55

Is it possible you've got a sign error there? What kind of probe are you using? And when you did the paper test, did you get the appropriate contact at z=0.55 or z=-0.55?

Festivejelly, 2017-12-09:
So I'm using a capacitive probe which is offset to the left of the nozzle.  I basically left the Z offset at zero then I did a home in the centre of the bed (safe z on)
And it basically moves down until the sensor triggers and then it stops. So then I move down the Z axis using repetier until it hits 0.
The nozzle was still a bit height so i used M851 Z-Number until I got the correct gap (around a sheet of paper) this number turned out to be -0.55

This is the way ive always done it but its not working now. I wonder if the probe is behaving differently to my previous probe which was a BL touch. I decided to move away from it because I wasnt happy with how that probe worked.

Festivejelly, 2017-12-09:
Right I think it know whats happening. When I do home, itle home fine and then go back up to 10mm height. But then when I try to go to 0 the capacitive sensor triggers and it stops it actually going down. (kinda like it being used as an endstop) I assume thats why it thinks its 0.7 mm higher. Any tips on what to do in order to stop this? With the BL touch I didnt notice this because it stows the probe when in use so it would never trigger when moving down. But the capacitive sensor is "always on" so triggers when it gets too close to the bed.

Roxy-3D, 2017-12-09:
After you home...   If you move the nozzle to the home location....  You should be able to (very carefully and slowly) lower the nozzle until it just barely touches the bed.   The value on the LCD (or with M114 if you don't have an LCD Display) should be very very close to 0.000 mm

If it isn't, UBL will still operate.   But it gets really hard to figure out what is wrong and how to correct things when your whole mesh is offset.

Festivejelly, 2017-12-09:
Ill try that right now. I'll need to use the encoder wheel to lower it as repetier doesnt let me doing it.  Its going to show minus something when its just above the bed, because the offset is set to 0.

Roxy-3D, 2017-12-09:
You can use PronterFace to lower it .1mm at a time.   That is what I use.   But how ever you do it, it will be fine.   For that matter, you can use G1 Z5.0    And then G1 Z4.5   And then G1 Z4.0   until it is where you want it.

> I wonder if it would be useful to have a warning printed to the LCD/console during mesh generation if UBL detects a grid mean height that's too far from zero - something to alert the user that they [...]

We can do what ever makes sense.   I'm leaning more toward not allowing the nozzle to move off the mesh during a print.    If the nozzle does move off the mesh, the printer gets shutdown.

Festivejelly, 2017-12-09:
Hrm it wont go down with the encoder wheel i'll try the G commands.

Festivejelly, 2017-12-09:
Nope, it wont go below 0. Strange since i have software endstops enabled

Festivejelly, 2017-12-09:
Doh! they need to be disabled 🙄

Roxy-3D, 2017-12-09:
With any of the mesh bed leveling systems...  You really ought to allow the nozzle to go below 0.000 mm.    Otherwise, you can't correct for dips in the bed.

Festivejelly, 2017-12-09:
Yeah I've corrected this. Its because I updated the firmware and its enabled by default.

Festivejelly, 2017-12-09:
Wow something has gone wrong in one of these updates :-/

So now when I level the bed and I did a mesh it somehow managed to probe all areas (weird since the probe cant physically get to them)

23:30:14.847: Unified Bed Leveling System v1.01 inactive.
23:30:14.896: No Mesh Loaded.
23:30:14.950: UBL object count: 1
23:30:14.951: planner.z_fade_height : 10.0000
23:30:14.954: # of samples: 100
23:30:14.958: Mean Mesh Height: 0.000000
23:30:14.962: Standard Deviation: 0.000000
23:30:14.963: zprobe_zoffset: -0.5000000
23:30:14.971: MESH_MIN_X ((((((300) / 2) - (300) / 2) + 10)>(0 + -65)?((((300) / 2) - (300) / 2) + 10):(0 + -65)))=10
23:30:14.983: MESH_MIN_Y ((((((300) / 2) - (300) / 2) + 10)>(0 + -15)?((((300) / 2) - (300) / 2) + 10):(0 + -15)))=10
23:30:15.016: MESH_MAX_X ((((((300) / 2) + (300) / 2) - (10))<(300 + 15 + -65)?((((300) / 2) + (300) / 2) - (10)):(300 + 15 + -65)))=250
23:30:15.028: MESH_MAX_Y ((((((300) / 2) + (300) / 2) - (10))<(300 + -15)?((((300) / 2) + (300) / 2) - (10)):(300 + -15)))=285
23:30:15.053: GRID_MAX_POINTS_X 10
23:30:15.053: GRID_MAX_POINTS_Y 10
23:30:15.081: MESH_X_DIST 26.67
23:30:15.081: MESH_Y_DIST 30.56
23:30:15.368: X-Axis Mesh Points at: 10.000 36.667 63.333 90.000 116.667 143.333 170.000 196.667 223.333 250.000
23:30:15.626: Y-Axis Mesh Points at: 10.000 40.556 71.111 101.667 132.222 162.778 193.333 223.889 254.444 285.000
23:30:15.682: Unified Bed Leveling sanity checks passed.

When i tried printing a test pattern that I use it started to print okay (albeit a bit low) but then when it got passed the probably area on the X axis it raised the Z by an infinite amount. I had to turn off the printer to stop it.

Festivejelly, 2017-12-09:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1545209/Marlin.zip)

Ive added my whole marlin folder. I wonder if the tripple probe thing is affecting this. Its a very peculiar thing to happen.

Festivejelly, 2017-12-09:
Looks like the mesh didnt save properly. Ill re probe and try again. Arrrg. Still dont see why it raised up infinitely though.

thinkyhead, 2017-12-10:
When you have a chance, do a probe log with `DEBUG_LEVELNG_FEATURE` plus `M111 S32` before `G29`. That might give us some indication where things are going wrong.

Festivejelly, 2017-12-10:
Okay so heres what we have with a fresh probing,

UBL Status:

Unified Bed Leveling System v1.01 active. (2)
Mesh 1 Loaded.
UBL object count: 1
planner.z_fade_height : 10.0000
# of samples: 100
Mean Mesh Height: 0.101233
Standard Deviation: 0.066597
zprobe_zoffset: -0.5000000
MESH_MIN_X ((((((300) / 2) - (300) / 2) + 10)>(0 + -65)?((((300) / 2) - (300) / 2) + 10):(0 + -65)))=10
MESH_MIN_Y ((((((300) / 2) - (300) / 2) + 10)>(0 + -15)?((((300) / 2) - (300) / 2) + 10):(0 + -15)))=10
MESH_MAX_X ((((((300) / 2) + (300) / 2) - (10))<(300 + 15 + -65)?((((300) / 2) + (300) / 2) - (10)):(300 + 15 + -65)))=250
MESH_MAX_Y ((((((300) / 2) + (300) / 2) - (10))<(300 + -15)?((((300) / 2) + (300) / 2) - (10)):(300 + -15)))=285
GRID_MAX_POINTS_X 10
GRID_MAX_POINTS_Y 10
MESH_X_DIST 26.67
MESH_Y_DIST 30.56
X-Axis Mesh Points at: 10.000 36.667 63.333 90.000 116.667 143.333 170.000 196.667 223.333 250.000
Y-Axis Mesh Points at: 10.000 40.556 71.111 101.667 132.222 162.778 193.333 223.889 254.444 285.000
Unified Bed Leveling sanity checks passed.

Mesh:

Bed Topography Report for CSV:
0.168,0.204,0.140,0.091,0.062,0.029,0.057,0.038,0.112,0.213
0.179,0.212,0.147,0.112,0.086,0.058,0.080,0.058,0.119,0.215
0.163,0.204,0.140,0.117,0.087,0.058,0.062,0.051,0.096,0.184
0.068,0.121,0.061,0.057,0.032,0.010,0.035,0.012,0.059,0.147
0.053,0.109,0.064,0.043,0.020,-0.002,0.014,0.007,0.029,0.126
0.016,0.085,0.047,0.044,0.015,-0.010,0.020,0.000,0.048,0.130
0.038,0.101,0.065,0.050,0.038,0.027,0.048,0.033,0.067,0.140
0.064,0.152,0.114,0.110,0.079,0.058,0.088,0.064,0.115,0.177
0.098,0.192,0.157,0.145,0.130,0.113,0.127,0.113,0.171,0.240
0.162,0.263,0.223,0.203,0.181,0.157,0.160,0.145,0.213,0.301

As I mentioned before i'm not getting a NAN now where the probe cant probe.

This is a problem because basically this means I can now only print in a probable part of the bed, which wasnt the case before. In the past I was able to manually enter these mesh values when they came up as NAN. I think there was a change to the Max mesh value and how its calculated:

MESH_MAX_X ((((((300) / 2) + (300) / 2) - (10))<(300 + 15 + -65)?((((300) / 2) + (300) / 2) - (10)):(300 + 15 + -65)))=250
MESH_MAX_Y ((((((300) / 2) + (300) / 2) - (10))<(300 + -15)?((((300) / 2) + (300) / 2) - (10)):(300 + -15)))=285

Looks like its deducting the offsets for the probe. In the past it would just report them as nan, you'd do the ole G29 P3 and happy days. Now you cant even get to that point.

Festivejelly, 2017-12-10:
Ill do the probe now

Festivejelly, 2017-12-10:
Looks like its changed to this:

``
#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Override the mesh area if the automatic (max) area is too large
  //#define MESH_MIN_X MESH_INSET
  //#define MESH_MIN_Y MESH_INSET
  //#define MESH_MAX_X X_BED_SIZE - (MESH_INSET)
  //#define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)
#endif``

Festivejelly, 2017-12-10:
I see this being an issue for people such as myself that have quite a large offset for their probe.

Festivejelly, 2017-12-10:
Right... theres something seriously wrong with the mesh system. I probed the bed and did some corrections using G29 P4. Saved the mesh and started to navigate around the bed to see if it worked then all of a sudden the Z axis again started raising up infinity. Interestingly when i tried a test print earlier I got that similar behaviour that people have mentioned before where the Left side of the bed was lower than the right. And so anything on the left was squished and the right was too high.

I downloaded and configured repetier firmware and used their probing feature and im not getting that issue. I did it more as a test than anything else just to make sure it wasnt a mechanical issue.

bjarchi, 2017-12-10:
> Looks like its changed to this:
#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_UBL)
// Override the mesh area if the automatic (max) area is too large
//#define MESH_MIN_X MESH_INSET
//#define MESH_MIN_Y MESH_INSET
//#define MESH_MAX_X X_BED_SIZE - (MESH_INSET)
//#define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)
#endif
I guess with them being commented out by default its now shrinking the mesh.

You almost certainly want those set to the default - that's how you let your mesh area cover (almost) all of your bed. The UBL system will figure out which areas can be probed base on the x,y probe offsets, and only probe a subset of the mesh. Then you use `G29 P3` (perhaps repeatedly) to auto-fill the parts of the mesh that couldn't be probed. 

This should still be how the system works. I participated in some UBL debugging recently and haven't had issues like this - but I'll go through the mesh probing and fill process tomorrow to be sure. 

> When i tried printing a test pattern that I use it started to print okay (albeit a bit low) but then when it got passed the probably area on the X axis it raised the Z by an infinite amount. I had to turn off the printer to stop it.

This really sounds like you've managed to define your mesh so that it's smaller than the bed area (perhaps only the probable area?) - which is not what you want with UBL. 

Can you post your current Configuration.h and Configuration_adv.h here so we can make sure this isn't a config issue?

bjarchi, 2017-12-10:
> I see this being an issue for people such as myself that have quite a large offset for their probe.

BTW My X probe offset is pretty large (-71) and I haven't run into this issue - so unless it's new in the past day or so I'm still guessing we can sort it with configuration changes. 

> Looks like the mesh didnt save properly. Ill re probe and try again. Arrrg. Still dont see why it raised up infinitely though.

This sometimes happens when you move outside the mesh area - I've seen it happen in another case where get_z_correction was asking for an out-of-range point. This is why you want your mesh to fill your bed. May be a bug, but as Roxy said the idea of UBL is to *never* move outside the mesh.

(re: Roxy)

> We can do what ever makes sense. I'm leaning more toward not allowing the nozzle to move off the mesh during a print. If the nozzle does move off the mesh, the printer gets shutdown. 

I'm starting to lean in that direction too. I suspect it will create lots of headaches (for us!) with new issues when it goes live, but it will force people to have the system configured correctly. If the goal of UBL is (as I understand) to have a mesh that covers the whole bed, so you never leave the mesh, then leaving the mesh (on a printing move) is a bug. 

My only concern there is the MESH_INSET - it means using UBL sacrifices some ~10 mm of bed size in each axis due to mesh insetting. We may need to go with defaulting to mesh covers the bed but probing is inset by some margin to avoid probing too near the edge?

Festivejelly, 2017-12-10:
@bjarchi So my co ordinate system is 0,0 which would be the left corner most nearest to you if you were standing at the printer:

x-------x
|           |
|           |
x-------x
^ here

I suppose the trick really is to use no inset if possible and continue to fill out the values that it cant probe.

Here are my configs:

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1545813/Marlin.zip)

bjarchi, 2017-12-10:
> @bjarchi So my co ordinate system is 0,0 which would be the left corner most nearest to you if you were standing at the printer:

OK that's good. That, and safe homing, are both as set up on my machine. I have MESH_INSET defined as 6 mm, so I lose a total of 12mm in each direction as un-meshed area. So far not a problem for me. You do want mesh inset to be small - just enough to make sure you won't accidentally probe off the edge of the bed on the edges your probe *can* reach (and to avoid clips, etc.)

I'll check out your configuration, and make sure I can still probe and fill a dense mesh as expected. Can you try the MESH_[MIN,MAX] defines I posted above to see if that helps you? If so it suggests that we need to re-evaluate the logic that computes the defaults for those. 

@Roxy-3D Have a look at how MESH_[MIN,MAX] is computed in conditionals_post.h and see what you think. They're computed (for UBL and MBL [mesh leveling]) based on [MIN,MAX]_PROBE_ defines, which in turn may depend on [X,Y]_PROBE_OFFSET_FROM_EXTRUDER. I suspect this may not be what we want. May need a separate case from MBL?

Festivejelly, 2017-12-10:
Its already set as I changed them back to what they were before:

19:39:34.857: Unified Bed Leveling System v1.01 inactive.
19:39:34.906: Mesh 0 Loaded.
19:39:34.959: UBL object count: 1
19:39:34.963: planner.z_fade_height : 10.0000
19:39:34.967: # of samples: 72
19:39:34.967: Mean Mesh Height: 0.063426
19:39:34.971: Standard Deviation: 0.063981
19:39:34.971: zprobe_zoffset: 0.0000000
19:39:34.975: MESH_MIN_X 10=10
19:39:34.975: MESH_MIN_Y 10=10
19:39:35.004: MESH_MAX_X 300 - (10)=290
19:39:35.004: MESH_MAX_Y 300 - (10)=290
19:39:35.032: GRID_MAX_POINTS_X 10
19:39:35.032: GRID_MAX_POINTS_Y 10
19:39:35.062: MESH_X_DIST 31.11
19:39:35.062: MESH_Y_DIST 31.11
19:39:35.349: X-Axis Mesh Points at: 10.000 41.111 72.222 103.333 134.444 165.556 196.667 227.778 258.889 290.000
19:39:35.606: Y-Axis Mesh Points at: 10.000 41.111 72.222 103.333 134.444 165.556 196.667 227.778 258.889 290.000
19:39:35.663: Unified Bed Leveling sanity checks passed.

bjarchi, 2017-12-10:
Update: I just went through the process of probing a new dense mesh to check things out, and the behavior on my machine was as expected - `G29 P1` probed the points it could reach, leaving a column of unprobed locations, which a subsequent `G29 P3` filled in. 

Don't worry about the mesh quality; it was a quick probe with everything cold.

`G29 T` output after initial probe:
```
Recv: (0,6)                                           (6,6)
Recv: (6,180)                                        (208,180)
Recv:  -0.226   -0.277   -0.121   -0.039    0.013    0.048       .
Recv: 
Recv:  -1.524   -0.157   -0.042   -0.009    0.045    0.120       .
Recv: 
Recv:  -1.467   -0.127   -0.085   -0.001    0.046    0.152       .
Recv: 
Recv:  -1.447   -0.077   -0.053  [ 0.001]   0.049    0.161       .
Recv: 
Recv:  -1.410   -0.046   -0.015    0.038    0.069    0.156       .
Recv: 
Recv:  -1.315   -0.037    0.033    0.035    0.077    0.126       .
Recv: 
Recv:   0.154   -0.076    0.026    0.077    0.089    0.116       .
Recv: (6,6)                                            (208,6)
Recv: (0,0)                                             (6,0)
Recv: ok
```

`G29 T` output after doing `G29 P3`:
```
Recv: (0,6)                                           (6,6)
Recv: (6,180)                                        (208,180)
Recv:  -0.226   -0.277   -0.121   -0.039    0.013    0.048    0.082
Recv: 
Recv:  -1.524   -0.157   -0.042   -0.009    0.045    0.120    0.195
Recv: 
Recv:  -1.467   -0.127   -0.085   -0.001    0.046    0.152    0.258
Recv: 
Recv:  -1.447   -0.077   -0.053  [ 0.001]   0.049    0.161    0.273
Recv: 
Recv:  -1.410   -0.046   -0.015    0.038    0.069    0.156    0.243
Recv: 
Recv:  -1.315   -0.037    0.033    0.035    0.077    0.126    0.175
Recv: 
Recv:   0.154   -0.076    0.026    0.077    0.089    0.116    0.144
Recv: (6,6)                                            (208,6)
Recv: (0,0)                                             (6,0)
Recv: ok
```

And, for completeness, output of `G29 W`: 
```
Send: G29 W
Recv: Unified Bed Leveling System v1.01 active.
Recv: Mesh 1 Loaded.
Recv: UBL object count: 1
Recv: planner.z_fade_height : 10.0000
Recv: # of samples: 49
Recv: Mean Mesh Height: -0.113220
Recv: Standard Deviation: 0.454829
Recv: zprobe_zoffset: -1.7999999
Recv: MESH_MIN_X  6=6
Recv: MESH_MIN_Y  6=6
Recv: MESH_MAX_X  214 - (6)=208
Recv: MESH_MAX_Y  186 - (6)=180
Recv: GRID_MAX_POINTS_X  7
Recv: GRID_MAX_POINTS_Y  7
Recv: MESH_X_DIST  33.67
Recv: MESH_Y_DIST  29.00
Recv: X-Axis Mesh Points at: 6.000  39.667  73.333  107.000  140.667  174.333  208.000
Recv: Y-Axis Mesh Points at: 6.000  35.000  64.000  93.000  122.000  151.000  180.000
Recv: Kill pin on :41  state:1
Recv: 
Recv: Unified Bed Leveling sanity checks passed.
Recv: ok
```

bjarchi, 2017-12-10:
> Its already set as I changed them back to what they were before:

OK, so what happens if you try to probe and fill a mesh now? Given those numbers, you should get a partially filled mesh with `G29 P1` that you can fill with `G29 P3` - are we at least back to that point of sanity?

Did you change your configs after posting them? I noticed that your MESH_INSET in the UBL section is defined as 5, but your `G29 W` output shows it being 10... 

Might also be interesting to try omitting the `MULTIPLE_PROBING` define - that's a rather new addition and I haven't tested it. If things work better without it defined then we have an idea of where to look.

Festivejelly, 2017-12-10:
So pre P3 we have:

Bed Topography Report for CSV:
NAN,NAN,NAN,NAN,NAN,NAN,NAN,NAN,NAN,NAN
0.157,0.192,0.132,0.085,0.023,0.062,0.029,0.075,NAN,NAN
0.143,0.180,0.135,0.083,0.015,0.043,0.031,0.072,NAN,NAN
0.051,0.113,0.065,0.022,-0.040,0.005,-0.011,0.036,NAN,NAN
0.027,0.088,0.047,0.003,-0.058,-0.010,-0.033,0.012,NAN,NAN
-0.015,0.064,0.031,-0.007,-0.063,-0.009,-0.018,0.027,NAN,NAN
0.007,0.093,0.070,0.037,-0.031,0.013,-0.011,0.027,NAN,NAN
0.019,0.114,0.084,0.065,0.008,0.065,0.038,0.086,NAN,NAN
0.081,0.179,0.143,0.112,0.063,0.106,0.097,0.141,NAN,NAN
0.106,0.215,0.172,0.139,0.085,0.137,0.132,0.187,NAN,NAN

Post P3 we have:

Bed Topography Report for CSV:
0.172,0.205,0.132,0.087,0.032,0.080,0.029,0.078,0.127,NAN
0.157,0.192,0.132,0.085,0.023,0.062,0.029,0.075,0.121,NAN
0.143,0.180,0.135,0.083,0.015,0.043,0.031,0.072,0.113,NAN
0.051,0.113,0.065,0.022,-0.040,0.005,-0.011,0.036,0.083,NAN
0.027,0.088,0.047,0.003,-0.058,-0.010,-0.033,0.012,0.058,NAN
-0.015,0.064,0.031,-0.007,-0.063,-0.009,-0.018,0.027,0.073,NAN
0.007,0.093,0.070,0.037,-0.031,0.013,-0.011,0.027,0.064,NAN
0.019,0.114,0.084,0.065,0.008,0.065,0.038,0.086,0.134,NAN
0.081,0.179,0.143,0.112,0.063,0.106,0.097,0.141,0.184,NAN
0.106,0.215,0.172,0.139,0.085,0.137,0.132,0.187,0.241,NAN

Yeah sorry I did change it but forgot to upload the new firmware :-/ but everything else is the same.

Regarding multiple probing good idea ill change that.

Looks like I need to run G29 P3 to fill in the unfilled areas. I thought it should have done this the first time I called it though?

bjarchi, 2017-12-10:
> Looks like I need to run G29 P3 to fill in the unfilled areas. I thought it should have done this the first time I called it though?

`G29 P3` will fill in one row/column per call, so you have more fine-grained control over the filling, which means that you need to call it multiple times if you have more than one row or column to fill. I'm pretty sure I mention this in the new UBL docs page(s) - have you read through them recently? I did an update for 1.1.6 a few weeks back. 

It looks like things are behaving correctly now - even with multple probing enabled.

Festivejelly, 2017-12-10:
Yep all looks good now. Im going to do a test print. I dont get on very well with G26 the jerky fast movements tend to make bed adhesion a pain. I have a 300 x 300 hollow spiral square im going to try.

Roxy-3D, 2017-12-11:
> I don't get on very well with G26 the jerky fast movements tend to make bed adhesion a pain. I have a 300 x 300 hollow spiral square im going to try.

G26 uses your Configuration.h settings to make reasonable guesses about what it takes to draw a line on your printer.    If you have crazy settings....  You might get 'jerky' or other behavior.    One of the things on my list is to add a parameter to G26 to scale the Configuration.h feedrates[].    Its probably 10 or 15 lines of code to do it.    I just haven't done it yet.

bjarchi, 2017-12-11:
@Roxy-3D I'm thinking about looking into the defaults for MESH_[MIN,MAX] in conditionals_post.h to see if I can come up with better values. Right now they're shared with mesh bed leveling - I'm thinking the cleanest solution might be to split UBL and MBL for those defaults, and use the (commented) Configuration_adv.h defaults for UBL:
MIN = MESH_INSET_foo
MAX = BED_SIZE_foo - MESH_INSET_foo
(for foo = [X,Y]

Does this seem like a reasonable way to go for default values? I think it's the expected behavior - mesh bed leveling uses PROBE_MIN to try to size the mesh to the area that can be probed, but that clearly isn't what we want for UBL.

Roxy-3D, 2017-12-11:
A couple of comments...

That probably works OK for Cartesian printers, but a different strategy would be needed for Delta's.

The Mesh Inset idea came from the original Mesh Bed Leveling.   And UBL does respect the Mesh Inset with regard to probing.   I wonder if it would make sense to change the original Mesh Bed Leveling to also respect the inset for its probing?

bjarchi, 2017-12-11:
> That probably works OK for Cartesian printers, but a different strategy would be needed for Delta's

Yes. I'll have to look at the code again but I think it calculates the mesh boundaries differently for the different machine classes. Definitely does for `PROBE_[MIN,MAX]`. I'll look into it today. 

> The Mesh Inset idea came from the original Mesh Bed Leveling. And UBL does respect the Mesh Inset with regard to probing. I wonder if it would make sense to change the original Mesh Bed Leveling to also respect the inset for its probing?

Yeah I think a longer-term solution will be more complex - like maybe this, or maybe separating the mesh_inset and probe_inset so UBL can have a full-bed mesh w/o trying to probe all the way to the edges. Right now I'm just thinking about how to fix the immediate problem of `MESH_[MIN,MAX]` defaults being affected by the probe offset w/ UBL.

bjarchi, 2017-12-11:
So here's what we have now (in conditionals_post.h):

```
#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_UBL)
    #if IS_KINEMATIC
      // Probing points may be verified at compile time within the radius
      // using static_assert(HYPOT2(X2-X1,Y2-Y1)<=sq(DELTA_PRINTABLE_RADIUS),"bad probe point!")
      // so that may be added to SanityCheck.h in the future.
      #define _MESH_MIN_X (MIN_PROBE_X + MESH_INSET)
      #define _MESH_MIN_Y (MIN_PROBE_Y + MESH_INSET)
      #define _MESH_MAX_X (MAX_PROBE_X - (MESH_INSET))
      #define _MESH_MAX_Y (MAX_PROBE_Y - (MESH_INSET))
    #else
      // Boundaries for Cartesian probing based on set limits
      #define _MESH_MIN_X (max(X_MIN_BED + MESH_INSET, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
      #define _MESH_MIN_Y (max(Y_MIN_BED + MESH_INSET, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
      #define _MESH_MAX_X (min(X_MAX_BED - (MESH_INSET), X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
      #define _MESH_MAX_Y (min(Y_MAX_BED - (MESH_INSET), Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
    #endif
```

Already splits delta and SCARA from  cartesian. I think all I need to do to fix the specific issue that arose here is to split the definition between MESH_BED_LEVELING and AUTO_BED_LEVELING_UBL, and for UBL make the defaults for MESH_[MIN,MAX] what used to be defined in Configuration_adv.h. I'll set up PRs for this when I have some cycles, maybe later tonight.

Roxy-3D, 2017-12-15:
@bjarchi    I'm not getting much time on the computer right now.   Please don't misinterpret my quietness for lack of interest.

> I think all I need to do to fix the specific issue that arose here is to split the definition between MESH_BED_LEVELING and AUTO_BED_LEVELING_UBL, and for UBL make the defaults for MESH_[MIN,MAX] [...]

Yes.   And from a higher level perspective, we have been trying to get all of the various bed leveling schemes to use common variable names and controlling defines.  Getting UBL and the original Mesh bed leveling more in tune with each other is a good thing.   The foundation that UBL was built on was very much the original Mesh Bed Leveling.

boelle, 2019-02-19:
@Festivejelly  have you tried latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle