MarlinFirmware/Marlin #8727: Probe Offset during calibration (1.1.x-bugfix)?

rszasz, 2017-12-09:
I just got a bltouch probe installed it on my delta printer, measured and set the PROBE_OFFSET_FROM_EXTRUDER definitions and am a bit confused about what should be happening when auto leveling is run. Shouldn't auto leveling start by aligning the probe with the center of the bed? Right now it seems to be ignoring the X and Y offsets and aligning the print nozzle instead.

Festivejelly, 2017-12-09:
Only when probing its offset. When you do normal moves it will not be offset.

bjarchi, 2017-12-09:
@rszasz This depends on which auto-leveling scheme you're using, and what commands you're running. Can you give more detail on your configuration and what you're trying to do? 

Specifically:
* What flavor of ABL you have defined as active (`#define AUTO_BED_LEVELING_foo` in Configuration.h)
* What commands are you running when you see behavior different from expected?
* Please zip up your Configuration.h and Configuration_adv.h and attach them to a reply (drag into the text box w/ most browsers)
* Also will be helpful to know what commit of `bugfix-1.1.x` you're compiling/running (when you cloned the repo) if this turns out to be a bug.

rszasz, 2017-12-09:
I set the UBL method as active, I noticed the same issues with both UBL and Delta configuration from the LCD control panel.

Homing on the delta is in the fully retracted position, rather than contacting the build plate, so the nozzle is centered, and the probe is offset, I dont use z_safe_homing

I'm using 95590adf07d4390dfd9b7e94e159a2662382dfce as a base to work from,

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1545174/config.zip)

rszasz, 2017-12-09:
probing with G30 commands seems to work as expected, (g30 e x0 y0 drops the probe, offsets to put the probe at 0,0 and probes properly. From the fully retracted home position it didn't respect the stops though and tried to drive one of the carriages into the stop.

thinkyhead, 2017-12-10:
Give the current `bugfix-1.1.x` code a test so we have a common baseline to find and fix the issue. It is possible we've fixed it already, or possibly it is worse! But we have to work from the current point.

rszasz, 2017-12-10:
Derpety derp. I'm still not used to sourcetree, pulling current and retrying

rszasz, 2017-12-10:
Grabbed the latest and cant compile, getting the following err in the arduino compiler

sketch\Marlin_main.cpp: In function 'void gcode_M18_M84()':

Marlin_main.cpp:8248: error: 'class unified_bed_leveling' has no member named 'lcd_map_control'

       ubl.lcd_map_control = defer_return_to_status = false;

           ^

sketch\Marlin_main.cpp: In function 'void manage_inactivity(bool)':

Marlin_main.cpp:13464: error: 'class unified_bed_leveling' has no member named 'lcd_map_control'

       ubl.lcd_map_control = defer_return_to_status = false;

           ^

exit status 1
'class unified_bed_leveling' has no member named 'lcd_map_control'

rszasz, 2017-12-10:
Disabled all but the Delta_auto_calibration and it compiled, A G33 after a G28 however seemed to ignore the probe offset. Perhaps the offset should just be subtracted from the standard probing radius for doing the delta calibration probes? I ended up shrinking the calibration radius until the probe stayed on the build platform during the entire calibration cycle and it seems to work

LVD-AC, 2017-12-11:
G33 ignores XY offsets because for calibration purpose one needs to know the probe results at certain positions of the carriages, rods and effector. So this is indeed the proper procedure for G33: shrink the calibration radius until the probe stays on the bed (this can be outside the printable radius !!!) and you're good to go. G29 (and G30) should take into account the XY offsets since bed levelling routines need to know the probe results not at nozzle position but at certain bed positions.

rszasz, 2017-12-11:
I guess this is turning into more of a feature request I guess. On the anycubic kossel the bed is just about the same size as the maximum build area . G29-30 make sure the probe stays on the bed, G33 doesn't. It would be nice if G33 at least sanity checked to make sure the probe would stay within either the maximum build area, or the probe area. (Automatically calculating the G33 probe area to keep the probe in the defined probing radius would be nice, and should be fairly simple)

LVD-AC, 2017-12-12:
Calibration is more precise at a larger calibration radius. This implies that deltas that use the nozzle to probe can be calibrated better then deltas with eccentric probes; and deltas with an eccentric probe that have a larger bed and as such some extra space where the probe can trigger can be calibrated better then deltas with eccentric probes with a more restricted bed size.

G33 needs to cater for all; not only the anycubic who seems to be the most restrictive: eccentric probe with almost no extra space on the bed for the probe to move outside the print radius.

BTW a "probing radius" (= limits where the probe still hits the bed and triggers) is undefined; hence the need to set a calibration radius.

boelle, 2019-02-19:
@rszasz tried the latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek