MarlinFirmware/Marlin #8731: [1.1.6] Compile error for delta configuration

blackus3r, 2017-12-09:
## Compile Error
When I compile I get an error:
```
sketch/ultralcd.cpp.o: In function `manage_manual_move()':
sketch/ultralcd.cpp:2770: undefined reference to `set_destination_to_current()'
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.
```

Compiling with an example configuration for deltas also didn't worked. 

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1544730/configuration.zip)

Tannoo, 2017-12-09:
Try `bugfix-1.1.x` and see if the issue is still there.

lumbric, 2017-12-10:
I stumpled upon a very similar problem. I wasn't able to reproduce the build error with your configuration (I get different errors, such as "You are using an old Configuration.h file, update it before building Marlin"), but I think pull request #8738 should fix your issue. You can try to apply only this patch and tell us here if it fixes your issue.

ethanzwolf, 2017-12-11:
I'm having this problem with compiling once I've added the example kossel_mini configuration.h and configuration_adv.h files to marlin. Once those files are added, I get this error message when trying to compile: 

Arduino: 1.6.7 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\ultralcd.cpp.o: In function `manage_manual_move()':
sketch/ultralcd.cpp:2784: undefined reference to `set_destination_from_current()'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling.

I tried downloading the latest marlin bugfix and it still didn't work. As soon as those example configuration files are added, the program won't compile.

fiveangle, 2017-12-11:
@lumbric - make is not used by Arduino IDE or PlatformIO on Windows platform, so not related to this issue.

@ethanzwolf - issue is fixed in bugfix1.1.x

```
avr-objcopy -O ihex -R .eeprom .pioenvs/megaatmega2560/firmware.elf .pioenvs/megaatmega2560/firmware.hex
avr-size --mcu=atmega2560 -C -d .pioenvs/megaatmega2560/firmware.elf
AVR Memory Usage
----------------
Device: atmega2560
Program:   97598 bytes (37.2% Full)
(.text + .data + .bootloader)
Data:       3141 bytes (38.3% Full)
(.data + .bss + .noinit)
========================= [SUCCESS] Took 10.33 seconds =========================
```

@blackus3r - hand-port your config to bugfix1.1.x as the `manage_manual_move()` compile error is fixed already, but your config needs to be updated:

<details><summary>errors due to configuration format changes</summary>

```
In file included from Marlin/MarlinConfig.h:39:0,
from Marlin/G26_Mesh_Validation_Tool.cpp:27:
Marlin/SanityCheck.h:226:4: error: #error "K1 is now PID_K1. Please update your configuration."
#error "K1 is now PID_K1. Please update your configuration."
^
avr-g++ -o .pioenvs/megaatmega1280/src/Marlin.ino.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega1280 -DF_CPU=16000000L -DPLATFORMIO=30500 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA -DARDUINO=10619 -I.pioenvs/megaatmega1280/src -IMarlin -I/Users/speedster/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/speedster/.platformio/packages/framework-arduinoavr/variants/mega -IMarlin -I.piolibdeps/U8glib_ID7/src -I/Users/speedster/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SPI/src -I.piolibdeps/TMC2130Stepper/src "-I.piolibdeps/Adafruit NeoPixel" -I/Users/speedster/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/Wire/src -I.piolibdeps/LiquidTWI2 -I.piolibdeps/TMC26XStepper -I/Users/speedster/.platformio/packages/framework-arduinoavr/libraries/LiquidCrystal/src -I.piolibdeps/LiquidCrystal_I2C_ID576 Marlin/Marlin.ino.cpp
Marlin/SanityCheck.h:283:6: error: #error "MIN_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MIN_SOFTWARE_ENDSTOP_Z."
#error "MIN_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MIN_SOFTWARE_ENDSTOP_Z."
^
Marlin/SanityCheck.h:291:6: error: #error "MAX_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MAX_SOFTWARE_ENDSTOP_Z."
#error "MAX_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MAX_SOFTWARE_ENDSTOP_Z."
^
In file included from Marlin/MarlinConfig.h:39:0,
from Marlin/I2CPositionEncoder.cpp:31:
Marlin/SanityCheck.h:226:4: error: #error "K1 is now PID_K1. Please update your configuration."
#error "K1 is now PID_K1. Please update your configuration."
^
Marlin/SanityCheck.h:283:6: error: #error "MIN_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MIN_SOFTWARE_ENDSTOP_Z."
#error "MIN_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MIN_SOFTWARE_ENDSTOP_Z."
^
Marlin/SanityCheck.h:291:6: error: #error "MAX_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MAX_SOFTWARE_ENDSTOP_Z."
#error "MAX_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MAX_SOFTWARE_ENDSTOP_Z."
^
In file included from Marlin/MarlinConfig.h:39:0,
from Marlin/M100_Free_Mem_Chk.cpp:51:
Marlin/SanityCheck.h:226:4: error: #error "K1 is now PID_K1. Please update your configuration."
#error "K1 is now PID_K1. Please update your configuration."
^
Marlin/SanityCheck.h:283:6: error: #error "MIN_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MIN_SOFTWARE_ENDSTOP_Z."
avr-g++ -o .pioenvs/megaatmega1280/src/MarlinSerial.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega1280 -DF_CPU=16000000L -DPLATFORMIO=30500 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA -DARDUINO=10619 -I.pioenvs/megaatmega1280/src -IMarlin -I/Users/speedster/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/speedster/.platformio/packages/framework-arduinoavr/variants/mega -IMarlin -I.piolibdeps/U8glib_ID7/src -I/Users/speedster/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SPI/src -I.piolibdeps/TMC2130Stepper/src "-I.piolibdeps/Adafruit NeoPixel" -I/Users/speedster/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/Wire/src -I.piolibdeps/LiquidTWI2 -I.piolibdeps/TMC26XStepper -I/Users/speedster/.platformio/packages/framework-arduinoavr/libraries/LiquidCrystal/src -I.piolibdeps/LiquidCrystal_I2C_ID576 Marlin/MarlinSerial.cpp
#error "MIN_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MIN_SOFTWARE_ENDSTOP_Z."
^
Marlin/SanityCheck.h:291:6: error: #error "MAX_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MAX_SOFTWARE_ENDSTOP_Z."
#error "MAX_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MAX_SOFTWARE_ENDSTOP_Z."
^
In file included from Marlin/MarlinConfig.h:39:0,
from /Users/speedster/dev/github/Marlin/Marlin/Marlin.ino:31:
Marlin/SanityCheck.h:226:4: error: #error "K1 is now PID_K1. Please update your configuration."
#error "K1 is now PID_K1. Please update your configuration."
^
Marlin/SanityCheck.h:283:6: error: #error "MIN_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MIN_SOFTWARE_ENDSTOP_Z."
#error "MIN_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MIN_SOFTWARE_ENDSTOP_Z."
^
Marlin/SanityCheck.h:291:6: error: #error "MAX_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MAX_SOFTWARE_ENDSTOP_Z."
#error "MAX_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MAX_SOFTWARE_ENDSTOP_Z."
^
[... cut ...]
Marlin/SanityCheck.h:283:6: error: #error "MIN_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MIN_SOFTWARE_ENDSTOP_Z."
#error "MIN_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MIN_SOFTWARE_ENDSTOP_Z."
^
Marlin/SanityCheck.h:291:6: error: #error "MAX_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MAX_SOFTWARE_ENDSTOP_Z."
#error "MAX_SOFTWARE_ENDSTOPS on DELTA/SCARA also requires MAX_SOFTWARE_ENDSTOP_Z."
^
*** [.pioenvs/megaatmega1280/src/Sd2Card.o] Error 1
*** [.pioenvs/megaatmega1280/src/Marlin_main.o] Error 1
```

</details>

lumbric, 2017-12-11:
@fiveangle ah yes, of course. I will remove the "fixes #8731" in my pull request again.

jorgempy, 2017-12-13:
Hi, i just cloned the repo, first tried the v1.1.6, and then i checked out the bugfix-1.1.x branch. I reapplied my configuration changes, and the error is the same,:
    ` ultralcd.cpp:2783: undefined reference to 
  'set_destination_from_current()'`

With arduino-ide-1.6.9. Please Help.

jorgempy, 2017-12-13:
Managed to compile, copied the line
    ` inline void set_destination_from_current() { COPY(destination, current_position); }` 
just above 
 `  inline void manage_manual_move() { ` (line ~2768).
in ultralcd.cpp

blackus3r, 2017-12-14:
Thanks for your help.
I already had outcommented the line, and the printer is working fine. I will give it a try with bugfix1.1.x.

commit ea345b1d (2017-12-20): Merge pull request #8738 from lumbric/bugfix-1.1.x

Add missing CPP files to Make file, fixes #8731
closed 2017-12-16 by thinkyhead