MarlinFirmware/Marlin #8732: Auto bed leveling with 3POINT and PROBE_MANUALLY does not move head after G29

lumbric, 2017-12-09:
### Bug Report
* Description: Trying to use auto bed leveling with manual probing and 3 point leveling.
* Expected behaviour: After the first G29 probe moves in X and Y direction to the first probe point.
* Actual behaviour: Probe does not move after the first G29.
* Steps to reproduce:
    - Set config for Auto bed leveing
        - AUTO_BED_LEVELING_3POINT
        - PROBE_MANUALLY
        - DEBUG_LEVELING_FEATURE
        - EEPROM_SETTINGS
        - points to measure:
            -  ABL_PROBE_PT_1_X -40
            - ABL_PROBE_PT_1_Y 40
            - ABL_PROBE_PT_2_X 70
            - ABL_PROBE_PT_2_Y 10
            - ABL_PROBE_PT_3_X -20
            - ABL_PROBE_PT_3_Y -80
        - LCD_BED_LEVELING (not used for reproducing, though)
    - M111 S247
    - G28
    - G29
    - --> Marlin prints the following but doesn't move the head:
```
Send: G28
Recv: echo:G29
Recv:   current_position=(0.00, -0.00, 219.07) :  >>> gcode_G29
Recv: Machine Type: Delta
Recv: Probe: PROBE_MANUALLY
Recv: Auto Bed Leveling: 3POINT (disabled)
Recv:
Recv:   current_position=(0.00, -0.00, 219.07) : sync_plan_position_kinematic
Recv:   current_position=(0.00, -0.00, 219.07) : setup_for_endstop_or_probe_move
Recv: > 3-point LevelingRecv: ok
```
Tested with: [Configuration.h](https://github.com/lumbric/Marlin/blob/972e3a5ae62fae15b4e25f41255936b86911490c/Marlin/Configuration.h) [Configuration_adv.h](https://github.com/lumbric/Marlin/blob/972e3a5ae62fae15b4e25f41255936b86911490c/Marlin/Configuration_adv.h)

Marlin prints `Auto Bed Leveling: 3POINT (disabled)`, what does this "disabled" mean?

Spawn32, 2017-12-09:
You have enabled #define AUTO_BED_LEVELING_GRID, so the if / else statement will select that and not 3-point..

lumbric, 2017-12-09:
Uh sorry, I've linked the wrong config files (in the wrong branch). I've fixed the links above. See [here](https://github.com/lumbric/Marlin/blob/d73d05c78a29b75c348411b0e3f27cf46e8e805a/Marlin/Configuration.h#L979).

thinkyhead, 2017-12-16:
Please test with `bugfix-1.1.x` because 1.1.6 is very old.

thinkyhead, 2017-12-25:
Try it with `G29 V4` to get more output from each invocation.

thinkyhead, 2017-12-25:
See if the patch linked above ([download branch](https://github.com/thinkyhead/Marlin/archive/bf1_action_on_action.zip)) fixes the issue.

lumbric, 2018-01-12:
G29 looks better now, thanks! Unfortunately the calibration does not what I expected. The head moves in Z when moving in X/Y after doing the G29.
I tested by cherry-picking 4f943a73bc on 1.1.7 (because my config is not ported to 1.1.8 yet), see [here](https://github.com/lumbric/Marlin/tree/test_issue_8732).

Is this a known thing? I can test it also with 1.1.8 if you think that would help, but I don't need i right now for myself.

thinkyhead, 2018-02-11:
> The head moves in Z when moving in X/Y after doing the G29.

As it should.

lumbric, 2018-02-11:
Hm I think I didn't describe the behavior very well. The head is not moving evenely in z when moving in x/y, but jumping up and down. I can try to make a video if you think it is worth debugging.

thinkyhead, 2018-02-14:
Surely, I would like to see what is happening. 3-point leveling should definitely not cause Z to move up and down, but always in one direction when moving X or Y in one direction.

boelle, 2019-02-19:
@lumbric have you tried latest bugfix 2.0?

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-06 by shitcreek

Youri-anet-a8, 2019-12-21:
hi I have this problem too plzz some one can you help?!